linux-sunxi/meta-sunxi #79: linux: do_configure fails - no rule to make target 'oldconfig'

KristofRobot, 2015-01-31:
When building against newer yocto project releases below failure
occurs.

```
DEBUG: Executing shell function do_configure
NOTE: make oldconfig
make: *** No rule to make target `oldconfig'.  Stop.
ERROR: oe_runmake failed
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_configure
```

Apparently this is already fixed in meta-raspberrypi, see https://github.com/djwillis/meta-raspberrypi/commit/1eda7e8b742bff698456facf4e8e79c3457cffc0

We should do something similar.

KristofRobot, 2015-02-01:
Note: I am working on this - should be able to submit a pull request shortly.

closed 2015-02-22 by naguirre

abehterev, 2015-08-14:
I have this trouble when i try build system for my cubie 1.

local.conf
```
BB_NUMBER_THREADS ?= "2"
PARALLEL_MAKE ?= "-j 2"
MACHINE ?= "cubieboard"
```
bblayers.conf
```
LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/bai/embedded/poky/meta \
  /home/bai/embedded/poky/meta-yocto \
  /home/bai/embedded/poky/meta-yocto-bsp \
  /home/bai/embedded/poky/meta-sunxi \
  "
BBLAYERS_NON_REMOVABLE ?= " \
  /home/bai/embedded/poky/meta \
  /home/bai/embedded/poky/meta-yocto \
  "

```
Truble:
```
bai@ubuntu-cons:~/embedded/poky/cubieboard$ bitbake core-image-minimal
Loading cache: 100% |##################################################################################################################| ETA:  00:00:00
Loaded 1301 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.26.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubieboard"
DISTRO            = "poky"
DISTRO_VERSION    = "1.8"
TUNE_FEATURES     = "arm armv7a vfp neon"
TARGET_FPU        = "vfp-neon"
meta
meta-yocto
meta-yocto-bsp    = "fido:08f9fbbc97909698b56200aa9be5fe50b99a44b2"
meta-sunxi        = "master:9ac263ebab41ea5229d4c74ff837ae6a623536b9"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: Function failed: do_configure (log file is located at /home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/temp/log.do_configure.20650)
ERROR: Logfile of failure stored in: /home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/temp/log.do_configure.20650
Log data follows:
| DEBUG: Executing python function sysroot_cleansstate
| DEBUG: Python function sysroot_cleansstate finished
| DEBUG: Executing shell function do_configure
| NOTE: make -C /home/bai/embedded/poky/cubieboard/tmp/work-shared/cubieboard/kernel-source O=/home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/build oldconfig
| make: Entering directory `/home/bai/embedded/poky/cubieboard/tmp/work-shared/cubieboard/kernel-source'
| make: *** No rule to make target `oldconfig'.  Stop.
| make: Leaving directory `/home/bai/embedded/poky/cubieboard/tmp/work-shared/cubieboard/kernel-source'
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/temp/log.do_configure.20650)
ERROR: Task 74 (/home/bai/embedded/poky/meta-sunxi/recipes-kernel/linux/linux_4.1.3.bb, do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1035 tasks of which 1033 didn't need to be rerun and 1 failed.
Waiting for 0 running tasks to finish:

Summary: 1 task failed:
  /home/bai/embedded/poky/meta-sunxi/recipes-kernel/linux/linux_4.1.3.bb, do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
* Folder "/home/bai/embedded/poky/cubieboard/tmp/work-shared/cubieboard/kernel-source" is empty.
* Kernel folder isn't empty:
```
bai@ubuntu-cons:~/embedded/poky/cubieboard$ ls -la /home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/
total 40
drwxrwxr-x  5 bai bai  4096 Aug 14 16:49 .
drwxrwxr-x  3 bai bai  4096 Aug 14 16:47 ..
drwxr-xr-x  2 bai bai  4096 Aug 14 17:07 build
-rw-r--r--  1 bai bai 17866 Aug 14 16:49 defconfig
drwxr-xr-x 23 bai bai  4096 Jul 21 20:10 linux-4.1.3
drwxrwxr-x  2 bai bai  4096 Aug 14 18:16 temp
```
* Folder "/home/bai/embedded/poky/cubieboard/tmp/work/cubieboard-poky-linux-gnueabi/linux/4.1.3-r0/build" contain ".config".
* File "linux.inc" same as in meta-oe.

commit e25b18fd (2015-02-22): linux: replacing custom linux.inc by openembedded linux.inc reference (resolves #79)