linux-sunxi/meta-sunxi #83: Migrate to Yocto Linux Kernel

KristofRobot, 2015-02-02:
I think it would be a good idea to migrate to using Yocto Linux Kernel recipes - to avoid issues like #79 

See http://www.yoctoproject.org/docs/1.7.1/kernel-dev/kernel-dev.html:

>  The Yocto Project also provides a powerful set of kernel tools for managing Linux kernel sources and configuration data. You can use these tools to make a single configuration change, apply multiple patches, or work with your own kernel sources. [...]

> If you do not maintain your own kernel sources and need to make only minimal changes to the sources, the released recipes provide a vetted base upon which to layer your changes. Doing so allows you to benefit from the continual kernel integration and testing performed during development of the Yocto Project.

I think meta-sunxi is in this case?

> If, instead, you have a very specific Linux kernel source tree and are unable to align with one of the official linux-yocto recipes, an alternative exists by which you can use the Yocto Project Linux kernel tools with your own kernel sources. 

Or are we in this case?

slapin, 2015-02-02:
I geerally disagree to this. Yocto kernel recipes were done to make task
easy for Intel processes.
Nobody else really use them.

KristofRobot, 2015-02-14:
> Yocto kernel recipes were done to make task easy for Intel processes.

It is true Intel is a participating company in the Yocto project [1]:

```
Companies contributing to the project today now include Cavium Networks, Dell, Freescale Semiconductor, Intel, LSI, Mentor Graphics, Mindspeed, MontaVista Software, NetLogic Microsystems, RidgeRun, Texas Instruments, Tilera, Timesys, and Wind River, among others.
```

However, as far as I know, there is nothing that really limits them to Intel. Reading their project charter [2], I do not see any reference to Intel, let alone a limitation to Intel hardware. Quite the contrary, they explicitly state:

```
To provide a place of collaboration for open source projects that further the development of customizable Linux platforms and embedded-style Linux-based systems.
[...]
To ensure as much as possible that the tooling for these developments is architecturally independent.
```

Also, Wikipedia [2] states:

```
The Yocto Project has the aim and objective of attempting to improve the lives of developers of customised Linux systems supporting the ARM, MIPS, PowerPC and x86/x86 64 architectures. 
```

Lastly, looking at Yocto's official hardware support list [4], I find following officially supported architectures/hardware - not all Intel:

```
  * ARM (qemuarm)
  * x86 (qemux86)
  * x86-64 (qemux86-64)
  * PowerPC (qemuppc)
  * MIPS (qemumips)

[...]

  * Texas Instruments Beaglebone (beaglebone)
  * Freescale MPC8315E-RDB (mpc8315e-rdb)

[...]

  * Intel x86 based PCs and devices (genericx86)
  * Ubiquiti Networks EdgeRouter Lite (edgerouter)
```

Or am I missing something?

[1] http://www.linuxfoundation.org/news-media/announcements/2011/03/yocto-project-aligns-technology-openembedded-and-gains-corporate-co
[2] https://www.yoctoproject.org/about/yocto-project-charter
[3] https://en.wikipedia.org/wiki/Yocto_Project
[4] http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/README.hardware?h=dizzy

KristofRobot, 2015-02-14:
To give a more practical example, the BSP layer for Raspberry Pi (1 and 2) is maintained under the Yocto project, and leverages the Yocto Linux Kernel. [1]

[1] http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/recipes-kernel/linux
[1]

ebutera, 2015-02-14:
I think there is just some confusion because we say "kernel recipes", while actually it's kernel recipe and kernel recipe include (linux.inc).

So i agree we don't need a custom *include* because we do nothing special there, but of course we need a custom kernel recipe to point to linux-sunxi kernel.
And that's only about kernel 3.4, for mainline we could just use official recipes too, the machine configuration should be enough to build a working kernel+devicetree (i never tested this).

Noxenious, 2015-02-25:
Sorry for jumping in on this issue, but:
I did want to commit some changes myself and to test them I wanted to setup a new build environment. Up until now I just could get Yocto, get the meta-sunxi layer and build images. With this change there is now a dependency on meta-oe, because meta-sunxi needs the linux.inc which comes only with meta-oe. Not an unsolveable problem to get the meta-oe, but I did like the self-containing layer better.

slapin, 2015-02-25:
So I'd add linux.inc back, as this is common practice to have this one in
BSP.

ebutera, 2015-02-25:
i agree, i tought the linux.inc was in oe core but i was wrong, sorry.
Just add it and we will try to keep it in sync with upstream then.

naguirre, 2015-03-02:
I did the same mistake, #86 fix it. I close the issue.

closed 2015-03-02 by naguirre