linux-sunxi/meta-sunxi #88: make: *** No rule to make target `A20-OLinuXino-Lime2_config'.  Stop.

shan-x, 2015-03-26:
When compiling an image for Olimex Lime2, I have this error :

    ERROR: Function failed: do_compile (log file is located at /home/user/yocto2/build/olimex/output/work/olinuxino_a20lime2-poky-linux-gnueabi/u-boot/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.29222)
    ERROR: Logfile of failure stored in: /home/user/yocto2/build/olimex/output/work/olinuxino_a20lime2-poky-linux-gnueabi/u-boot/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.29222
    Log data follows:
    | DEBUG: Executing shell function do_compile
    | NOTE: make -j 16 CROSS_COMPILE=arm-poky-linux-gnueabi- CC=arm-poky-linux-gnueabi-gcc  --sysroot=/home/user/yocto2/build/olimex/output/sysroots/olinuxino-a20lime2 A20-OLinuXino-Lime2_config
    | make: *** No rule to make target `A20-OLinuXino-Lime2_config'.  Stop.
    | make: *** [A20-OLinuXino-Lime2_config] Error 1
    | ERROR: oe_runmake failed
    | WARNING: /home/user/yocto2/build/olimex/output/work/olinuxino_a20lime2-poky-linux-gnueabi/u-boot/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/run.do_compile.29222:1 exit 1 from
    |   exit 1
    | ERROR: Function failed: do_compile (log file is located at /home/user/yocto2/build/olimex/output/work/olinuxino_a20lime2-poky-linux-gnueabi/u-boot/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.29222)
    ERROR: Task 262 (/home/user/yocto2/layers-externes/meta-sunxi/recipes-bsp/u-boot/u-boot_git.bb, do_compile) failed with exit code '1'
    NOTE: Tasks Summary: Attempted 5001 tasks of which 4998 didn't need to be rerun and 1 failed.
    Waiting for 0 running tasks to finish:

    Summary: 1 task failed:
    /home/user/yocto2/layers-externes/meta-sunxi/recipes-bsp/u-boot/u-boot_git.bb, do_compile

I use poky and openembedded up to date.

Noxenious, 2015-04-02:
Oh sorry adding the lime2 was my commit, although I tried it it seems a little error slipped into the config.

The U-Boot name for the lime2 is not A20-OLinuXino-Lime2 but A20-OLinuXino_Lime2 (where the name for the A-20-Lime is A20-OLinuXino-Lime - all the different dashes ...)

There is also a little problem with the kernel config for the lime2 I have fixed, I will do a new push request with my patches.

shan-x, 2015-04-02:
Thanks for your answer, I'll wait for the patches.

Btw, I don't know if it's linked, but it seems that the ethernet card isn't supported (no way to bring eth0 up). But I don't know to what it's due.

Noxenious, 2015-04-02:
I just did a pull request, you can wait for them to be integrated or manually integrated them (just need to change two files).

Yes the ethernet issue was the other issue I was talking about. I still had the emac (ethernet 10/100) driver enabled, when i did some tests with the lime2 I saw that it needed the gmac( gigabit ethernet) driver. This has also been fixed in my pull request.

See
https://github.com/linux-sunxi/meta-sunxi/pull/89/files

shan-x, 2015-04-27:
Your patch has corrected the problems, thanks a lot.
Other problem, maybe related (again) : the nand memory doesn(t work. When booting with a Debian, I can see /dev/nand1 and /dev/nand2. None of them appear with the image generated. Is it a problem with the BSP ?

Noxenious, 2015-04-27:
The NAND driver is not enabled in the kernel. You just need to add the following two lines to your defconfig (/recipes-kernel/linux/linux/olinuxino-a20lime2/defconfig)

CONFIG_SUNXI_NAND=y
CONFIG_SUNXI_NAND_COMPAT_DEV=y

I can enable them in a later commit.

shan-x, 2015-05-28:
It's me, again.
It seems that the network doesn't work with u-boot (no dhcp request sent to the server, even with the dhcp command). Is it related ?

closed 2015-08-10 by ebutera