linux-sunxi/meta-sunxi #94: Socarch definition missing?

Noxenious, 2015-06-25:
In sunxi.inc the following definition is done:

# Sub-architecture support
MACHINE_SOCARCH_SUFFIX ?= ""
MACHINE_SOCARCH_SUFFIX_sun4i = "-sun4i"

does it not need to define the other sufixes as well?

e.g.
MACHINE_SOCARCH_SUFFIX_sun5i = "-sun5i"
MACHINE_SOCARCH_SUFFIX_sun7i = "-sun7i"

?

Noxenious, 2015-06-25:
and another issue: it includes

include conf/machine/include/soc-family.inc

but where is this file?

naguirre, 2015-06-25:
soc-family.inc is included from here : https://github.com/openembedded/oe-core/blob/master/meta/conf/machine/include/soc-family.inc

About the subarch support, i need to check.

Noxenious, 2015-06-25:
Should it not be included in the meta-sunxi layer? Otherwise it depens on meta-oe (was the same case for linux.inc in kernel)

naguirre, 2015-06-26:
No, it depends on oe-core, which is the default one.
linux.inc is included in meta-oe wich is a different layer.

closed 2019-08-07 by nandra