linux-sunxi/meta-sunxi #97: meta-sunxi/conf/machine/olinuxino-a20.conf - wrong UBOOT_MACHINE name

gschmottlach, 2015-07-03:
In the master branch in the file:

 meta-sunxi/conf/machine/olinuxino-a20.conf

The UBOOT_MACHINE name is incorrect.

It is currently:

UBOOT_MACHINE = "A20-OLinuXino-Micro_config"

But must be:

UBOOT_MACHINE = "A20-OLinuXino_MICRO_defconfig"

naguirre, 2015-07-03:
Could you please provide a pull request ?

gschmottlach, 2015-07-03:
Sorry, I think you guys already fixed this issue in your u-boot_2015.07rc2.bb recipe where you override the UBOOT_MACHINE to be the correct name. I guess I missed that. Not sure if that's the best solution since it's still incorrect in:

meta-sunxi/conf/machine/olinuxino-a20.conf

according to the upstream u-boot naming convention. Honestly, I'm not sure how you'd like to best address this. Shouldn't both the recipe (u-boot_2015.07rc2.bb) and the olinuxino-a20.conf file use consistent names? If so you wouldn't need to override the UBOOT_MACHINE in the recipe.

ebutera, 2015-07-17:
since it is a rc release, so not really "supported", it is ok to override it in the recipe, now that 2015.07 was released the correct way to do it should be to add the recipe and keep consistent names as you suggested.

Noxenious, 2015-07-17:
Actually the overwrite was for the difference between sunxi-uboot and mainline u-boot. First idea was to default to sunxi-uboot, but now it is default mainline. They both use different names, the one in the machine config is for sunxi-u-boot, the overwrite is for mainline. I think i can manage a little bit time today or on monday to change that and update to release 2015.07,

Noxenious, 2015-09-22:
The configs have been switched to use mainline u-boot name. The overwrite is done for old sunxi-uboot. Done in commit be0e113009761d9c38e673d2654dd073091d3a91
Can be closed?

naguirre, 2015-09-22:
Yes. it's ok

closed 2015-09-22 by naguirre