linux-sunxi/meta-sunxi #106: Image build fails for some machines due to missing/wrong fex support

koenkooi, 2015-07-31:
'meleg' and 'olinuxino-a10' fail for me due to the fex stuff failing to build. With a recent u-boot and mainline being the default, do we still need to build the fex stuff by default?

naguirre, 2015-07-31:
What's the error you get ?
Does sunxi boards boot without fex ? I think that we still need it, i didn't tried with recent uboot though.

koenkooi, 2015-07-31:
I don't know why I said fex, the error is in u-boot:

Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j8 CROSS_COMPILE=arm-angstrom-linux-gnueabi- CC=arm-angstrom-linux-gnueabi-gcc  --sysroot=/build/v2014.12/build/tmp-angstrom_v2014_12-glibc/sysroots/meleg Mele_A1000G_config
|   HOSTCC  scripts/basic/fixdep
|   HOSTCC  scripts/kconfig/conf.o
|   SHIPPED scripts/kconfig/zconf.tab.c
|   SHIPPED scripts/kconfig/zconf.lex.c
|   SHIPPED scripts/kconfig/zconf.hash.c
|   HOSTCC  scripts/kconfig/zconf.tab.o
|   HOSTLD  scripts/kconfig/conf
| ***
| *** Can't find default configuration "arch/../configs/Mele_A1000G_defconfig"!
| ***
| scripts/kconfig/Makefile:111: recipe for target 'Mele_A1000G_defconfig' failed
| make[1]: *** [Mele_A1000G_defconfig] Error 1
| Makefile:475: recipe for target 'Mele_A1000G_config' failed
| make: *** [Mele_A1000G_config] Error 2

ebutera, 2015-07-31:
probably is another change in u-boot machines (like the previous one in olinuxino micro->MICRO etc...), i don't know why they keep changing them while keeping inconsistent up/down case...

Noxenious, 2015-08-10:
It´s not failing because of fex support, but U-Boot machine names. The fex is actually not needed but does not hurt either and is needed for older kernel/u-boot. The correct U-Boot machine name for the meleG is "Mele_A1000G_quad", i´ll make a patch for it.

Noxenious, 2015-08-10:
Which other board is failing? There is no olinuxino-a10 machine config any more, it is now named olinuxino-a10lime.

Noxenious, 2015-08-10:
Ok just did a pull request fixing the u-boot name.

ebutera, 2015-08-10:
merged

closed 2015-08-10 by ebutera