linux-sunxi/meta-sunxi #107: Why complex kernel config again?

Noxenious, 2015-08-10:
Is it now ok to add kernel config options again? When I added the kernel recipe for 4.x I had a complex config with many options enabled which I removed again:

- ebutera - 
"please as @whirm said it's better to just use the sunxi_defconfig instead of adding those files, it's much easier to maintain"

If it´s ok, I´ll add back a few options myself.

naguirre, 2015-08-10:
I preffer to use a complete defconfig as it's need anyway for modern DISTRO.
Most of embedded distro has systemd and udev, these days, so we need most of them.
If you like a smaller kernel footprint by using less options it's up to you to override this meta with a bbappend and include you own defconfig.

ebutera, 2015-08-10:
i thought that the sunxi_defconfig had at least some of the most useful options enabled, but i was wrong.
However i was more against adding a defconfig for each board, not for enabling this or that option. Of course make a pull request if you want to add other useful options :)

Noxenious, 2015-09-22:
Ok will add options that are useful along the way. Issue can be closed.

closed 2015-09-22 by Noxenious