linux-sunxi/meta-sunxi #111: Custom dts in yocto

abehterev, 2015-08-25:
Hello,
i tried build system without rtc (in cubieboard isn't active on board), and in previously versions i used a small script for sys which add i2c rtc ds1307 device as rtc and it work fine. But in current version have embedded rtc driver and i can not use external rtc  device, because internal rtc is in priority for hwclock. And i want add some new devices through IO ports. For u-boot i created own repo and link to it with new commit id and this solution work fine (i see my dtb inspl file), but kernel rewrite all dts's and i can not use this hook. What is correct way to use custom dts in meta-sunxi. Google hasn't this info. Thank you.

Noxenious, 2015-08-26:
One way to solve both problems would be to create a patch with your differences and apply it to u-boot and linux-sunxi. Just create a patch file with diff (with correct path) and add it with file://mypatch.patch to the recipe.

abehterev, 2015-08-26:
Yes, but for u-boot i know correct path for patch diff, but what is path would be used for kernel?

Noxenious, 2015-08-26:
Here is an example patch I am using (but I am using the git kernel recipe, hope this works for the tarball as well):

diff --git a/arch/arm/boot/dts/sun7i-a20-olinuxino-lime.dts b/arch/arm/boot/dts/sun7i-a20-olinuxino-lime.dts
index d54d684..9b5ac12 100644
--- a/arch/arm/boot/dts/sun7i-a20-olinuxino-lime.dts
+++ b/arch/arm/boot/dts/sun7i-a20-olinuxino-lime.dts
@@ -180,6 +199,10 @@
        status = "okay";
 };

+&rtp {
+     allwinner,ts-attached;
+};
+
 &uart0 {
        pinctrl-names = "default";
        pinctrl-0 = <&uart0_pins_a>;

abehterev, 2015-08-26:
Me too. I use kernel from git. Thank you i'll try today.
I see that you use kernel folder relativity path, and you say: 'use file://mypatch.patch' with correct path. How yocto know what patch in which moment to use (when for u-boot, when kernel)?

Noxenious, 2015-08-26:
No you just have to have a correct path inside your patch (like above) and add a 

    file://mypatch.patch 

to bb file and put the file mypatch.pach in 

    recipes-kernel/linux/linux/

the file defconfig should be in this folder

abehterev, 2015-08-26:
Thank you. This way work fine!

closed 2015-08-26 by abehterev