linux-sunxi/meta-sunxi #115: Problems with sunxi-mail on current master

Noxenious, 2015-09-22:
I am trying to compile my layer with meta-sunxi on current master (which will be Yocto 2.0). I got many build errors complaining about missing EGL, which is fixed by my current pull request, but I still get an error when bitbaking sunxi-mali:

ERROR: runstrip: ''arm-poky-linux-gnueabi-strip' --remove-section=.comment --remove-section=.note --strip-unneeded '/home/jens/poky/build/tmp/work/armv7a-vfp-neon-poky-linux-gnueabi/sunxi-mali/git-r0/sysroot-destdir/usr/lib/libMali.so.3'' strip command failed with 1 (arm-poky-linux-gnueabi-strip: /home/jens/poky/build/tmp/work/armv7a-vfp-neon-poky-linux-gnueabi/sunxi-mali/git-r0/sysroot-destdir/usr/lib/stpvDXfe: Not enough room for program headers, try linking with -N
arm-poky-linux-gnueabi-strip:/home/jens/poky/build/tmp/work/armv7a-vfp-neon-poky-linux-gnueabi/sunxi-mali/git-r0/sysroot-destdir/usr/lib/stpvDXfe[.note.gnu.build-id]: Bad value
)

The recipe already contains
INHIBIT_PACKAGE_DEBUG_SPLIT = "1"
INHIBIT_PACKAGE_STRIP = "1"
which should skip the most split actions. I am not getting this error on current yocto fido. Any ideas?

ebutera, 2015-09-22:
no idea, is libMali a precompiled binary? forgive my ignorance about mali libs!
you can search if the INHIBIT_... are not supported anymore or just their syntax changed....

Noxenious, 2015-09-23:
Had a look at the package.bbclass but the keywords are still the same. Again searching for any updates on oe I found this patch:
http://patchwork.openembedded.org/patch/95661/
which seems to be responsible for the error.

I´ll test it and prepare a push.

naguirre, 2015-09-23:
I'm using master ( 29 august version , but i didn't noticed the problem. I force a cleansstate cache for sunxi-mali, update all my meta to master, and i get the same error.
I tried the fix you mention, and it fixes the problem, nice catch !

I'm waiting for your pull request :-)

You are also using 3.4 kernel with gcc 4.9 right ?

Noxenious, 2015-09-23:
Ok pull request is there. My project uses 3.4 kernel and gcc 4.9, but I haven´t tested with it yet. I always do the fixes with a clean copy of meta-sunxi and then backport the fixes to my layer because I have changed a lot of things. Will test it with that, but with the fixes I can build a working core-image-sato.

I have seen the question on the mailinglist about mainline and mali, next thing would be to figure out what to do about it.

naguirre, 2015-09-23:
Ok thanks for the infos.

closed 2015-09-23 by naguirre