linux-sunxi/meta-sunxi #129: Yocto Project 2.0 Jethro

leon-anavi, 2016-01-07:
Hi,

When do you plan to add a branch compatible with Yocto Project 2.0 Jethro?

Best regards, Leon

naguirre, 2016-01-07:
As far as i know, master is compatible with Jethro.
At least i use it for more than 4 month now, without problem.

leon-anavi, 2016-01-08:
Hi, thank you for the feedback. Using the master branch I have successfully built core-image-minimal for A10-OLinuXino-LIME. 

Anyway do you plan to create branch jethro? A branch with that name exists for some of the other BSP layers that I am using: meta-fsl-arm and meta-raspberrypi.

Thanks, Leon

ebutera, 2016-01-08:
yes, a jethro branch will be created soon.

leon-anavi, 2016-01-08:
OK, thank you. Looking forward to it. I will close the issue.

closed 2016-01-08 by leon-anavi