linux-sunxi/meta-sunxi #131: mainline 4.1 kernel (as long as 4.4) are broken regarding input devices

slapin, 2016-02-28:
mainline 4.1 kernel (as long as 4.4) are broken regarding input devices as defconfig doesn't
contain CONFIG_INPUT_EVDEV which prevents /dev/input/eventX creation
as latest udev doesn't create device nodes and kernel doesn't pick up evdev (and keyboard + mouse are disabled too). dition of CONFIG_INPUT_EVDEV is enough to make X11/weston work with current
openembedded-core/systemd

jlucius, 2016-02-29:
Created a pull request for new kernel versions. 4.4.3 and 4.5rc6.

Should that fix your problems?

slapin, 2016-02-29:
if CONFIG_INPUT_EVDEV was added to defconfig, then I like this :)

diegorondini, 2017-04-14:
I think this can be closed now:
`# zcat /proc/config.gz | grep CONFIG_INPUT_EVDEV
CONFIG_INPUT_EVDEV=y`

closed 2017-04-15 by ebutera