linux-sunxi/meta-sunxi #134: mali driver x11 problem

yamak, 2016-03-11:
Hi,
I try to build an image for bananaPro withot x11 and wayland support.But I get following error during mali test binary compilation

> make: Leaving directory '/home/yusuf/yocto/poky/bananaProHf/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/sunxi-mali/git-r0/git/include'
make -C test test
make[1]: Entering directory '/home/yusuf/yocto/poky/bananaProHf/tmp/work/cortexa7hf-vfp-vfpv4-neon-poky-linux-gnueabi/sunxi-mali/git-r0/git/test'
arm-poky-linux-gnueabi-gcc  -march=armv7-a -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a7 --sysroot=/home/yusuf/yocto/poky/bananaProHf/tmp/sysroots/bananapro  -O2 -pipe -g -feliminate-unused-debug-types -I../include -L../../image/usr/lib -o test test.c -lEGL -lGLESv2 -lX11
In file included from ../include/EGL/egl.h:36:0,
                 from test.c:32:
../include/EGL/eglplatform.h:89:22: fatal error: X11/Xlib.h: No such file or directory
compilation terminated.
Makefile:8: recipe for target 'test' failed

When I view 0001-fix-test-build.patch file,I see following lines

>  test: ../config.mk test.c
-	$(CC) $(CFLAGS) -o $@ test.c -lEGL -lGLESv2
+	$(CC) $(CFLAGS) -I../include -L../../image/usr/lib -o $@ test.c -lEGL -lGLESv2 -lX11
 
 clean:
 	rm -f test
-- 
Since the image doesn't support X11, I assumed the X11 libraries shouldn't be included. Isn't that true?
How can I solve this problem?

Lefttyre, 2016-05-27:
I met the same issue too...

likewise, 2017-01-10:
Same issue still there in krogoth.

slapin, 2017-01-10:
Sorry for delay, I will look into this. Please report your full bitbake log
(for me to lazily test your overlay/layer setup).

likewise, 2017-01-11:
I am on krogoth branches with local.conf:
MACHINE = "olinuxino-a13"
DISTRO_FEATURES_remove = "x11 wayland"

I do not have the original log anymore, and already started hacking. So far:

The current sunxi-mali recipe supports X11 and Wayland, not without both. However I suspect it only works for X11.

I discovered a few problems so far:
1) PACKAGECONFIG is not set in case neither x11 nor wayland is set. New logic must added to cover this case. I think it is best we just set EGL_TYPE=framebuffer for the default case, and change it only for the X11 case. 

For now I hacked around this, to see for further problems:

PACKAGECONFIG = "egl"
PACKAGECONFIG[egl] = "EGL_TYPE=framebuffer,,,"

2) The EGL_TYPE set above is not passed to the do_configure stage. EXTRA_OEMAKE should be changed to EXTRA_OECONF.

do_configure() {
         DESTDIR=${D}/ VERSION=r3p0 ABI=armhf make ${EXTRA_OECONF} config

3) eglplatform_fb.h must be used instead of eglplatform.h. I am not sure (yet) if eglplatform_fb.h is intended to be renamed (moved) over eglplatform.h during install, or if applications should include the _fb.h variant of the include.

likewise, 2017-01-11:
I have forked and committed 3 patches to the krogoth branch. This is NOT YET finished. At least one remaining item is make sure the -lX11 is handled correctly for the test application build for the x11 case.

https://github.com/linux-sunxi/meta-sunxi/compare/krogoth...likewise:krogoth

nandra, 2019-05-28:
@macpijan can you pls create PR. If it is resolved I'll be happy to merge this in. :+1: Thanks.

macpijan, 2019-05-28:
@nandra Hi. I think it was not an acceptable solution as it could break other cases. I'm not using it presently so I would need to get back to it. I will try to find some time next week.

nandra, 2019-05-28:
@macpijan ok thanks a lot. I'll ping you in case you forgot ;) in 2 weeks. Thx.

nandra, 2019-08-07:
@macpijan hi any update on this? Thanks.

slapin, 2019-09-07:
Will try to build for wailand and see what is going on. Is there any wayland support out there in the drivers btw?

nandra, 2019-09-07:
@slapin I have in my repos mali opensource driver + kernel driver (taken from bootlin repos). Can we maybe join forces and get it in? I tried to test with GLES and QT5 but I failed. Repo commits: https://github.com/nandra/meta-sunxi/commits/mbe-gles-mali Thx

slapin, 2019-09-07:
I hope so, does glmark2-es2 work on it?

slapin, 2019-09-07:
Anyway, touchscreen on my tablet comes first...

nandra, 2019-09-07:
Didn't try it yes. I'll try and share results. Thx.

slapin, 2019-09-07:
Well, I have had a look on blobs, are you sure they are better than what we have now? Ned to check core versions... I need to make myself a stable test platform first, from a13-mit tablet I have. Can we chat on IRC about this issue first? That would make lots of things go faster I think. We need to support fbdev, X11 and wayland and add proper settings for each one. We can start with X11-only build then extend to other two.

nandra, 2019-09-07:
Sure, on which IRC channel? Thx

-- 
as simple and primitive as possible
-------------------------------------------------
Marek Belisko - OPEN-NANDRA
Freelance Developer

Ruska Nova Ves 219 | Presov, 08005 Slovak Republic
Tel: +421 915 052 184
skype: marekwhite
twitter: #opennandra
web: http://open-nandra.com

slapin, 2019-09-07:
I think #linux-sunxi on freenode is official place unless something changed.

slapin, 2019-09-10:
Looks like we live in very different timezones and can't meet on IRC. As soon as I finally make firefox work on my image and do some wiki editing, I will look into how we can support Mali for different display systems. As I understand, blobs now support DMAbuf for x11, wayland and fbdev, which is great progress. Also we have lima in Mesa. And there is no way to choose blob/open source driver in OE which I found. So we need to prepare include files for each blob configuration and each mesa configuration I guess so that proper PREFERRED_PROVIDER entries are set. we have features for wayland and x11 which we can use so that we use proper package for each but also we need the same for mesa. For now I think we can use blobs by default and make include file for Mesa version which could be included from machine conf. Also I have lots of other topics to discuss, but them are not part of this issue anyway...

nandra, 2021-03-31:
Closed due inactivity. We can use now opensource lima + mesa driver for GPU acceleration.

closed 2021-03-31 by nandra