linux-sunxi/meta-sunxi #137: dtb file is missing from sdimage

chenbd, 2016-06-08:
test with cubietruck, core-image-base, build without any problem. but after flash the sdimage to sdcard and try to boot, the image can't boot, and seems the dtb file is missing from boot dir.
after manually copy uImage-sun7i-a20-cubietruck.dtb to boot dir and named sun7i-a20-cubietruck.dtb, the image can boot now.
i also checked core-image-base-cubietruck-20160607130700.rootfs.tar.gz and boot dir contains the dtb file

jlucius, 2016-06-08:
What version of yocto / oe are you using?

chenbd, 2016-06-09:
this is my configuration:
Build Configuration:
BB_VERSION        = "1.31.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubietruck"
DISTRO            = "poky"
DISTRO_VERSION    = "2.1+snapshot-20160609"
TUNE_FEATURES     = "arm armv7ve vfp  neon  vfpv4  callconvention-hard  cortexa7"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "master:4b4e1821fb1a00e16206b7e0bddc19de3e12e95e"
meta-oe           = "master:7be16bce816dee5e9ea1dd241b5a3543fdf78356"
meta-sunxi        = "master:35d98632f33b81e9e918a5033c72cae08b60697d"
meta-qt5          = "master:27617dde3560e2fe97989670725eb19747a0f48d"
meta-ruby         
meta-multimedia   = "master:7be16bce816dee5e9ea1dd241b5a3543fdf78356

ebutera, 2016-06-25:
i merged #141, i think it should be applied to master too right?

stephanroslen, 2016-06-27:
Assuming, that future versions of Yocto will be based on krogoth, that might be the right choice, I guess.

closed 2016-08-09 by chenbd