linux-sunxi/meta-sunxi #138: sunxi-mali: --disable-static: not found

chenbd, 2016-06-09:
sunxi-mali configure error, the following is the error log:
DEBUG: Executing python function sysroot_cleansstate
DEBUG: Python function sysroot_cleansstate finished
DEBUG: Executing shell function do_configure
/home/chen/linux/linux-sunxi/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/sunxi-mali/git-r0/temp/run.do_configure.6832: 103: /home/chen/linux/linux-sunxi/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/sunxi-mali/git-r0/temp/run.do_configure.6832: --disable-static: not found
WARNING: exit code 127 from a shell command.
ERROR: Function failed: do_configure (log file is located at /home/chen/linux/linux-sunxi/yocto/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/sunxi-mali/git-r0/temp/log.do_configure.6832)

ebutera, 2016-06-09:
you should always provide information about your setup (oe/yocto version, meta-sunxi revision used etc...), anyway in this case updating to latest meta-sunxi should fix it (see 69f56d3)

chenbd, 2016-06-09:
ok, i will provide information next time when fire new issues.

yes, with meta-sunxi        = "master:69f56d3f94cf79a6a2a38f1c342b4618f3a4cd02"
the issue is fixed.

closed 2016-06-09 by chenbd