linux-sunxi/meta-sunxi #144: MACHINE configs set DISTRO policy

koenkooi, 2016-09-09:
The machines in meta-sunxi include https://github.com/linux-sunxi/meta-sunxi/blob/master/conf/machine/include/sunxi-mali.inc which sets DISTRO policy. 

Even when ignoring that, an OE build should never install a big blob of binary without explicitly telling the user, especially binaries where redistribution wasn't allowed in past versions.

What I'd like to propose is:

1.  Remove the include from the MACHINE.configs
2. Tag the machines with a mali gpu with 'mali<ID>' in MACHINE_FEATURES, e.g. 'mali200'
3. Add something like https://github.com/96boards/meta-rpb/blob/master/conf/distro/include/mali.inc to meta-sunxi, located in conf/distro/include
4. Describe in the README how to include said .inc and what the license terms are

This would be a big step into making mali behave the same between BSPs. If not that, it makes this BSP more well behaved for not stepping on DISTRO settings.

ebutera, 2016-09-12:
I agree what you propose is much better then the current situation.

If possibile a warning could be displayed if mali is in machine_features but MALI_USERLAND_LIBARIES is empty, so if someone forgets to set it then he gets a feedback and if a distro doesn't want mali binaries can set it to mesa.

koenkooi, 2016-09-12:
Sure, I'll try to adapt the stuff from work to meta-sunxi, hopefully next week.

nandra, 2019-05-28:
@koenkooi any upodate on this? Thanks.

koenkooi, 2019-05-28:
Yes! I think this issue is moot now that the opensource drivers (lima/panfrost) are both in the mainline kernel and in upstream mesa.

nandra, 2019-05-28:
OK so we can maybe create recipes for that and drop sunxi-mali or make it optional? Thx.

koenkooi, 2019-05-28:
Mesa is in upstream OE-core, so only the kernel would need to get updated to 5.1+

nandra, 2019-05-28:
@koenkooi ok just to get clean picture mesa in OE-core contains support for (lima/panfrost - are they configurable via PACKAGECONFIG?) and then we need appropriate drivers  from later kernel. Thx.

koenkooi, 2019-05-28:
I just checked and OE-core only has mesa 19.0, lima/panfrost were merged in 19.1 :( I expect that they will have their own packageconfig after OE updates it to 19.1: https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-graphics/mesa/mesa.inc#L94

And lima/panfrost are in linux 5.2, not 5.1.

nandra, 2019-05-28:
@koenkooi ok thanks. I plan to push warrior branch and then hopefuly in next release we will get it for free so we can try to play with that. Thanks a lot.

slapin, 2019-09-07:
@koenkooi as time passed and I was out of context of a while, do you see any issues with distro features still? Also except for mali there are other gles binary blob drivers out there, so the goog order with things is still relevant.

koenkooi, 2019-09-08:
Using a DISTRO_FEATURE to enable the blob would be a good idea, the default should be the open source mesa driver.
OE-Core has mesa 19.1 nowadays.

nandra, 2020-05-26:
Closing as we have lima support in place already,