linux-sunxi/meta-sunxi #151: Function failed: do_image_sunxi_sdimg

xpete, 2016-10-11:
I keep getting the error below. The sunxi-sdimg is not even created. It works without the meta-sunxi.

>  bitbake core-image-base
> WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
> Parsing recipes: 100% |######################################################################################################################################################################| Time: 00:00:24
> Parsing of 879 .bb files complete (0 cached, 879 parsed). 1311 targets, 67 skipped, 0 masked, 0 errors.
> NOTE: Resolving any missing task queue dependencies
> 
> Build Configuration:
> BB_VERSION        = "1.30.0"
> BUILD_SYS         = "x86_64-linux"
> NATIVELSBSTRING   = "Ubuntu-16.04"
> TARGET_SYS        = "arm-poky-linux-gnueabi"
> MACHINE           = "olinuxino-a20"
> DISTRO            = "poky"
> DISTRO_VERSION    = "2.1.1"
> TUNE_FEATURES     = "arm armv7ve vfp  neon  vfpv4  callconvention-hard  cortexa7"
> TARGET_FPU        = "hard"
> meta              
> meta-poky         
> meta-yocto-bsp    = "krogoth:e93596fe74927e2e2f4dd7f671994ccb9744cff8"
> meta-sunxi        = "master:4ab93061c2f0e22cc91cfd22e31200de92b3eff5"
> 
> NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/1.0.1/x86_64-nativesdk-libc.tar.bz2;sha256sum=acf1e44a0ac2e855e81da6426197d36358bf7b4e88e552ef933128498c8910f8
> NOTE: Preparing RunQueue
> NOTE: Executing SetScene Tasks
> NOTE: Executing RunQueue Tasks
> ERROR: core-image-base-1.0-r0 do_image_sunxi_sdimg: Function failed: do_image_sunxi_sdimg (log file is located at /home/user/yocto/poky/build/tmp/work/olinuxino_a20-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_image_sunxi_sdimg.6889)
> ERROR: Logfile of failure stored in: /home/user/yocto/poky/build/tmp/work/olinuxino_a20-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_image_sunxi_sdimg.6889
> Log data follows:
> | DEBUG: Executing python function set_image_size
> | DEBUG: Python function set_image_size finished
> | DEBUG: Executing shell function do_image_sunxi_sdimg
> | dd: failed to open '/core-image-base-olinuxino-a20-20161011145937.rootfs.sunxi-sdimg': Permission denied
> | WARNING: exit code 1 from a shell command.
> | ERROR: Function failed: do_image_sunxi_sdimg (log file is located at /home/user/yocto/poky/build/tmp/work/olinuxino_a20-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_image_sunxi_sdimg.6889)
> ERROR: Task 17 (/home/user/yocto/poky/meta/recipes-core/images/core-image-base.bb, do_image_sunxi_sdimg) failed with exit code '1'
> NOTE: Tasks Summary: Attempted 2504 tasks of which 8 didn't need to be rerun and 1 failed.
> Waiting for 0 running tasks to finish:
> 
> Summary: 1 task failed:
>   /home/user/yocto/poky/meta/recipes-core/images/core-image-base.bb, do_image_sunxi_sdimg
> Summary: There was 1 WARNING message shown.
> Summary: There was 1 ERROR message shown, returning a non-zero exit code.
>

xpete, 2016-10-12:
I've found the problem. IMGDEPLOYDIR is not defined. It is trying to write the image to the root of the file system. I had to define IMGDEPLOYDIR.

twoerner, 2016-10-12:
It's usually a bad idea to mix layers from different branches. In your case you're using the "krogoth" branch of poky with the "master" branch of meta-sunxi. You should either use the krogoth branch or the master branch of both.

xpete, 2016-10-14:
Thank you. That was the problem.

closed 2016-10-14 by xpete

JOE-JOE-NGIGI, 2017-01-17:
How did you define IMGDEPLOYDIR please?