linux-sunxi/meta-sunxi #160: dd: failed to open '/core-image-minimal-cubieboard2-20170117143941.rootfs.sunxi-sdimg': Permission denied

JOE-JOE-NGIGI, 2017-01-17:
I am building an image for the cubieboard2 and it has not been successful so far. The do rootfs function fails and the dd if="" command fails in the sunxi-sdimg script.Here is my console output.

ngigijoe@ngigijoe-HP-2000-Notebook-PC:~/src/git/poky/build$ bitbake  core-image-minimal
WARNING: Host distribution "Ubuntu-16.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |###############################################################################################################################| ETA:  00:00:00
Loaded 1346 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.3"
TUNE_FEATURES     = "arm armv7a vfp neon"
TARGET_FPU        = "vfp-neon"
meta              
meta-yocto        
meta-yocto-bsp    = "jethro:e08994ce953ff5ab49b260e7763a40bfad4285ca"
meta-sunxi        = "master:7f88e6c78e7014960654c8f4b3bb0c4e188f0f43"

NOTE: Preparing RunQueue
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: Error: The image creation script '/home/ngigijoe/src/git/poky/build/tmp/work/cubieboard2-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/create_image.sunxi-sdimg' returned 1:
dd: failed to open '/core-image-minimal-cubieboard2-20170117143941.rootfs.sunxi-sdimg': Permission denied
WARNING: exit code 1 from a shell command.

ERROR: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/ngigijoe/src/git/poky/build/tmp/work/cubieboard2-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs.10750
ERROR: Task 7 (/home/ngigijoe/src/git/poky/meta/recipes-core/images/core-image-minimal.bb, do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2119 tasks of which 2118 didn't need to be rerun and 1 failed.
No currently running tasks (2118 of 2123)

Summary: 1 task failed:
  /home/ngigijoe/src/git/poky/meta/recipes-core/images/core-image-minimal.bb, do_rootfs
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

ebutera, 2017-01-17:
As you already noticed in #151 you should use the meta-sunxi "jethro" branch, not master.

JOE-JOE-NGIGI, 2017-01-18:
well, I'm a yocto newbie.What exactly would you suggest I do at this point please?

ebutera, 2017-01-18:
you have 2 options:
1) update your poky version to latest release (morthy), and use meta-sunxi morty branch
2) keep your poky version and "downgrade" meta-sunxi to jethro branch (cd meta-sunxi; git checkout jethro)

if you have no reason to keep using an old poky version i suggest you 1)

JOE-JOE-NGIGI, 2017-01-18:
Thanks so much.

ciprianmanea, 2017-03-06:
I'm experiencing a similar issue, trying to build the "orange-pi-one": 

`bitbake core-image-minimal
Parsing recipes: 100% |###############################################################################| Time: 0:00:33
Parsing of 878 .bb files complete (0 cached, 878 parsed). 1334 targets, 76 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "orange-pi-one"
DISTRO            = "poky"
DISTRO_VERSION    = "2.2.1"
TUNE_FEATURES     = "arm armv7ve vfp  neon  vfpv4  callconvention-hard  cortexa7"
TARGET_FPU        = "hard"
meta              
meta-poky         
meta-yocto-bsp    = "morty:6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f"
meta-sunxi        = "morty:1848d697ad2c57eebaef7c9826fe9f0e9f55f091"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/1.4/x86_64-nativesdk-libc.tar.bz2;sha256sum=101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca
--2017-03-05 21:24:33--  http://downloads.yoctoproject.org/releases/uninative/1.4/x86_64-nativesdk-libc.tar.bz2
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)... 198.145.20.127
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|198.145.20.127|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2473216 (2.4M) [application/octet-stream]
Saving to: ‘/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/downloads/uninative/101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca/x86_64-nativesdk-libc.tar.bz2’

2017-03-05 21:24:35 (1.39 MB/s) - ‘/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/downloads/uninative/101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca/x86_64-nativesdk-libc.tar.bz2’ saved [2473216/2473216]

Initialising tasks: 100% |############################################################################| Time: 0:00:04
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: xz-native-5.2.2-r0 do_fetch: Failed to fetch URL http://tukaani.org/xz/xz-5.2.2.tar.gz, attempting MIRRORS if available
WARNING: libpng-native-1.6.24-r0 do_fetch: Failed to fetch URL http://distfiles.gentoo.org/distfiles/libpng-1.6.24.tar.xz, attempting MIRRORS if available
ERROR: core-image-minimal-1.0-r0 do_image_sunxi_sdimg: Function failed: do_image_sunxi_sdimg (log file is located at /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_image_sunxi_sdimg.32003)
ERROR: Logfile of failure stored in: /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_image_sunxi_sdimg.32003
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_sunxi_sdimg
| 0+0 records in
| 0+0 records out
| 0 bytes copied, 0.000109502 s, 0.0 kB/s
| Model:  (file)
| Disk /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/deploy-core-image-minimal-image-complete/core-image-minimal-orange-pi-one-20170305202432.rootfs.sunxi-sdimg: 35.7MB
| Sector size (logical/physical): 512B/512B
| Partition Table: msdos
| Disk Flags:
| 
| Number  Start   End     Size    Type     File system  Flags
|  1      2097kB  23.1MB  21.0MB  primary               boot, lba
|  2      23.1MB  33.6MB  10.5MB  primary
| 
| mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
| mkfs.fat 4.0 (2016-05-06)
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Copying image-version-info
| 10+0 records in
| 10+0 records out
| 20971520 bytes (21 MB, 20 MiB) copied, 0.0154594 s, 1.4 GB/s
| 0+1 records in
| 0+1 records out
| 10485760 bytes (10 MB, 10 MiB) copied, 0.0117678 s, 891 MB/s
| dd: failed to open '/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/deploy/images/orange-pi-one/u-boot-sunxi-with-spl.bin': No such file or directory
| WARNING: /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/run.do_image_sunxi_sdimg.32003:1 exit 1 from 'dd if=/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/deploy/images/orange-pi-one/u-boot-sunxi-with-spl.bin of=/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/deploy-core-image-minimal-image-complete/core-image-minimal-orange-pi-one-20170305202432.rootfs.sunxi-sdimg bs=1024 seek=8 conv=notrunc'
[... cut ...]
ERROR: Task (/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/meta/recipes-core/images/core-image-minimal.bb:do_image_sunxi_sdimg) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1838 tasks of which 6 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/meta/recipes-core/images/core-image-minimal.bb:do_image_sunxi_sdimg
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
`

Looks like the u-boot-sunxi-with-spl.bin was not installed in the expected location.
Anyone experienced this before ? 

Thanks

closed 2017-04-12 by JOE-JOE-NGIGI