linux-sunxi/meta-sunxi #161: Keyboard and Mouse not working after building core-image-sato for cubieboard2

JOE-JOE-NGIGI, 2017-01-20:
I successfully built core-image-sato for cubieboard2. It boots perfectly, the Xorg server even launches the desktop. The only problem is I can't get the mouse and keyboard to work. Any help is highly welcome. I have been using the jethro branch.Below is the output I get from:
**/var/log/Xorg.0.log**

X.Org X Server 1.17.2
Release Date: 2015-06-16
[3132137.737] X Protocol Version 11, Revision 0
[3132137.737] Build Operating System: Linux 4.4.0-59-generic x86_64
[3132137.737] Current Operating System: Linux cubieboard2 4.1.15 #1 SMP Thu Jan
[3132137.737] Kernel command line: console=ttyS0,115200 console=tty1 root=/dev/m
[3132137.737] Build Date: 19 January 2017  07:33:48PM
[3132137.737]
[3132137.737] Current version of pixman: 0.32.6
[3132137.737]   Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[3132137.737] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[3132137.738] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Jan 20 07:04:14 20
[3132137.778] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[3132137.798] (==) No Layout section.  Using the first Screen section.
[3132137.798] (==) No screen section available. Using defaults.
[3132137.798] (**) |-->Screen "Default Screen Section" (0)
[3132137.798] (**) |   |-->Monitor "<default monitor>"
[3132137.799] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[3132137.799] (**) |   |-->Device "Allwinner A10/A13 FBDEV"                     
[3132137.799] (==) No monitor specified for screen "Default Screen Section".    
        Using a default monitor configuration.                                  
[3132137.799] (==) Automatically adding devices                                 
[3132137.799] (==) Automatically enabling devices                               
[3132137.799] (==) Automatically adding GPU devices                             
[3132137.802] (WW) The directory "/usr/share/fonts/X11/misc/" does not exist.   
[3132137.802]   Entry deleted from font path.                                   
[3132137.802] (WW) The directory "/usr/share/fonts/X11/TTF/" does not exist.    
[3132137.803]   Entry deleted from font path.                                   
[3132137.803] (WW) The directory "/usr/share/fonts/X11/OTF/" does not exist.    
[3132137.803]   Entry deleted from font path.                                   
[3132137.803] (WW) The directory "/usr/share/fonts/X11/Type1/" does not exist.  
[3132137.803]   Entry deleted from font path.                                   
[3132137.803] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist. 
[3132137.803]   Entry deleted from font path.                                  
[3132137.803] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist. 
[3132137.803]   Entry deleted from font path.                                  
[3132137.803] (==) FontPath set to:                                            
                                                                               
[3132137.803] (==) ModulePath set to "/usr/lib/xorg/modules"                   
[3132137.803] (II) The server relies on udev to provide the list of input device
[3132137.803] (==) ModulePath set to "/usr/lib/xorg/modules"                    
[3132137.803] (II) The server relies on udev to provide the list of input device
        If no devices become available, reconfigure udev or disable AutoAddDevic
[3132137.803] (II) Loader magic: 0x1bfa80                                       
[3132137.803] (II) Module ABI versions:                                         
[3132137.803]   X.Org ANSI C Emulation: 0.4                                     
[3132137.803]   X.Org Video Driver: 19.0                                        
[3132137.803]   X.Org XInput driver : 21.0                                      
[3132137.803]   X.Org Server Extension : 9.0                                    
[3132137.804] (II) no primary bus or device found                               
[3132137.806] (II) LoadModule: "glx"                                            
[3132137.813] (WW) Warning, couldn't open module glx                            
[3132137.813] (II) UnloadModule: "glx"                                          
[3132137.813] (II) Unloading glx                                                
[3132137.813] (EE) Failed to load module "glx" (module does not exist, 0)       
[3132137.813] (II) LoadModule: "fbturbo"                                        
[3132137.813] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so         
[3132137.823] (II) Module fbturbo: vendor="X.Org Foundation"                    
[3132137.823]   compiled for 1.17.2, module version = 0.5.1                     
[3132137.823]   Module class: X.Org Video Driver                                
[3132137.823]   ABI class: X.Org Video Driver, version 19.0                     
[3132137.823] (II) FBTURBO: driver for framebuffer: fbturbo  
[3132137.824] (--) using VT number 3                                            
[... 32 lines cut ...]
[3132137.938] (==) Depth 24 pixmap format is 32 bpp                             
[3132137.939] (EE) FBTURBO(0): FBIOBLANK: Invalid argument                      
[3132137.960] (II) FBTURBO(0): using backing store heuristics                   
[3132137.967] (II) FBTURBO(0): can't load 'g2d_23' kernel module                
[3132137.967] (II) FBTURBO(0): failed to enable the use of sunxi display control
[3132137.967] (II) FBTURBO(0): No sunxi-g2d hardware detected (check /dev/disp a
[3132137.968] (II) FBTURBO(0): G2D hardware acceleration can't be enabled       
[3132137.968] (II) FBTURBO(0): enabled VFP/NEON optimizations                   
[3132137.968] (==) FBTURBO(0): Backing store enabled                            
[3132137.969] (==) FBTURBO(0): DPMS enabled                                     
[3132137.969] (II) FBTURBO(0): failed to enable hardware cursor                 
[3132137.984] (II) FBTURBO(0): can't load 'mali' kernel module                  
[3132137.992] (II) FBTURBO(0): can't load 'mali_drm' kernel module        
[3132137.992] (II) FBTURBO(0): can't load 'mali_drm' kernel module              
[3132138.000] (II) FBTURBO(0): can't load 'sunxi_cedar_mod' kernel module       
[3132138.000] (II) Loading sub module "dri2"                                    
[3132138.000] (II) LoadModule: "dri2"                                           
[3132138.000] (II) Module "dri2" already built-in                               
[3132138.399] SunxiMaliDRI2_Init: drmOpen failed!                               
[3132138.400] (II) FBTURBO(0): failed to enable DRI2 integration for Mali GPU   
[3132138.400] (==) RandR enabled                                                
[3132138.759] (EE) FBTURBO(0): FBIOBLANK: Invalid argument                      
[3132738.774] (EE) FBTURBO(0): FBIOBLANK: Invalid argument                      
[3132738.774] (EE) FBTURBO(0): FBIOBLANK: Invalid argument                      
[3132738.807] (EE) FBTURBO(0): FBIOBLANK: Invalid argument

yangjiel, 2017-11-02:
I have the same problem that FontPath set to: empty, such that startx won't work because lack of libraries or modules.

nandra, 2019-05-28:
@JOE-JOE-NGIGI any update on this? Thanks.

JOE-JOE-NGIGI, 2019-05-28:
@Marek Belisko,

You will have to do enable those in the kernel.
It should be easy, just find out the bitbake command for accessing the
kernel menuconfig.
Then navigate to kernel/drivers/input and enable both mouse and keyboard.
Also check kernel/drivers/usb/input and enable usb mouse and keyboard.
I did this a while ago I may not be correct on the directories but that
fixed it for me and I am currently able to use those inputs on my device
that runs
core-image-sato.

Regards,

nandra, 2019-05-28:
OK so it was only missing kernel configurations. Did you check on sunxi latest branch? Thanks.

JOE-JOE-NGIGI, 2019-05-28:
I haven't checked because our project made some changes and we had to move
from linux-sunxi kernel to mainline kernel.

nandra, 2019-05-28:
OK I'll close it for now. If it appear again pls re-open.

closed 2019-05-28 by nandra