linux-sunxi/meta-sunxi #168: orange-pi-one build fails on morty

ciprianmanea, 2017-03-08:
I'm experiencing a similar issue, trying to build the "orange-pi-one":

bitbake core-image-minimal
Parsing recipes: 100% |###############################################################################| Time: 0:00:33
Parsing of 878 .bb files complete (0 cached, 878 parsed). 1334 targets, 76 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION = "1.32.0"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "Ubuntu-16.04"
TARGET_SYS = "arm-poky-linux-gnueabi"
MACHINE = "orange-pi-one"
DISTRO = "poky"
DISTRO_VERSION = "2.2.1"
TUNE_FEATURES = "arm armv7ve vfp neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU = "hard"
meta
meta-poky
meta-yocto-bsp = "morty:6a1f33cc40bfac33cf030fe41e1a8efd1e5fad6f"
meta-sunxi = "morty:1848d697ad2c57eebaef7c9826fe9f0e9f55f091"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/1.4/x86_64-nativesdk-libc.tar.bz2;sha256sum=101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca
--2017-03-05 21:24:33-- http://downloads.yoctoproject.org/releases/uninative/1.4/x86_64-nativesdk-libc.tar.bz2
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)... 198.145.20.127
Connecting to downloads.yoctoproject.org (downloads.yoctoproject.org)|198.145.20.127|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2473216 (2.4M) [application/octet-stream]
Saving to: ‘/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/downloads/uninative/101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca/x86_64-nativesdk-libc.tar.bz2’

2017-03-05 21:24:35 (1.39 MB/s) - ‘/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/downloads/uninative/101ff8f2580c193488db9e76f9646fb6ed38b65fb76f403acb0e2178ce7127ca/x86_64-nativesdk-libc.tar.bz2’ saved [2473216/2473216]

Initialising tasks: 100% |############################################################################| Time: 0:00:04
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: xz-native-5.2.2-r0 do_fetch: Failed to fetch URL http://tukaani.org/xz/xz-5.2.2.tar.gz, attempting MIRRORS if available
WARNING: libpng-native-1.6.24-r0 do_fetch: Failed to fetch URL http://distfiles.gentoo.org/distfiles/libpng-1.6.24.tar.xz, attempting MIRRORS if available
ERROR: core-image-minimal-1.0-r0 do_image_sunxi_sdimg: Function failed: do_image_sunxi_sdimg (log file is located at /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_image_sunxi_sdimg.32003)
ERROR: Logfile of failure stored in: /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_image_sunxi_sdimg.32003
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: Python function set_image_size finished
| DEBUG: Executing shell function do_image_sunxi_sdimg
| 0+0 records in
| 0+0 records out
| 0 bytes copied, 0.000109502 s, 0.0 kB/s
| Model: (file)
| Disk /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/deploy-core-image-minimal-image-complete/core-image-minimal-orange-pi-one-20170305202432.rootfs.sunxi-sdimg: 35.7MB
| Sector size (logical/physical): 512B/512B
| Partition Table: msdos
| Disk Flags:
|
| Number Start End Size Type File system Flags
| 1 2097kB 23.1MB 21.0MB primary boot, lba
| 2 23.1MB 33.6MB 10.5MB primary
|
| mkfs.fat: warning - lowercase labels might not work properly with DOS or Windows
| mkfs.fat 4.0 (2016-05-06)
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Total number of sectors (40960) not a multiple of sectors per track (63)!
| Add mtools_skip_check=1 to your .mtoolsrc file to skip this test
| Copying image-version-info
| 10+0 records in
| 10+0 records out
| 20971520 bytes (21 MB, 20 MiB) copied, 0.0154594 s, 1.4 GB/s
| 0+1 records in
| 0+1 records out
| 10485760 bytes (10 MB, 10 MiB) copied, 0.0117678 s, 891 MB/s
| dd: failed to open '/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/deploy/images/orange-pi-one/u-boot-sunxi-with-spl.bin': No such file or directory
| WARNING: /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/run.do_image_sunxi_sdimg.32003:1 exit 1 from 'dd if=/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/deploy/images/orange-pi-one/u-boot-sunxi-with-spl.bin of=/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/deploy-core-image-minimal-image-complete/core-image-minimal-orange-pi-one-20170305202432.rootfs.sunxi-sdimg bs=1024 seek=8 conv=notrunc'
| ERROR: Function failed: do_image_sunxi_sdimg (log file is located at /home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/build-sunxi-orange-pi-one/tmp/work/orange_pi_one-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_image_sunxi_sdimg.32003)
ERROR: Task (/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/meta/recipes-core/images/core-image-minimal.bb:do_image_sunxi_sdimg) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1838 tasks of which 6 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
/home/ciprian/work/f-elec/trainig/yocto/yocto-labs/poky-morty/meta/recipes-core/images/core-image-minimal.bb:do_image_sunxi_sdimg
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Looks like the u-boot-sunxi-with-spl.bin was not installed in the expected location.
Anyone experienced this before ?

Thanks

ebutera, 2017-04-06:
should be fixed with 941171c912f468edb5deddcf92a04cee99a2b0dd

ciprianmanea, 2017-06-12:
Tested and works fine now with this 941171c change.

closed 2017-06-12 by ciprianmanea