linux-sunxi/meta-sunxi #174: recipes-devtools/rpm/rpm_5.4.16.bb:do_compile) failed with exit code '1

JOE-JOE-NGIGI, 2017-04-12:
I can't build the recipes-devtools/rpm in poky-morty because apparently the do_compile function in rpm_5.4.16.bb fails. The do_compile function does not even exist in the .bb file.Anybody knows a fix to this problem?

nandra, 2019-05-28:
@JOE-JOE-NGIGI do you still experience an issue? Thanks.

JOE-JOE-NGIGI, 2019-05-28:
Hey Marek,

I figured it out.

At the time I had lots of dependencies I had not installed and compiled and
my gcc and g++ versions were different.

That was causing most of my build errors but I have since solved that and
compiled my images a couple of times successfully.

My image is already booting on my device.
I have done a couple of cool stuff on it.

I enabled the GUI log in using lxdm. I am yet to figure out how to launch a
virtual keyboard on the log in screen.

nandra, 2019-05-28:
OK fine. Thanks for update. Closing...

closed 2019-05-28 by nandra