linux-sunxi/meta-sunxi #177: error  rpi-basic-image (sunxi-mali dependencies) Toaster branch Morty 2.2.1

denyste, 2017-04-24:
Hi,

I am creating and building an embedded Linux distribution for the raspberry pi3.
I use the version of Yocto Project 2.2.1 - Morty and the layer meta-sunxi.
I am integrating the xf86-video-fbturbo (meta-sunxi recipes) into the image rpi-basic-image.
I am not able to solve the problems of dependencies between the recipes (sunxi-mali) of this image recipe (see the Error message below):
"1  error: Required build target 'rpi-basic-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['rpi-basic-image', 'sunxi-mali']"

Could you help me to resolve the sunxi-mali dependencies. I have wrote this diagram of the dependencies for xf86-video-fbturbo originating from the "OpenEmbedded Layer Index" below :
morty → meta-sunxi → xf86-video-fbturbo (MIT-X) DEPENDS += "sunxi-mali libump xf86driproto"
 morty → meta-sunxi → sunxi-mali (Proprietary) DEPENDS = "libdrm dri2proto libump"
  morty → openembedded-core → libdrm (MIT) DEPENDS = "libpthread-stubs udev libpciaccess"
   morty → openembedded-core → libpthread-stubs  (MIT)
   morty → openembedded-core → eudev  (GPLv2.0+)
   morty → openembedded-core → libpciaccess (MIT & MIT-style)
 morty → openembedded-core → dri2proto (MIT)
morty → meta-sunxi → libump (Apache-2)
morty → openembedded-core → xf86driproto (MIT)

See also the copy of the IMAGE_INSTALL_append (Toaster Morty 2.2.1) below: 
"apt dpkg dbus sudo tzdata glibc-utils localedef networkmanager pointercal xinit libmatchbox matchbox-terminal matchbox-desktop matchbox-keyboard xkeyboard-config matchbox-session matchbox-wm libx11 libxext consolekit base-passwd liberation-fonts pkgconfig xserver-xorg xf86-video-fbdev libpthread-stubs udev libpciaccess dri2proto renderproto videoproto xextproto fontsproto xextproto sunxi-mali libump libdrm xf86driproto xf86-video-fbturbo libxext libxfixes xf86-input-libinput" 
I thank you in advance for your reply. 

Denis

ebutera, 2017-04-25:
i never used fbturbo but you can try to replace the sunxi-mali and libump dependencies with rpi3 equivalent (vc-graphics?).

denyste, 2017-04-27:
Thanks Enrico!
I am testing the recipe vc-graphics but there is a lot problems with dependencies. An issue has been opened on the site of the maintainer of meta-rasperrypi, see the link below.    
https://github.com/agherzan/meta-raspberrypi/issues/58

nandra, 2019-05-28:
@denyste any update on this? I saw that agherzan/meta-raspberrypi#58 was closed due inactivity.

denyste, 2019-05-29:
Hi Marek Belisko,

Sorry, but there is no update.
I did not use xf86-video-fbturbo, I used another driver less complicated (I cannot remember which).
But, you will be able to find it easily.
Howewer, the recipe xf86-video-fbturbo informations could be available are listed here, see below:
https://layers.openembedded.org/layerindex/recipe/23093/.  

Good luck.

denyste

----- Mail original -----

closed 2019-08-07 by nandra