linux-sunxi/meta-sunxi #178: Dependency loop when using initramfs

agherzan, 2017-04-24:
The current implementation inherits sdcard_image-sunxi.bbclass unconditionally which in turn, sets two depends on do_rootfs unconditionally again. 

```
do_rootfs[depends] += "virtual/kernel:do_deploy"                                                                                                                         
do_rootfs[depends] += "virtual/bootloader:do_deploy" 
```

This means that if I want to build an initramfs image in the kernel image I will get a dependency loop.

Can we change that by including these in IMAGE_DEPENDS_sunxi-sdimg?

```
IMAGE_DEPENDS_sunxi-sdimg += " \
                        parted-native \
                        mtools-native \
                        dosfstools-native \
                        virtual/kernel:do_deploy \
                        virtual/bootloader:do_deploy \
                        "
```

Any concerns against this change? Can we have it in morty too?

ebutera, 2017-04-25:
yes it looks good to me, did you already test it?

agherzan, 2017-04-26:
Haven't tested it in this specific case but we are using something similar in meta-raspberrrypi for a while without any issues.

ebutera, 2017-04-26:
i've pushed the fix on master, i was running a test build on morty but it's taking forever so i'll push it later or if someone confirms it works.

ebutera, 2017-04-27:
cherry-picked on morty too

closed 2017-04-27 by ebutera

agherzan, 2017-04-27:
Thanks @ebutera . @shaunmulligan I know you've been testing this, could you please give the fix a try?