linux-sunxi/meta-sunxi #183: Legacy kernel build

sergey-suloev, 2017-06-06:
Hello,
can you explain what should be changed in  conf/machine/include/sunxi.inc to  build legacy kernel ?
Can you update this section in README file, please ?

Thank you

ebutera, 2017-06-06:
it's already there:

> PREFERRED_PROVIDER_u-boot="u-boot-sunxi"
> PREFERRED_PROVIDER_virtual/bootloader="u-boot-sunxi"
> PREFERRED_PROVIDER_virtual/kernel="linux-sunxi"

in other words: you change sunxi.inc to that, OR add it to local.conf

sergey-suloev, 2017-06-06:
Thanks a lot!
I tried that but I got errors compiling u-boot:

--------------------------------------------------------------------------------------------------
bitbake u-boot-sunxi
Loading cache: 100% |######################################################################################| Time: 0:00:01
Loaded 1333 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.32.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-17.04"
TARGET_SYS        = "arm-oe-linux-gnueabi"
MACHINE           = "orange-pi-one"
DISTRO            = "nodistro"
DISTRO_VERSION    = "nodistro.0"
TUNE_FEATURES     = "arm armv7ve vfp  neon  vfpv4  callconvention-hard  cortexa7"
TARGET_FPU        = "hard"
meta-sunxi        = "master:a84d6a40c1c786c64824e8e13b289fccc0b32b92"
meta              = "morty:b206497bf2a68b1b009ae5a6b1c5a29df3966443"

Initialising tasks: 100% |#################################################################################| Time: 0:00:01
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: u-boot-sunxi-1_v2014.04+gitAUTOINC+ea1ac32bf7-r0 do_compile: oe_runmake failed
ERROR: u-boot-sunxi-1_v2014.04+gitAUTOINC+ea1ac32bf7-r0 do_compile: Function failed: do_compile (log file is located at /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.3323)
ERROR: Logfile of failure stored in: /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.3323
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 2 CROSS_COMPILE=arm-oe-linux-gnueabi- CC=arm-oe-linux-gnueabi-gcc  --sysroot=/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/orange-pi-one V=1 HOSTCC=gcc  -isystem/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/usr/include -O2 -pipe -L/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/usr/lib -L/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/lib -Wl,-rpath,/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/sysroots/x86_64-linux/lib -Wl,-O1 -C /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git O=/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/build orangepi_one_defconfig
| ERROR: oe_runmake failed
| make: Entering directory '/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git'
| make -C /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/build \
| KBUILD_SRC=/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git \
| KBUILD_EXTMOD="" -f /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git/Makefile \
| orangepi_one_defconfig
| make[1]: *** No rule to make target 'orangepi_one_defconfig'.  Stop.
| Makefile:138: recipe for target 'sub-make' failed
| make: *** [sub-make] Error 2
| make: Leaving directory '/home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git'
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/sergey/Projects/raspberry/openembedded/oe-core/orangepione/tmp-glibc/work/orange_pi_one-oe-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/temp/log.do_compile.3323)
ERROR: Task (/home/sergey/Projects/raspberry/openembedded/oe-core/meta-sunxi/recipes-bsp/u-boot/u-boot-sunxi.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 393 tasks of which 387 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/sergey/Projects/raspberry/openembedded/oe-core/meta-sunxi/recipes-bsp/u-boot/u-boot-sunxi.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

ebutera, 2017-06-06:
> make[1]: *** No rule to make target 'orangepi_one_defconfig'. Stop.

looks like that u-boot-sunxi version doesn't support orangepi_one

closed 2017-12-21 by ebutera