linux-sunxi/meta-sunxi #185: Nothing RPROVIDES 'libegl' (but /poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.19.1.bb RDEPENDS on or otherwise requires it)

JOE-JOE-NGIGI, 2017-07-17:
I am on pyro branch, building a core-image-sato for Allwinner's cubieboard2.My meta layers include the following:-
1.meta
2.meta-poky
3.meta-sunxi
4.meta-openembedded/meta-oe
5.meta-openembedded/meta-gnome

However, my bitbake does not go beyond the following:

Loading cache: 100% |########################################################################################################################################################| Time: 0:00:03
Loaded 2168 entries from dependency cache.
Parsing recipes: 100% |######################################################################################################################################################| Time: 0:00:00
Parsing of 1567 .bb files complete (1566 cached, 1 parsed). 2168 targets, 201 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'libegl' (but /home/ngigijoe/src/git/pyro/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.19.1.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'libegl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libegl']
NOTE: Runtime target 'xf86-input-keyboard-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-keyboard-dev', 'virtual/xserver', 'libegl']
NOTE: Runtime target 'xf86-input-keyboard' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-keyboard', 'virtual/xserver', 'libegl']
NOTE: Runtime target 'packagegroup-core-x11-xserver' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-xserver', 'xf86-input-keyboard', 'virtual/xserver', 'libegl']
NOTE: Runtime target 'packagegroup-core-x11-base' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-base', 'packagegroup-core-x11-xserver', 'xf86-input-keyboard', 'virtual/xserver', 'libegl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-base', 'packagegroup-core-x11-xserver', 'xf86-input-keyboard', 'virtual/xserver', 'libegl']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
What could be a possible fix?

twoerner, 2017-07-17:
Could you try adding the following to your conf/local.conf:

    PACKAGECONFIG_remove_pn-xserver-xorg = "glamor"

    PREFERRED_PROVIDER_u-boot = "u-boot-sunxi"
    PREFERRED_PROVIDER_virtual/bootloader = "u-boot-sunxi"
    PREFERRED_PROVIDER_virtual/kernel = "linux-sunxi"

    IMAGE_FEATURES += "x11"
    DISTRO_FEATURES_append = " opengl x11"
    DISTRO_FEATURES_remove = "wayland"

?

I don't have a cubieboard2, but I do have a cubietruck (i.e. cubieboard3) and the above builds core-image-sato for me. As well, the resulting image works on my board.

I am able to perform a poky build of core-image-sato using MACHINE=cubieboard2 with the above, and it builds for me. Without a board, however, I can't test the resulting image.

Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "opensuse-42.2"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3.1"
TUNE_FEATURES     = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU        = "hard"
meta              
meta-poky         = "pyro:6bd890d9e011014cf323e61267f8b256949d44aa"
meta-sunxi        = "pyro:0056643fcf2c496a0f2cf005fb67d626ab0e2c10"
meta-oe           
meta-gnome        = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"

JOE-JOE-NGIGI, 2017-07-18:
Thanks Trevor. Let me try it out and then get back to you.

JOE-JOE-NGIGI, 2017-07-18:
I tried that and now it throws some do_compile errors. I am thinking it
could be an issue with gcc within the host system but I don't have a
workaround for it. Here's my build output now:

ngigijoe@ngigijoe-HP-2000-Notebook-PC:/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2$
bitbake core-image-sato
Parsing recipes: 100%
|#################################################################################################################|
Time: 0:02:38
Parsing of 1567 .bb files complete (0 cached, 1567 parsed). 2168 targets,
206 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3.1"
TUNE_FEATURES     = "arm armv7ve vfp neon vfpv4 callconvention-hard
cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky         = "pyro:6bd890d9e011014cf323e61267f8b256949d44aa"
meta-oe
meta-gnome        = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"
meta-sunxi        = "pyro:0056643fcf2c496a0f2cf005fb67d626ab0e2c10"

NOTE: Fetching uninative binary shim from
http://downloads.yoctoproject.org/releases/uninative/1.6/x86_64-nativesdk-libc.tar.bz2;sha256sum=2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a
--2017-07-18 12:19:18--
http://downloads.yoctoproject.org/releases/uninative/1.6/x86_64-nativesdk-libc.tar.bz2
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)...
198.145.20.127
Connecting to downloads.yoctoproject.org
(downloads.yoctoproject.org)|198.145.20.127|:80...
connected.
HTTP request sent, awaiting response... 200 OK
Length: 2535308 (2.4M) [application/octet-stream]
Saving to:
‘/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/downloads/uninative/2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a/x86_64-nativesdk-libc.tar.bz2’

2017-07-18 12:19:27 (305 KB/s) -
‘/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/downloads/uninative/2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a/x86_64-nativesdk-libc.tar.bz2’
saved [2535308/2535308]

Initialising tasks: 100%
|##############################################################################################################|
Time: 0:01:37
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: icu-native-58.2-r0 do_fetch: Failed to fetch URL
http://download.icu-project.org/files/icu4c/58.2/icu4c-58_2-src.tgz,
attempting MIRRORS if available
ERROR: nspr-native-4.13.1-r0 do_compile: Function failed: do_compile (log
file is located at
/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/nspr-native/4.13.1-r0/temp/log.do_compile.29184)
ERROR: mpfr-native-3.1.5-r0 do_unpack: Error executing a python function in
exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:extend_recipe_sysroot(d)
     0003:
File:
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/meta/classes/staging.bbclass',
lineno: 598, function: extend_recipe_sysroot
[... 904 lines cut ...]
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/build/tests/misc'
| Making all in cxx
| make[3]: Entering directory
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/build/tests/cxx'
| make[3]: Nothing to be done for 'all'.
| make[3]: Leaving directory
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/build/tests/cxx'
| make[2]: Leaving directory
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/build/tests'
| Making all in mpn
| make[2]: Entering directory
'/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/build/mpn'
| ../x86_64-linux-libtool  --tag=CC   --mode=compile gcc  -DHAVE_CONFIG_H
-I. -I../../gmp-6.1.2/mpn -I..  -D__GMP_WITHIN_GMP -I../../gmp-6.1.2
-DOPERATION_`echo fib_table | sed 's/_$//'`
-isystem/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/recipe-sysroot-native/usr/include
-O2 -pedantic -fomit-frame-pointer -m64 -mtune=k8 -march=k8
-isystem/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/recipe-sysroot-native/usr/include
-O2 -pipe -c -o fib_table.lo fib_table.c
| ../x86_64-linux-libtool  --tag=CC   --mode=compile gcc  -DHAVE_CONFIG_H
-I. -I../../gmp-6.1.2/mpn -I..  -D__GMP_WITHIN_GMP -I../../gmp-6.1.2
-DOPERATION_`echo mp_bases | sed 's/_$//'`
-isystem/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/work/x86_64-linux/gmp-native/6.1.2-r0/recipe-sysroot-native/usr/include
-O2 -pedantic -fomit-frame-pointer -m64 -mtune=k8 -march=k8
-isystem/media/ngigijoe/3202EFFC49995A0F/yocto_pyro/poky/cubieboard2/tmp/wor

twoerner, 2017-07-18:
The problem appears to be here:

| In file included from ../../../../nspr-4.13.1/nspr/pr/include/md/_unixos.h:152:0,
| from ../../../../nspr-4.13.1/nspr/pr/include/md/prosdep.h:93,
| from ../../../../nspr-4.13.1/nspr/pr/include/private/primpl.h:44,
| from ../../../../nspr-4.13.1/nspr/pr/src/misc/pripc.c:12:
| /usr/include/string.h:635:28: fatal error: ../../../../nspr-4.13.1/nspr/pr/include/private/bits/string3.h: Interrupted system call
| compilation terminated.

I've never seen this before (interrupted system call). Is your machine stable otherwise? Googling "gcc interrupted system call" doesn't give much help.

Maybe just try:

    $ bitbake -c cleansstate nspr-native

then try the build again?

There are lots of reasons for a generic interrupted system call... bad RAM, running out of RAM, running out of resources in general. Are you building in a VM, and/or using NFS disks? Have you done other successful Yocto/OE builds before?

JOE-JOE-NGIGI, 2017-07-18:
I had initially done a successful build with jethro branch. Only reason I
am doing a re-run is to include meta-openembedded/meta-gnome to allow
support for graphical log in.
As fate would have it, I already deleted that directory and now I seem to
be having lots of trouble.
Sure I am using an NTFS partition. That could also be a problem.

JOE-JOE-NGIGI, 2017-07-18:
I've switched to and ext4 partion and did a clean build maintaining the
same configurations. However I am still getting the do_compile errors. This
is weird.
ngigijoe@ngigijoe-HP-2000-Notebook-PC:~/src/git/pyro/poky/cubieboard2$
bitbake core-image-sato
Parsing recipes: 100% |##########################################| Time:
0:03:41
Parsing of 1567 .bb files complete (0 cached, 1567 parsed). 2168 targets,
206 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION        = "1.34.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "ubuntu-16.04"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "cubieboard2"
DISTRO            = "poky"
DISTRO_VERSION    = "2.3.1"
TUNE_FEATURES     = "arm armv7ve vfp neon vfpv4 callconvention-hard
cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky         = "pyro:6bd890d9e011014cf323e61267f8b256949d44aa"
meta-sunxi        = "pyro:0056643fcf2c496a0f2cf005fb67d626ab0e2c10"
meta-oe
meta-gnome        = "pyro:5e82995148a2844c6f483ae5ddd1438d87ea9fb7"

NOTE: Fetching uninative binary shim from
http://downloads.yoctoproject.org/releases/uninative/1.6/x86_64-nativesdk-libc.tar.bz2;sha256sum=2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a
--2017-07-18 17:44:23--
http://downloads.yoctoproject.org/releases/uninative/1.6/x86_64-nativesdk-libc.tar.bz2
Resolving downloads.yoctoproject.org (downloads.yoctoproject.org)...
198.145.20.127
Connecting to downloads.yoctoproject.org
(downloads.yoctoproject.org)|198.145.20.127|:80...
connected.
HTTP request sent, awaiting response... 200 OK
Length: 2535308 (2.4M) [application/octet-stream]
Saving to:
‘/home/ngigijoe/src/git/pyro/poky/cubieboard2/downloads/uninative/2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a/x86_64-nativesdk-libc.tar.bz2’

2017-07-18 17:44:32 (321 KB/s) -
‘/home/ngigijoe/src/git/pyro/poky/cubieboard2/downloads/uninative/2b4fffa308d9f19e0742a1a404ff42495fb50c165e5ca0458cedca157372691a/x86_64-nativesdk-libc.tar.bz2’
saved [2535308/2535308]

Initialising tasks: 100% |#######################################| Time:
0:00:51
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: gperf-native-3.0.4-r0 do_compile: oe_runmake failed
ERROR: gperf-native-3.0.4-r0 do_compile: Function failed: do_compile (log
file is located at
/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/temp/log.do_compile.17022)
ERROR: Logfile of failure stored in:
/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/temp/log.do_compile.17022
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8
| cd lib; make all
| make[1]: Entering directory
'/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/build/lib'
| gcc
-isystem/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/recipe-sysroot-native/usr/include
-O2 -pipe -I../../gperf-3.0.4/lib -c ../../gperf-3.0.4/lib/getopt.c
| gcc
-isystem/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/recipe-sysroot-native/usr/include
-O2 -pipe -I../../gperf-3.0.4/lib -c ../../gperf-3.0.4/lib/getopt1.c
| g++
-isystem/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/recipe-sysroot-native/usr/include
[... 321 lines cut ...]
delete[](void*)'
| getline.cc:(.text+0x156): undefined reference to `operator new[](unsigned
long)'
| collect2: error: ld returned 1 exit status
| Makefile:74: recipe for target 'gperf' failed
| ERROR: oe_runmake failed
| make[1]: *** [gperf] Error 1
| make[1]: Leaving directory
'/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/build/src'
| Makefile:33: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at
/home/ngigijoe/src/git/pyro/poky/cubieboard2/tmp/work/x86_64-linux/gperf-native/3.0.4-r0/temp/log.do_compile.17022)
ERROR: Task
(virtual:native:/home/ngigijoe/src/git/pyro/poky/meta/recipes-extended/gperf/gperf_3.0.4.bb:do_compile)
failed with exit code '1'
NOTE: Tasks Summary: Attempted 423 tasks of which 0 didn't need to be rerun
and 1 failed.

Summary: 1 task failed:

virtual:native:/home/ngigijoe/src/git/pyro/poky/meta/recipes-extended/gperf/gperf_3.0.4.bb:
do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

twoerner, 2017-07-18:
Yes, you are seeing a do_compile error, but it's in a completely different component (gperf-native this time instead of nspr-native) and for a completely different reason ("undefined reference" this time vs "interrupted system call" last time). I think these build issues are beyond "nothing provides libegl", and beyond the metadata in general. Sorry!

JOE-JOE-NGIGI, 2017-07-18:
Does it matter whether one uses a newer version of gcc on the host compared
to the target machine?I am suspecting that could be the problem.

twoerner, 2017-07-18:
Are you using the compiler that came with Ubuntu 16.04 or have you added or built your own? If you're not using the 16.04 compiler, that could be the issue.

According to the documentation (http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#detailed-supported-distros) Ubuntu 16.04 is a support distro. That means builds are done with that distro regularly (nightly?). Even though the nightly builds don't do cubietruck2 builds, the compiler issues you're seeing are with -native recipes, and the -native recipes would be built quite regularly.

Do you have all the required build machine packages installed?
http://www.yoctoproject.org/docs/latest/ref-manual/ref-manual.html#ubuntu-packages

NFS disks aren't supported for building https://lists.yoctoproject.org/pipermail/yocto/2013-August/015452.html

NTFS is out: https://lists.yoctoproject.org/pipermail/yocto/2012-January/004133.html

I don't know if you mentioned whether this is a VM or not. I think some people are able to build using VMs, but how the disks are used and connected is probably very important.

I think your original problem has been solved (nothing provides libegl). I think we're onto something else that is unrelated.

JOE-JOE-NGIGI, 2017-07-19:
> docs/latest/ref-manual/ref-manual.html#detailed-supported-distros) Ubuntu
> manual.html#ubuntu-packages
> org/pipermail/yocto/2013-August/015452.html
> January/004133.html

It's not a VM.

closed 2017-07-27 by JOE-JOE-NGIGI