linux-sunxi/meta-sunxi #194: Ethernet not working on NanoPi Neo for core-image-base

TheMeaningfulEngineer, 2017-12-08:
The following build steps don't result in an image with working ethernet.
Just copy pasting them should result in an image build.

```
mkdir yocto
cd yocto
export YOCTO_DIR=`pwd`
git clone git://git.openembedded.org/meta-openembedded meta-openembedded --branch rocko
git clone git://git.openembedded.org/openembedded-core oe --branch rocko
git clone git://git.openembedded.org/bitbake oe/bitbake --branch 1.36
git clone git://github.com/linux-sunxi/meta-sunxi.git meta-sunxi 
source oe/oe-init-build-env tmp
sed -i 's,MACHINE ??= .*,MACHINE= "nanopi-neo",' conf/local.conf
sed -i '5,10d' conf/bblayers.conf
echo 'BBLAYERS ?= " \'                                     >> conf/bblayers.conf
echo  $YOCTO_DIR'/oe/meta \'                             >> conf/bblayers.conf
echo  $YOCTO_DIR'/meta-openembedded/meta-oe \'           >> conf/bblayers.conf
echo  $YOCTO_DIR'/meta-openembedded/meta-networking \'   >> conf/bblayers.conf
echo  $YOCTO_DIR'/meta-openembedded/meta-python \'       >> conf/bblayers.conf
echo  $YOCTO_DIR'/meta-sunxi "'                          >> conf/bblayers.conf
bitbake core-image-base

#Flashed the core-image-base-nanopi-neo-20171208175722.rootfs.sunxi-sdimg
```

Boot log and trying to run ethernet:

```
U-Boot SPL 2017.03-g8537ddd (Dec 08 2017 - 16:37:16)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2017.03-g8537ddd (Dec 08 2017 - 16:37:16 +0100) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi NEO
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
reading /boot.scr
714 bytes read in 18 ms (38.1 KiB/s)
## Executing script at 43100000
reading sun8i-h3-nanopi-neo.dtb
13890 bytes read in 25 ms (542 KiB/s)
reading zImage
** Unable to read file zImage **
reading boot/zImage
** Unable to read file boot/zImage **
reading uImage
3853256 bytes read in 217 ms (16.9 MiB/s)
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.14.0-rc3
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3853192 Bytes = 3.7 MiB
[... 187 lines cut ...]
[    3.031826] Bluetooth: BNEP filters: protocol 
[    3.036378] Bluetooth: BNEP socket layer initialized

OpenEmbedded nodistro.0 nanopi-neo /dev/ttyS0

nanopi-neo login: root
root@nanopi-neo:~# ifconfig -a
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@nanopi-neo:~# 
```

NguyenTrongThinh, 2017-12-14:
In Kernel 4.14, emac driver is unavailable. see this link http://linux-sunxi.org/Sun8i_emac

TheMeaningfulEngineer, 2017-12-14:
From http://linux-sunxi.org/Sun8i_emac:
> The second is dwmac-sun8i, this a glue to the stmmac driver. You can find the latest version at https://github.com/montjoie/linux/tree/dwmac-sun8i-current
This driver is mainline, but DT was reverted in 4.13-rc7. DT should be back soon.

The DT (Device Tree) was reverted not to support the mainlined driver?
Am I getting this right (because it seems somewhat strange)?

So it's just a matter of a correct device tree with the mainline 3.14 to get it running again?

NguyenTrongThinh, 2017-12-18:
I use the first driver which is discontinued. Here is source kernel  https://github.com/montjoie/linux/tree/sun8i-emac-wip

I add these to device tree

**file sun8i-h3-nanopi.dtsi**

```
+&emac {
+	phy = <&phy1>;
+	phy-mode = "mii";
+	allwinner,use-internal-phy;
+	allwinner,leds-active-low;
+	status = "okay";
+	phy1: ethernet-phy@1 {
+		reg = <1>;
+	};
+};
```
**file sun8i-h3.dtsi**
```
+		emac: ethernet@1c30000 {
+			compatible = "allwinner,sun8i-h3-emac";
+			reg = <0x01c30000 0x104>, <0x01c00030 0x4>;
+			reg-names = "emac", "syscon";
+			interrupts = <GIC_SPI 82 IRQ_TYPE_LEVEL_HIGH>;
+			resets = <&ccu RST_BUS_EMAC>, <&ccu RST_BUS_EPHY>;
+			reset-names = "ahb", "ephy";
+			clocks = <&ccu CLK_BUS_EMAC>, <&ccu CLK_BUS_EPHY>;
+			clock-names = "ahb", "ephy";
+			#address-cells = <1>;
+			#size-cells = <0>;
+			status = "disabled";
+		};
```

MicroJoe, 2018-02-02:
> So it's just a matter of a correct device tree with the mainline 3.14 to get it running again?

I hope you meant 4.14 here. I am using 4.14 from linux-megous [1] that provides a patched device-tree for Nanopi NEO dwmac-sun8i support [2] as it was not present in mainline 4.14 [3]. It is however available in 4.15 release [4].

However I am in the process of migration from linux-megous to linux-mainline with this meta-sunxi layer. I guess I can propose a patch for the device tree for stable 4.13 kernel used here.

[1] : https://github.com/megous/linux
[2] : https://github.com/megous/linux/blob/orange-pi-4.14/arch/arm/boot/dts/sun8i-h3-nanopi-neo.dts
[3] : http://elixir.free-electrons.com/linux/v4.14.16/source/arch/arm/boot/dts/sun8i-h3-nanopi-neo.dts
[4] : http://elixir.free-electrons.com/linux/v4.15/source/arch/arm/boot/dts/sun8i-h3-nanopi-neo.dts

ebutera, 2018-06-10:
linux-mainline is now at 4.16.13 so it should be fixed, please reopen if it is not.

closed 2018-06-10 by ebutera