linux-sunxi/meta-sunxi #203: U-Boot Recipe bleeding out of sysroot into host environment...

madscientist42, 2018-01-25:
While it doesn't break much right now, it doesn't work right.  While the recipe specifies python3-native, it doesn't use it and ends up using the host's libpython3-dev Python.h  If you don't have the host's libpython3-dev packaging installed, it can't find Python.h and won't complete a build.

Can't say if it's an oops in the recipe or if their current buildsystem metadata needs to be patched.  Will work on a pull request for this shortly once I figure out what needs patching to do this "right".

xuhcc, 2018-02-18:
Hi @madscientist42 

Have you found a solution?

For me recipe fails even when libpython3-dev is installed on the host system. I tried to add

```
inherit python3native
DEPENDS += " python3 "
```

However, it still throws an error:

```
build/tmp/work/orange_pi_one-poky-linux-gnueabi/u-boot/2_v2017.11+gitAUTOINC+c253573f3e-r0/recipe-sysroot/usr/include/python3.5m/pyconfig.h:57:36: fatal error: python3.5m/pyconfig-64.h: No such file or directory
```

macpijan, 2018-03-16:
Hi @xuhcc . You may try to install python packages from this PR as well: https://github.com/3mdeb/yocto-docker/pull/10
Or just use this docker container to perform the build. It seems to be working.

xuhcc, 2018-03-26:
@macpijan Thanks, it worked.

closed 2018-06-10 by ebutera