linux-sunxi/meta-sunxi #209: Eathernet MAC

Hitendraprajapatisls, 2018-03-06:
I'm working on Orange-pi zero board.
Sometimes , after rebooting of system , I Noticed that MAC address is changed.
From where It is reading MAC Address??
any help on that?

Hitendraprajapatisls, 2018-03-12:
any help ?? please update?

NguyenTrongThinh, 2018-04-23:
MAC Address is generated randomly on each reboot. The address is generated in device driver then written to chip Ethernet. Below is code to generate MAC address in driver. 

`	/* Read MAC-address from DT */
	mac_addr = of_get_mac_address(np);
	if (mac_addr)
		memcpy(ndev->dev_addr, mac_addr, ETH_ALEN);

	/* Check if the MAC address is valid, if not get a random one */
	if (!is_valid_ether_addr(ndev->dev_addr)) {
		eth_hw_addr_random(ndev);
		dev_warn(&pdev->dev, "using random MAC address %pM\n",
			 ndev->dev_addr);
	}` 

This is [full source driver](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/ethernet/allwinner/sun4i-emac.c).

closed 2018-06-10 by ebutera