linux-sunxi/meta-sunxi #211: putty issue in mainline 4.14.15 kernel on orange-pi zero

Hitendraprajapatisls, 2018-03-19:
Hi,
From the windows PC, Using Putty i'm not able to login on orange-pi Zero via ssh.
can anyone help  ?

Hitendraprajapatisls, 2018-03-19:
Dear sir/madam, We are currently use the h2+ soc based board for our product. We use the "4.14.15" kernel and yocto build system for Linux. now we found one issue on our board as per below description. We are connecting our board in network using Ethernet cable and we get easily remote support of board from other Linux system(PC) but when we try to connect with windows system the board does not connect using SSH/scp/webpage and it gives the "Incoming packet was garbled on decryption" error on Windows system(PC). So Please help us to solve our issue. also find the some another information which will help us for knowing the basic OS environment.
 root@orange-pi-zero:~# uname -a
 Linux orange-pi-zero 4.14.15 #6 SMP Wed Mar 14 10:21:30 IST 2018 armv7l GNU/Linux 
root@orange-pi-zero:~# ssh -V
 OpenSSH_7.5p1, OpenSSL 1.0.2n 7 Dec 2017
 root@orange-pi-zero:~# openssl version OpenSSL 1.0.2n 7 Dec 2017

nandra, 2019-05-28:
@Hitendraprajapatisls which tool do you use on windows to connect? Thanks.

nandra, 2019-08-07:
Closing due to the lack of reply.

closed 2019-08-07 by nandra