linux-sunxi/meta-sunxi #224: u-boot v2018.03 fails compilation (Python.h: No such file or directory)

MaxPayne86, 2018-07-30:
Hi,

I'm building my yocto image using meta-sunxi, poky and meta-openembedded master branches.
During compilation it fails with this error

libfdt_wrap.c:149:21: fatal error: Python.h: No such file or directory

NOTE: in my image I'm including Python3 and in 
nanopi_neo2-poky-linux/u-boot/2_v2018.03+gitAUTOINC+f95ab1fb6e-r0/recipe-sysroot-native/usr/include/python3.5m
all the header files required can be found, but it seems the compiler is searching them under
nanopi_neo2-poky-linux/u-boot/2_v2018.03+gitAUTOINC+f95ab1fb6e-r0/recipe-sysroot-native/usr/include

MaxPayne86, 2018-07-31:
solved installing python-devel on my host machine (python3-devel is not enough)

closed 2018-07-31 by MaxPayne86

akohlsmith, 2022-02-16:
Just a minor bump: on Ubuntu at least, the package is `python-dev-is-python3`