linux-sunxi/meta-sunxi #227: X11 image failure - dependency on libgbm

macpijan, 2018-08-10:
The build of standard `X11` image (`core-image-sato`) fails due to the:

```
ERROR: Nothing PROVIDES 'virtual/libgbm' (but /home/user/storage/projects/yocto-sunxi/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.19.6.bb DEPENDS on or otherwise requires it)
mesa PROVIDES virtual/libgbm but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa-gl, not mesa
NOTE: Runtime target 'xserver-xorg' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-xserver' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-utils' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-base' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-base', 'packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-base', 'packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
```

It seems to be due to the `xserver-xorg` dependency on `libgm`. By default, there is `PACKAGECONFIG[glamor]` enabled: http://cgit.openembedded.org/openembedded-core/tree/meta/recipes-graphics/xorg-xserver/xserver-xorg.inc#n135
which pulls-in `libgbm` as dependency (and sunxi-mali does not provide it). Do you think we should remove this `glamor` from `PACKAGECONFIG` in `xserver-xorg` bbapend file to avoid failing at this point?

slapin, 2018-08-10:
> openembedded-core/tree/meta/recipes-graphics/xorg-xserver/
> xserver-xorg.inc#n135

I thing glamor will not work on gles2 anyway, so I'd disable it from
building.

slapin, 2019-09-07:
will merge this one into master too

commit 26f427ca (2019-09-07): linux-sunxi: disable glamor build if Mali is used

Unless somebody writes libgbm replacement for binary Mali driver
(sort of converter for gbm-UMB) there is no glamor for us.

Closes #227
commit b20940ed (2019-09-07): linux-sunxi: disable glamor build if Mali is used

Unless somebody writes libgbm replacement for binary Mali driver
(sort of converter for gbm-UMB) there is no glamor for us.

Closes #227
commit 345cd0d0 (2019-09-07): linux-sunxi: disable glamor build if Mali is used

Unless somebody writes libgbm replacement for binary Mali driver
(sort of converter for gbm-UMB) there is no glamor for us.

Closes #227
closed 2019-09-07 by slapin