linux-sunxi/meta-sunxi #235: nanopi-neo2 not in COMPATIBLE_MACHINE

humariel, 2018-11-05:
Hi.

I created an image to run on my nanopi-neo2, but when booting I have an error related to the device tree.
So I followed your suggestion to add the following lines to my conf/local.conf file:
PREFERRED_PROVIDER_u-boot="u-boot-sunxi" 
PREFERRED_PROVIDER_virtual/bootloader="u-boot-sunxi" PREFERRED_PROVIDER_virtual/kernel="linux-sunxi"

But now I get an error saying that nothing providesvirtual/kernel because my machine is not in COMPATIBLE_MACHINE.
 
==============================================================================
Here is the output of the first error:
U-Boot SPL 2018.03-gf95ab1f (Oct 07 2018 - 23:32:30 +0000)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2018.03-gf95ab1f (Oct 07 2018 - 23:32:30 +0000) Allwinner Technology

CPU:   Allwinner H5 (SUN50I)
Model: FriendlyARM NanoPi NEO 2
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

Failed (-5)                                                                     
Loading Environment from MMC... *** Warning - bad CRC, using default environment
                                                                                
Failed (-5)                                                                     
In:    serial                                                                   
Out:   serial                                                                   
Err:   serial                                                                   
Net:   No ethernet found.                                                       
starting USB...                                                                 
No controllers found                                                            
Hit any key to stop autoboot:  0                                                
switch to partitions #0, OK                                                     
mmc0 is current device                                                          
Scanning mmc 0:1...                                                             
Found U-Boot script /boot.scr                                                   
495 bytes read in 13 ms (37.1 KiB/s)                                            
## Executing script at 4fc00000                                                 
** Unable to read file allwinner/sun50i-h5-nanopi-neo2.dtb **                   
19495424 bytes read in 973 ms (19.1 MiB/s)                                      
ERROR: Did not find a cmdline Flattened Device Tree                             
Could not find a valid device tree                                              
SCRIPT FAILED: continuing... 
==============================================================================

And the output of the second error:

ERROR: Nothing PROVIDES 'virtual/kernel'
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-mainline PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-sunxi, not linux-mainline
linux-yocto-dev PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
**linux-sunxi PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)**
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-mainline PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-sunxi, not linux-mainline
linux-dummy PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-sunxi, not linux-dummy
linux-yocto-rt PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-dummy PROVIDES virtual/kernel but was skipped: PREFERRED_PROVIDER_virtual/kernel set to linux-sunxi, not linux-dummy
linux-yocto-tiny PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto-dev PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)
linux-yocto PROVIDES virtual/kernel but was skipped: incompatible with machine nanopi-neo2 (not in COMPATIBLE_MACHINE)

geomatsi, 2018-11-05:
IIUC this board is a fairly new one, so it is not supported by legacy linux-sunxi and u-boot-sunxi. So it makes sense to use default (mainline) kernel and bootloader for this board. What kind of DTS problem do you have when you use default images for that board ?

humariel, 2018-11-05:
If I build the mainline kernel, when the machine boots, the boot fails with the first error in the previous message

geomatsi, 2018-11-06:
Hi,
A series of quick questions:
1. Could you please post here your local.conf and bblayers.conf
2. Could you please check the contents of boot partition on your sdcard: does allwinner/sun50i-h5-nanopi-neo2.dtb exist ? If dtb does not exists, then no need to proceed to the next question.
3. You may have encountered the issue #232. To do a quick check, could you please rebuild kernel image using the following commands:
```
$ bitbake -c cleanall virtual/kernel
$ bitbake virtual/kernel
```
Then copy new Image to sdcard and try to boot again.

Regards,
Sergey

humariel, 2018-11-06:
In my local.conf file the only difference is the machine variable which I set to **MACHINE ??= "nanopi-neo2"** .

This is my bblayers.conf: 
==============================================================================
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/gabriel/Desktop/kek/poky/meta \
  /home/gabriel/Desktop/kek/poky/meta-poky \
  /home/gabriel/Desktop/kek/poky/meta-yocto-bsp \
  /home/gabriel/Desktop/kek/openembedded-core/meta \
  /home/gabriel/Desktop/kek/meta-openembedded/meta-oe  \
  /home/gabriel/Desktop/kek/meta-openembedded/meta-python \
  /home/gabriel/Desktop/kek/meta-openembedded/meta-networking \
  /home/gabriel/Desktop/kek/meta-sunxi \
  "
==============================================================================
I'm using the default branch on all of the layers.

I checked my sd card and the file sun50i-h5-nanopi-neo2.dtb does exist but it's not inside a folder named allwinner. It's a file inside the boot directory.

humariel, 2018-11-06:
In my local.conf file the only difference is the machine variable which I set to `MACHINE ??= "nanopi-neo2"` .

This is my bblayers.conf:
```
POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " 
/home/gabriel/Desktop/kek/poky/meta 
/home/gabriel/Desktop/kek/poky/meta-poky 
/home/gabriel/Desktop/kek/poky/meta-yocto-bsp 
/home/gabriel/Desktop/kek/openembedded-core/meta 
/home/gabriel/Desktop/kek/meta-openembedded/meta-oe 
/home/gabriel/Desktop/kek/meta-openembedded/meta-python 
/home/gabriel/Desktop/kek/meta-openembedded/meta-networking 
/home/gabriel/Desktop/kek/meta-sunxi 
"
```
I'm using the default branch on all of the layers.

I checked my sd card and the file sun50i-h5-nanopi-neo2.dtb does exist but it's not inside a folder named allwinner. It's a file inside the boot directory.

geomatsi, 2018-11-06:
Well, boot.scr scripts expects to find dtb in allwinner directory. Could you please create that directory and move dtb there, and then to try boot the board again ?

I built images for neo2 today and checked sdimg: dtb is in allwinner directory as expected. By the way, how do you prepare your sdcard ? Manually or you are using 'dd' for sdimg image ?

Regards,
Sergey

humariel, 2018-11-06:
I've already created the directory and moved the dtb file into it, but got the same error when booting.
I used 'dd' to flash the card. Here's what I did : 
`sudo dd if=tmp/deploy/images/nanopi-neo2/core-image-base-nanopi-neo2.sunxi-sdimg of=/dev/mmcblk0`

geomatsi, 2018-11-06:
You mean the same u-boot log ?
```
...
Found U-Boot script /boot.scr                                                   
495 bytes read in 13 ms (37.1 KiB/s)                                            
## Executing script at 4fc00000       
** Unable to read file allwinner/sun50i-h5-nanopi-neo2.dtb **   
...
```

humariel, 2018-11-06:
Yes, exactly.

geomatsi, 2018-11-07:
Well, but that is weird. Could you please double-check again the content of your boot partition on sdcard and make sure that dtb is where it should be, including name spelling check.

Besides, try to boot the board manually using u-boot CLI commands. I assume that nanopi-neo2 does not have a emmc or nand, so you may try the following commands (check command help if needed):
```
U-Boot> mmc rescan
// check contents of boot partition and make sure that dtb is there:
U-Boot> fatls mmc 0:1 /
U-Boot> fatls mmc 0:1 /allwinner/
U-Boot> fatload mmc 0:1 ${kernel_addr_r} Image
// what is the output of this command ?
U-Boot> fatload mmc 0:1 ${fdt_addr_r} /allwinner/sun50i-h5-nanopi-neo2.dtb
// if previous command succeeded then try boot kernel w/ dtb:
U-Boot> booti ${kernel_addr_r} - ${fdt_addr_r}
```

humariel, 2018-11-07:
You were right, it was weird. I only found the dtb file in the partition that wasn't the boot one. So I placed it under allwinner/ there. I just copied it to the boot partition and the machine booted.

Thank you so much for your time and your help!
Best regards,
Gabriel

geomatsi, 2018-11-08:
Ok, I could reproduce the same issue (missing dtb file on boot partition) for another arm64 board  orange-pi-zero-plus2. I will take a look.

ebutera, 2018-11-21:
fixed merging #236

closed 2018-11-21 by ebutera