linux-sunxi/meta-sunxi #237: Opi zero how get wifi on core-image-minimal ? How to edit device tree ?

PJfixer, 2018-11-10:
Hi there, 
Firstly i want to apologise because i am not sure this is a real issue...i think this is more a newbie question since i am student that just begin to work with Yocto.

i would just ask if the wifi on Opi-zero is currently supported ? if yes,how configure it in yocto ?
an other things, how i can edit the device tree to add the spi NOR memory support in the image 
http://linux-sunxi.org/Xunlong_Orange_Pi_Zero#Installing_from_linux

thanks for your time

geomatsi, 2018-11-10:
Hi,
WiFi support is not [yet] in master. It depends on maintainers whether it will be or not. For the time being you may try the following sources:
- submitted pull-request #233 
- my additional [layer](https://github.com/geomatsi/meta-sunxi-contrib) which contains both wifi support and spi flash patches
- alternatively, you may use buildroot where wifi on orange-pi-zero is supported out of the box 

Regards,
Sergey

PJfixer, 2018-11-10:
Hi @geomatsi
I want to thank you,your quick answer and this additional layer! 

> alternatively, you may use buildroot where wifi on orange-pi-zero is supported out of the box

yes,i know that because i built my first images with buildroot , now i try to train with Yocto which is more complex ,but also more powerfull 

best Regards,

Pierre

PJfixer, 2018-11-11:
Hi @geomatsi after build core-image-minimal,i am not unable to start the kernel...it seems u-boot is not able to read the device tree 
`mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
714 bytes read in 18 ms (38.1 KiB/s)
## Executing script at 43100000
** Unable to read file sun8i-h2-plus-orangepi-zero.dtb **
** Unable to read file boot/sun8i-h2-plus-orangepi-zero.dtb **
** Unable to read file zImage **
** Unable to read file boot/zImage **
4026392 bytes read in 201 ms (19.1 MiB/s)
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.16.13
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4026328 Bytes = 3.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...
`
At the beginning i was thinking that the sdcard corrupted or incomplete when i use the dd command , so i have formated my sd card and perfom a new dd but i got the same problem...do you have some idee of what is wrong here ? 

maybe i am wrong, but i have understand that i need to build all of this with master branch,so all the layers here are master banch is it a problem ?

**here the layers **
`NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta                  /home/hdd/poky/meta                       5
meta-poky             /home/hdd/poky/meta-poky                  5
meta-yocto-bsp        /home/hdd/poky/meta-yocto-bsp             5
meta-sunxi-contrib    /home/hdd/poky/meta-sunxi-contrib         9
meta-sunxi            /home/hdd/poky/meta-sunxi                 10
meta-oe               /home/hdd/poky/meta-openembedded/meta-oe  6
`
i have add : IMAGE_INSTALL_append = " xradio-mod " in the local.conf file

`U-Boot SPL 2018.03-gf95ab1f (Nov 11 2018 - 13:42:03 +0000)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2018.03-gf95ab1f (Nov 11 2018 - 13:42:03 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: Xunlong Orange Pi Zero
DRAM:  512 MiB
MMC:   SUNXI SD/MMC: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

Failed (-5)
Loading Environment from MMC... *** Warning - bad CRC, using default environment

Failed (-5)
In:    serial
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
714 bytes read in 18 ms (38.1 KiB/s)
## Executing script at 43100000
** Unable to read file sun8i-h2-plus-orangepi-zero.dtb **
** Unable to read file boot/sun8i-h2-plus-orangepi-zero.dtb **
** Unable to read file zImage **
** Unable to read file boot/zImage **
4026392 bytes read in 201 ms (19.1 MiB/s)
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.16.13
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4026328 Bytes = 3.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...

Device 0: device type unknown
... is now current device
** Bad device usb 0 **
** Bad device usb 0 **
_sun8i_emac_eth_init: Timeout
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/01-02-42-03-c5-53-5c
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-sunxi
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
*** ERROR: `serverip' not set
Config file not found
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5

Abort
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4

Abort
`

geomatsi, 2018-11-11:
Hi @PJfixer ,
Yeah, there is a problem on master due to recent changes of device tree files naming conventions in poky. Could you please apply the patch from PR #236 . Besides, take a look at two other PRs: #234 and #238 in the case of further problems.

Regards,
Sergey

nandra, 2018-11-26:
@PJfixer all pull requests should be merged (wifi support + fixing kernel boot).

PJfixer, 2018-11-28:
Hi @nandra @geomatsi   this pretty cool thanks a lot for your work, it will be usefull for me soon 

Regards,

Pierre

nandra, 2018-11-28:
@PJfixer :+1:

closed 2018-11-28 by ebutera