linux-sunxi/meta-sunxi #240: bananapi core-image-sato thud

voloviq, 2018-11-28:
When trying to build core-image-sato for bananapi get the following error

> ERROR: Nothing PROVIDES 'dri2proto' (but /home/mw/yocto/poky/meta-sunxi/recipes-graphics/libgles/sunxi-mali_git.bb DEPENDS on or otherwise requires it)
> NOTE: Runtime target 'gstreamer1.0-libav' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> NOTE: Runtime target 'gst-examples' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> NOTE: Runtime target 'packagegroup-core-x11-sato-games' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato-games', 'gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> NOTE: Runtime target 'packagegroup-core-x11-sato-apps' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato-apps', 'gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> NOTE: Runtime target 'packagegroup-core-x11-sato-base' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato-base', 'gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> NOTE: Runtime target 'packagegroup-core-x11-sato' is unbuildable, removing...
> Missing or unbuildable dependency chain was: ['packagegroup-core-x11-sato', 'gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> ERROR: Required build target 'core-image-sato' has no buildable providers.
> Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-sato', 'gst-examples', 'gstreamer1.0-libav', 'gstreamer1.0-plugins-base', 'virtual/libgles2', 'dri2proto']
> 

of course in local.conf I added 

```
PACKAGECONFIG_remove_pn-xserver-xorg = "glamor"
IMAGE_FEATURES += "x11"
DISTRO_FEATURES_append = " opengl x11"
DISTRO_FEATURES_remove = "wayland"
```
Missing some dependancies?

medemi68, 2019-01-16:
Hi,

I actually came across your issue because I was looking for the PACKAGECONF option that you had on the first line. 

It seems meta-sunxi hasn't been updated recently. dri2proto has been replaced with xorgproto in Thud, so has xextproto (replaced with xorgproto), you may have to modify the dependencies in all the files that contain it, or create some form of bbappend file to modify the dependencies.

Best of luck,

Thanks.

voloviq, 2019-01-18:
Thanks for the answer, I do as You suggested and compilation start. Unfortunately, I stack on gstreamer. Please take a look

> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: QA Issue: /usr/lib/libgstgl-1.0.so.0.1404.0 contained in package libgstgl-1.0 requires libGLESv2.so, but no providers found in RDEPENDS_libgstgl-1.0? [file-rdeps]
> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: QA Issue: /usr/lib/libgstgl-1.0.so.0.1404.0 contained in package libgstgl-1.0 requires libEGL.so, but no providers found in RDEPENDS_libgstgl-1.0? [file-rdeps]
> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: QA Issue: /usr/lib/gstreamer-1.0/libgstopengl.so contained in package gstreamer1.0-plugins-base-opengl requires libEGL.so, but no providers found in RDEPENDS_gstreamer1.0-plugins-base-opengl? [file-rdeps]
> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: QA Issue: /usr/lib/gstreamer-1.0/libgstopengl.so contained in package gstreamer1.0-plugins-base-opengl requires libGLESv2.so, but no providers found in RDEPENDS_gstreamer1.0-plugins-base-opengl? [file-rdeps]
> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
> ERROR: gstreamer1.0-plugins-base-1.14.4-r0 do_package_qa: Function failed: do_package_qa
> ERROR: Logfile of failure stored in: /home/mw/yocto/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/gstreamer1.0-plugins-base/1.14.4-r0/temp/log.do_package_qa.13062
> ERROR: Task (/home/mw/yocto/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.14.4.bb:do_package_qa) failed with exit code '1'

How can I resolve this issue? What happens with gstreamer?

Regards
Michal

medemi68, 2019-01-30:
Weird.

I seem to be having the same issue when trying to compile something that uses gstreamer as well.

Any luck?

medemi68, 2019-01-31:
Hello,

So I managed to find a bit of a work around, that involves using Mesa's software rendering as opposed to the hardware rendering. I'm using this temporarily until we can get this working.

In your local.conf file, add:

```
PREFERRED_PROVIDER_virtual/libgl = "mesa"
PREFERRED_PROVIDER_virtual/libgles2 = "mesa"
PREFERRED_PROVIDER_virtual/libgles1 = "mesa"
PREFERRED_PROVIDER_virtual/egl = "mesa"
PREFERRED_PROVIDER_virtual/mesa = "mesa"
```

Also, ensure that you have opengl selected (I put it in `DISTRO_FEATURES`), and that you also have `meta-openembedded` layer added to your configuration.

Hope that helps for now. I'll have a look at the hardware rendering portion using sunxi-mali at some point in the future.

voloviq, 2019-01-31:
Hello Medemi68,
You have right. I add as You suggested but I need to remove from my config files sunxi.inc and sunxi-mali.inc.
Tomorrow I check on board if works.

As I good understand the problem is with mali graphics packages, am I right?

Thanks a lot.

medemi68, 2019-02-01:
I think it has something to do with those, yeah. It seems as though this repository is not well maintained. 

Then again, I can pretty much say that for the entirety of openembedded

nandra, 2019-05-27:
@voloviq can you pls try merge #252 and #253 and try? #253 is kind of hack but looking for proper solution. Thanks.

voloviq, 2019-05-27:
Hi Mark,
I started compilation, in my host, it takes about 2 hours.
Shall I still keep 
Regards
Michal

nandra, 2019-05-28:
No pls drop those changes. Just use thud branch and merge #252 and #253 That should be it. Thanks.

voloviq, 2019-05-28:
Hello Mark,
Unfortunately, I get an error

> ERROR: Nothing PROVIDES 'virtual/libgbm' (but /home/mw/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.20.1.bb DEPENDS on or otherwise requires it)
mesa PROVIDES virtual/libgbm but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa-gl, not mesa
NOTE: Runtime target 'xserver-xorg' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-xserver' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-base' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-base', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-base', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm'

I do not touch anything than mentioned #252 and #253. Just clear take meta-sunxi thud branch applied patch.
It seems that some PREFERRED_ are missing.

nandra, 2019-05-28:
@voloviq ok thanks for reporting. Can you pls post conf/local.conf + bblayer.conf (stripped if contains customer stuff) and provide which command you're running to build? I'll try it here. Did you do clean build (wipe out everything?).Thanks.

voloviq, 2019-05-28:
The content is not confident I can share it, this is my hobby

To build I use simple
> souce oe ....
> bitbake core-image-sato

bblayers.conf

> POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
> changes incompatibly
> POKY_BBLAYERS_CONF_VERSION = "2"
> 
> BBPATH = "${TOPDIR}"
> BBFILES ?= ""
> 

> BBLAYERS ?= " \
>  ${HOME}/yocto/poky/meta \
>  ${HOME}/yocto/poky/meta-poky \
>  ${HOME}/yocto/poky/meta-openembedded/meta-oe \
>  ${HOME}/yocto/poky/meta-openembedded/meta-networking \
>  ${HOME}/yocto/poky/meta-openembedded/meta-python \
>  ${HOME}/yocto/poky/meta-qt5 \
>  ${HOME}/yocto/poky/meta-sunxi \
>   "

local.conf

Just minor changes from original 

> MACHINE ??= "bananapi"
> DL_DIR ?= "${HOME}/yocto/downloads"
> SSTATE_DIR ?= "${HOME}/yocto/sstate-cache
> TMPDIR = "${HOME}/yocto/tmp"
> PACKAGE_CLASSES ?= "package_ipk"
> RM_OLD_IMAGE = "1"
> INHERIT += "rm_work"

Everythings was clear before start compilation.

nandra, 2019-05-28:
Are you on actual master? I have those when run build:
```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'virtual/libgbm' (but /home/builder/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.20.4.bb DEPENDS on or otherwise requires it)
mesa PROVIDES virtual/libgbm but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa-gl, not mesa
NOTE: Runtime target 'xserver-xorg' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-xserver' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-utils' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11-base' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11-base', 'packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'core-image-sato' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-core-x11-base', 'packagegroup-core-x11-utils', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
NOTE: Runtime target 'packagegroup-core-x11' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-core-x11', 'packagegroup-core-x11-xserver', 'xserver-xorg', 'virtual/libgbm']
ERROR: Nothing PROVIDES 'virtual/xserver' (but /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-keyboard_1.9.0.bb, /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-evdev_2.10.6.bb, /home/builder/sunxi/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb, /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-mouse_1.9.3.bb DEPENDS on or otherwise requires it)
No eligible PROVIDERs exist for 'virtual/xserver'
NOTE: Runtime target 'xf86-input-keyboard' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-keyboard', 'virtual/xserver']
NOTE: Runtime target 'xf86-input-evdev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-evdev', 'virtual/xserver']
NOTE: Runtime target 'xf86-video-fbturbo' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-video-fbturbo', 'virtual/xserver']
NOTE: Runtime target 'xf86-input-mouse' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-mouse', 'virtual/xserver']
ERROR: Nothing RPROVIDES 'xf86-video-fbturbo-dev' (but /home/builder/sunxi/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'xf86-video-fbturbo-dev'
NOTE: Runtime target 'xf86-video-fbturbo-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-video-fbturbo-dev']
ERROR: Nothing RPROVIDES 'xf86-input-evdev-dev' (but /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-evdev_2.10.6.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'xf86-input-evdev-dev'
NOTE: Runtime target 'xf86-input-evdev-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-evdev-dev']
NOTE: Runtime target 'xf86-input-libinput' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-libinput']
ERROR: Nothing RPROVIDES 'xserver-xorg-dev' (but /home/builder/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.20.4.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'xserver-xorg-dev'
NOTE: Runtime target 'xserver-xorg-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xserver-xorg-dev']
ERROR: Nothing RPROVIDES 'libegl' (but /home/builder/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.20.4.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'libegl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libegl']
ERROR: Nothing RPROVIDES 'xf86-input-keyboard-dev' (but /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-keyboard_1.9.0.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'xf86-input-keyboard-dev'
NOTE: Runtime target 'xf86-input-keyboard-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-keyboard-dev']
ERROR: Nothing RPROVIDES 'xf86-input-mouse-dev' (but /home/builder/poky/meta/recipes-graphics/xorg-driver/xf86-input-mouse_1.9.3.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'xf86-input-mouse-dev'
NOTE: Runtime target 'xf86-input-mouse-dev' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['xf86-input-mouse-dev']
```

nandra, 2019-05-28:
Sorry was using poky warrior (not thud).

voloviq, 2019-05-28:
I check it twice still the same

> bitbake core-image-minimal
> Loading cache: 100% |############################################| Time: 0:00:00
> Loaded 3132 entries from dependency cache.
> NOTE: Resolving any missing task queue dependencies
> 
> Build Configuration:
> BB_VERSION           = "1.40.0"
> BUILD_SYS            = "x86_64-linux"
> NATIVELSBSTRING      = "universal"
> TARGET_SYS           = "arm-poky-linux-gnueabi"
> MACHINE              = "bananapi"
> DISTRO               = "poky"
> DISTRO_VERSION       = "2.6.2"
> TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
> TARGET_FPU           = "hard"
> meta                 
> meta-poky            = "thud:50f33d3bfebcbfb1538d932fb487cfd789872026"
> meta-oe              
> meta-networking      
> meta-python          = "thud:4cd3a39f22a2712bfa8fc657d09fe2c7765a4005"
> meta-qt5             = "thud:0630018c0033c91fddda62a49f59a82ba6ec6850"
> meta-sunxi           = "thud:72ece33639990d8dccdf0513a07f444c2fe0192c"

> ERROR: Nothing PROVIDES 'virtual/libgbm' (but /home/mw/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_1.20.1.bb DEPENDS on or otherwise requires it)
> mesa PROVIDES virtual/libgbm but was [...]

Something is missing

voloviq, 2019-05-28:
> ~/yocto/poky/meta-sunxi$ git branch
> * thud
>

voloviq, 2019-05-28:
Do You build Your Yocto environment using 

> repo

voloviq, 2019-05-28:
Sorry, I don't see Your post above. I check it.

nandra, 2019-05-28:
I've tried and was able to workaround issue with below patch (it is combination of above + some other fixes). I need to start looking about all those dependencies as it's really mess :) (with this patch on SW rendering works though) :
```
diff --git a/conf/machine/include/sunxi-mali.inc b/conf/machine/include/sunxi-mali.inc
index 8367195..5ae0d05 100644
--- a/conf/machine/include/sunxi-mali.inc
+++ b/conf/machine/include/sunxi-mali.inc
@@ -1,13 +1,13 @@
-PREFERRED_PROVIDER_virtual/mesa ?= "mesa-gl"
-PREFERRED_PROVIDER_virtual/libgl ?= "mesa-gl"
+PREFERRED_PROVIDER_virtual/mesa ?= "mesa"
+PREFERRED_PROVIDER_virtual/libgl ?= "sunxi-mali"
 PREFERRED_PROVIDER_virtual/libgles1 ?= "sunxi-mali"
 PREFERRED_PROVIDER_virtual/libgles2 ?= "sunxi-mali"
 PREFERRED_PROVIDER_virtual/egl ?= "sunxi-mali"
 
-XSERVER += "sunxi-mali \
-            sunxi-mali-dev"
+#XSERVER += "sunxi-mali \
+#            sunxi-mali-dev"
 
-MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS += "\
- kernel-module-mali \
- kernel-module-mali-drm \
-  "
+#MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS += "\
+# kernel-module-mali \
+# kernel-module-mali-drm \
+#  "
diff --git a/conf/machine/include/sunxi.inc b/conf/machine/include/sunxi.inc
index 05109b4..ff36e6b 100644
--- a/conf/machine/include/sunxi.inc
+++ b/conf/machine/include/sunxi.inc
@@ -7,7 +7,6 @@ MACHINE_SOCARCH_SUFFIX_sun4i = "-sun4i"
 
 PREFERRED_PROVIDER_virtual/xserver = "xserver-xorg"
 XSERVER = "xserver-xorg \
-           xf86-video-fbturbo \
            xf86-input-evdev \
            xf86-input-mouse \
            xf86-input-keyboard"
diff --git a/recipes-graphics/libgles/sunxi-mali_git.bb b/recipes-graphics/libgles/sunxi-mali_git.bb
index 8153b52..3ec880b 100644
--- a/recipes-graphics/libgles/sunxi-mali_git.bb
+++ b/recipes-graphics/libgles/sunxi-mali_git.bb
@@ -100,3 +100,4 @@ FILES_${PN}-test = "${bindir}/sunximali-test"
 # These are closed binaries generated elsewhere so don't check ldflags & text relocations
 INSANE_SKIP_${PN} = "dev-so ldflags textrel"
 INSANE_SKIP_${PN}-test = "dev-so ldflags textrel"
+
diff --git a/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb b/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb
index edf7bbb..da5929d 100644
--- a/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb
+++ b/recipes-graphics/xorg-xserver/xf86-video-fbturbo_git.bb
@@ -5,7 +5,7 @@ DESCRIPTION = "X.Org X server -- A10/A13 display driver"
 LICENSE = "MIT-X"
 LIC_FILES_CHKSUM = "file://COPYING;md5=f91dc3ee5ce59eb4b528e67e98a31266"
 
-DEPENDS += "sunxi-mali libump xf86driproto"
+DEPENDS += "sunxi-mali libump xorgproto"
 
 PE = "3"
 PV = "0.5.1+git${SRCPV}"
```

voloviq, 2019-05-28:
Sorry Mark, I still not able to build core-image-sato. I make a combination of poky warrior, meta-sunxi thud. I also try to apply above Your patch but still without success. Opengl is not an easy thing to do. If You look on similar bb layer form other manufacturer it looks the same.

nandra, 2019-05-28:
OK pls give me some time. Looking into it ;). Thanks.

voloviq, 2019-05-28:
No rush, just let me know when You be ready.
Thanks

slapin, 2019-09-07:
See this for reference: https://github.com/96boards/meta-96boards/issues/195
In particular this change: https://github.com/akuster/meta-odroid/blob/master/recipes-graphics/mali/mali-450_r6p1.bb#L29

closed 2019-09-07 by slapin

slapin, 2019-09-07:
Please cherry-pick that for thud and warrior please, tested for master.

nandra, 2019-09-07:
Thanks a lot for reference. I'll do. Keep you informed.

commit 63ee6333 (2019-09-07): Fix soname in Mali.so to make auto-RDEPENDS work

The auto-RDEPENDs system reqires that all .so libraries have
proper soname set. If library soes not have soname set, it will
not be added to RDEPENDs list of other elf files.

Closes #240
commit 12b85f0d (2019-09-07): Fix soname in Mali.so to make auto-RDEPENDS work

The auto-RDEPENDs system reqires that all .so libraries have
proper soname set. If library soes not have soname set, it will
not be added to RDEPENDs list of other elf files.

Closes #240
commit 4a083b66 (2019-09-07): Fix soname in Mali.so to make auto-RDEPENDS work

The auto-RDEPENDs system reqires that all .so libraries have
proper soname set. If library soes not have soname set, it will
not be added to RDEPENDs list of other elf files.

Closes #240