linux-sunxi/meta-sunxi #244: [Master] [Orange-pi-pc-plus]  Support/Enable Wifi ? Precisions about CPUFreq & thermal?

PJfixer, 2019-01-16:
Hello all, 

I have managed to build "core-image-minimal" for a orange-pi-pc-plus. I have successfully boot on it and I have now several questions : 

1. **How to get wifi working ?**
I have tried  "ifup wlan0" but the command is not supported, then I print out the available network interfaces with "ls sys/class/net". It returns : 

-  Eth0
-Lo
-sit0
is there a way to get wlan0 or something else? 
sit0 is a "device" used to tunnel IPv6 through an IPv4 connection right?

2.**CPUfreq & thermal **
by typing " ls  /sys/devices/system/cpu/cpufreq/ "I see that the cpufreq folder is empty, does it means that we cant tune the H3 parameters? because of the hardware design?

/sys/class/thermal is also empty  ?

Thanks for your time 

Regards.

geomatsi, 2019-01-16:
Hi @PJfixer ,

1. WiFi
This board has one of the numerous Realtek chips. AFAIK this particular chip is not yet supported by mainline or staging Realtek drivers. Which means that an out-of-tree wireless driver needs to be used - see branch rtl8189 of [rfl8189x driver](github.com/jwrdegoede/rtl8189ES_linux.git) repository. This particular driver has not yet been integrated into meta-sunxi layer. So you may try to use rtl8189fs-mod_git.bb recipe from my [meta-sunxi-contrib](https://github.com/geomatsi/meta-sunxi-contrib) layer. This recipe will pull rtlwifi firmware packages as well. 

2. cpufreq and thermal
There may be two different reasons. In good case those features are just not enabled in meta-sunxi default kernel config - see recipes-kernel/linux/linux-mainline/arm/defconfig. So you may tweak that config or build your own custom kernel and dtb. However there is also a chance that some of those features are not yet available in v4.16.x kernel. In this case you may check newer kernels. I would suggest either to check [mainlining page](http://linux-sunxi.org/Linux_mainlining_effort) on linux-sunxi wiki or check kernel git log.

Regards,
Sergey

PJfixer, 2019-01-16:
Hi @geomatsi, 
thanks a lot for your help. I have made some progress : 
1.Wifi 
I have integrated meta-sunxi-contrib in bblayers.conf then I added "IMAGE_INSTALL_append = " rtl8189fs-mod". I got now the wlan0 device in sys/class/net.

next step is to connect to a router with ssid & psk. For this I have followed these link instructions in order to get wpa_supplicant: https://www.lynxbee.com/how-to-enable-wifi-with-yocto-raspberry-pi3/  is it the right way?
because I didn't know in which file add DISTRO_FEATURES_append = "wifi" so currently I added it to local.conf 

I have rebuild the images but it seems it didn't have wpa_supplicant features  
did you have some advice to solve this?

2. cpufreq & Thermal

regarding linux_mainlining page thermal features for H3 seems to be not supported yet, and cpufreq is supported on linux mainline 4.18.
get these features working is not an emergency at the moment for me, but I would ask you, in the case I want to use kernel 4.18 to get cpufreq features working how should I proceed? did I need to find a  **mainline_4.18.bb** recipes or something like that? and then edit some files in meta-sunxi ? ( I think this is probably not as simple like that  )

regards,

Pierre.

geomatsi, 2019-01-16:
Hi @PJfixer 

1. wifi
I don't know what 'wifi' feature is about. I would suggest to add wpa_supplicant to your image. This can be done either via local.conf or using custom package group and image. Take a look at Yocto documentation. Or you may use any top level layer as a reference example, e.g. [meta-raspberry](git.yoctoproject.org/meta-raspberrypi) - see recipes for images and packagegroup in recipes-core.

2. new kernel
You may switch to linux-mainline_git.bb recipe in meta-sunxi layer. Edit that recipe (see comments inside): set default preference and update git tag. But be ready for experiments since new kernel may bring some build failures, which haven't yet been addressed in meta-sunxi.

Regards,
Sergey

PJfixer, 2019-01-17:
Hi @geomatsi, 
Thanks a lot for your suggestion, I finally successfully install  wpa_supplicant by adding "IMAGE_INSTALL_append = "wpa**-**supplicant"(take care to use minus instead of underscore)  which will use the "wpa-supplicant_2.6.bb" recipes in poky/meta/recipes-connectivity

 After configuring "/etc/wpa_suppllicant.conf am now able to connect to my router with my Orange-pc-plus by running **ifup wlan0** 👍 

1. The next step is to run wlan0 and connect after booting, for that I was thinking to add an InitWlan0.sh script file in init.d, do you think its a good idea(didn't tried it yet)?

2.And one last thing after running "ifup wlan0" and succesfully connect to my router, the  rfl8189x driver is a little bit  "talkative" and it flood the serial console by logging  the events : 
912.083639] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 57
[  914.163644] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 53
[  916.243650] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 53
[  918.323645] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 53
[  919.897225] RTL871X: adaptive_early_32k(): bcn_cnt = 101
[  920.403647] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 53
[  922.483660] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 56
[  924.563649] RTL871X: CurIGValue=53, EntryMinUndecoratedSmoothedPWDB = 56

is there a way to mute this little guy? :)

Regards,

Pierre.

PJfixer, 2019-01-17:
Adding **auto wlan0** in etc/network/interfaces solve the "connect a startup" problem the orange pi now connect just after booting 👍

geomatsi, 2019-01-18:
Hi @PJfixer 

1. wifi
Yeah, you are using sysvinit, not systemd build. So it should be ok to set create proper wpa_supplicant config and auto-enable interface in /etc/network/interfaces.

2. rtlwifi
True, driver in its current build configuration is too noisy. I plan to fix this before I submit PR with this driver into meta-sunxi. There should be some compilation option to make it less verbose. You may check driver repository for README or something like that, and modify your driver recipe or its patch if needed.

P.S. By the way, if you are done with this particular thread of questions, go ahead and close the issue.

Regards,
Sergey

PJfixer, 2019-01-18:
Ok @geomatsi, thanks for your help.

Best regards,

Pierre

closed 2019-01-18 by PJfixer