linux-sunxi/meta-sunxi #258: i2c not working on nanopi neo plus2

XDeschuyteneer, 2019-06-06:
I am building a custom OS for nanopi neo plus 2 and facing issue with i2c bus using the meta-sunxi layer.

I tried "i2cdetect -y 0" without success.
Then checked if the bus was present: "ls /dev/i2c*" without success.
Then checked the controller status: "head /sys/firmware/devicetree/base/soc/i2c*/status"
`
==> /sys/firmware/devicetree/base/soc/i2c@1c2ac00/status <==
disabled
==> /sys/firmware/devicetree/base/soc/i2c@1c2b000/status <==
disabled
==> /sys/firmware/devicetree/base/soc/i2c@1c2b400/status <==
disabled
`

I checked the device tree file:
`fdtget /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status`
and get:
`disabled`

I enabled it:
`fdtput -t s /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status okay`

It was applied correctly on the dtb but the i2c is still not working.
So I tried to take this dtb and change it in my yocto layer in case it was the issue.
Anyway, I need to do it later on.

What I did:

Update de default (override) dts: https://pastebin.com/rS4pwuSR
add a linux-mainline_%.bbappend file: https://pastebin.com/kaHsReC7

And at this point, neither my dts, neither my cfg part are "working".

nandra, 2019-06-06:
This won't work:
```
SRC_URI += "file://cyanos-cygwl.dts;subdir=linux-4.16.13/arch/arm64/boot/dts \
            file://kname.cfg \
            "

```

you need to patch dts (add new dts to build and then reference it in your machine.conf). If you're creating new machine I would suggest to use only include (dtsi) and create new one.

XDeschuyteneer, 2019-06-06:
(my name.cfg isn't working neither)

Do you have an example of a dts patch?
Shouldn't i2c work out of the box with the kernel in meta-sunxi?

I am creating new machine, but just basically aliases and used internally in my meta-layer to adjust build parameters or image content.

nandra, 2019-06-06:
Look at linux-mainline_4.16.13.bb there is patch for dts ```file://0003-ARM-dts-nanopi-neo-air-Add-WiFi-eMMC.patch ``` I doubt cfg will not work so you need to explicitly update defconfig.

XDeschuyteneer, 2019-06-07:
but why i2c isn't working out of the box?

nandra, 2019-06-07:
Usually i2c is disabled in main dtsi file and should be enabled how you do it in your dts. Did you check if kernel config have I2C enabled? Thanks.

XDeschuyteneer, 2019-06-07:
I assume from what I see here:
`cat /proc/config.gz | gunzip  | grep I2C`
https://pastebin.com/uetGTrAh

nandra, 2019-06-07:
Yes looks like all is fine (this line should be present: CONFIG_I2C_MV64XXX=y). Can you post bootlog if there are some issues about i2c? Thanks.

XDeschuyteneer, 2019-06-07:
here is dmesg output:

https://pastebin.com/A4iKaFxd

nandra, 2019-06-07:
OK thanks looks good. Can you pls post also dts with changes for inspection? Thanks.

XDeschuyteneer, 2019-06-07:
The controller is disabled (as mentioned in my first post):

https://pastebin.com/GyryEzuV

nandra, 2019-06-07:
I think it's because you don't apply dts patch properly thus by default nodes are disabled. As asked can you pls post modified devicetree (not only patch). Thanks.

XDeschuyteneer, 2019-06-07:
This is my complete dts: https://pastebin.com/en5AxEM2
Which is referred in my linux-mainline_%.bbappend: https://pastebin.com/SGyTybrr

nandra, 2019-06-07:
Hmm this is weird. I2C nodes seems to be enabled (not sure why status report disabled on running system). Also do you have patch which adds your dts do compilation? Are you sure you're loading correct dts? ;)

XDeschuyteneer, 2019-06-07:
nop. (Not used to modifying DTS/kernel config in yocto.. so probably I did something wrong).
my bbappend with the dts and cfg are the only kernel changes that I apply.

How to check which dts is loaded? I didn't changed anything. I hoped that overriding the default dts from your meta-layer would be enough :-D
And in the DTS everything seems OK (pins, status, etc.).

nandra, 2019-06-07:
dts is checked from u-boot. Pls check image dts (shouuld be in /boot) if it really contains i2c changes.

XDeschuyteneer, 2019-06-07:
I assume this is this DTS which is loaded (only one present):
/boot/sun50i-h5-nanopi-neo-plus2.dtb

but when I check the status is disabled:
```
root@cy-gwl:~# fdtget /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status
disabled
```

but when I try to change it, it works on the file:
```
root@cy-gwl:~# fdtput -t s /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status okay
root@cy-gwl:~# fdtget /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status
okay
```

but after a reboot, it doesn't change nothing on the applied device tree:
```
root@cy-gwl:~# head /sys/firmware/devicetree/base/soc/i2c*/status
==> /sys/firmware/devicetree/base/soc/i2c@1c2ac00/status <==
disabled
==> /sys/firmware/devicetree/base/soc/i2c@1c2b000/status <==
disabled
==> /sys/firmware/devicetree/base/soc/i2c@1c2b400/status <==
disabled
```

nandra, 2019-06-07:
In 4.16 kernel I cannot find devicetree ```sun50i-h5-nanopi-neo-plus2.dts``` where did you get it?

XDeschuyteneer, 2019-06-07:
It is like that by default, I didn't changed anything to change that.
I just included the meta-sunxi layer.
Using the nanopi-neo-plus2 machine.

nandra, 2019-06-07:
If you look at https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/allwinner/sun50i-h5-nanopi-neo-plus2.dts thre is no setup for i2c nodes s this is the reason why no i2c node on target

XDeschuyteneer, 2019-06-07:
Yeah... And i tried to search from where does this dts come from: https://pastebin.com/JZRfwnJL
And apparently... it's nowhere...
So back to basic, how to activate the i2c bus?

nandra, 2019-06-07:
Add patch which patch above dts file with i2cnodes and add it to bbappend file (look at linux-mainline.bb file how we do it for other machines). Thx.

XDeschuyteneer, 2019-06-07:
I will check that. But pretty sure it will not fix my issue... I already override the DTS with my bbappend... without success. Will keep you updated.

XDeschuyteneer, 2019-06-07:
> Look at linux-mainline_4.16.13.bb there is patch for dts `file://0003-ARM-dts-nanopi-neo-air-Add-WiFi-eMMC.patch ` I doubt cfg will not work so you need to explicitly update defconfig.

I tried to patch existing dts:
my patch (just copied what I had on my system and generated a patch using quilt): https://pastebin.com/V4zzKgwN
compile log: https://pastebin.com/GzAicWWY

Any idea?
I checked this patch: 0003-ARM-dts-nanopi-neo-air-Add-WiFi-eMMC.patch
But it's totally different from what I generated with quilt.

nandra, 2019-06-07:
Patch is not correctly generated. Usually I do that clone linux-mailine repo then do changes generate patch using git-format-patch and then add this patch to bbappend.

XDeschuyteneer, 2019-06-07:
OK! usually I use the bitbake devshell, then generate patch using quilt :-D
Will try that!

XDeschuyteneer, 2019-06-07:
I just had time to implement what you suggested.
Something changed but it's not fixed yet.

I generated this patch: https://pastebin.com/ctVD04Yv
I updated my linux-mainline_%.bb : https://pastebin.com/7TtsGZ2N

I cleaned and rebuild everything and just tested it:

Now, the dts present on the boot partition is correct without any modification (which is a change and improvement):

```
root@cy-gwl:~# fdtget /boot/sun50i-h5-nanopi-neo-plus2.dtb /soc/i2c@1c2ac00 status
okay
```

But the active device tree is still not correct:

```
root@cy-gwl:~# head /sys/firmware/devicetree/base/soc/i2c*/status
==> /sys/firmware/devicetree/base/soc/i2c@1c2ac00/status <==
okay
==> /sys/firmware/devicetree/base/soc/i2c@1c2b000/status <==
disabled
==> /sys/firmware/devicetree/base/soc/i2c@1c2b400/status <==
disabled
```

It seems that this device tree is just not loaded.

XDeschuyteneer, 2019-06-08:
I took the DTS from the friendly elec (ubuntu core based... no source available...).
Patched the kernel with that and I went one step further:

device tree seems OK:

```
root@cy-gwl:~# head /sys/firmware/devicetree/base/soc/i2c*/status
==> /sys/firmware/devicetree/base/soc/i2c@01c2ac00/status <==
okay
==> /sys/firmware/devicetree/base/soc/i2c@01c2b000/status <==
okay
==> /sys/firmware/devicetree/base/soc/i2c@01c2b400/status <==
okay
```

bus seems present:

```
root@cy-gwl:~# ls /dev/i2c*
/dev/i2c-0  /dev/i2c-1  /dev/i2c-2
```

but when running:

`root@cy-gwl:~# i2cdetect -y 0`

I get that in dmesg:

`[ 1212.897916] i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0`

nandra, 2019-06-08:
‘’’i2c detect -r -y 0’’’

XDeschuyteneer, 2019-06-12:
I fixed some stuff and managed to get it working and no errors anymore.
Just need to cleanup/refactor some stuff...

nandra, 2019-06-13:
OK fine. Thanks for reporting back. Closing.

closed 2019-06-13 by nandra

XDeschuyteneer, 2019-06-13:
I would not close it since it's not yet fixed in this meta-layer... I have no time now to cleanup my patch. But I will try to do a pull request soon.

nandra, 2019-06-13:
OK please post pull request then we can discuss there. Thx

ZhelnovSergey, 2020-12-09:
I had same problem while connecting NanoPi Duo2 to pressure sensor via i2c
"i2c i2c-0: mv64xxx: I2C bus locked, block: 1, time_left: 0"
I had to solder pull up resistors 10k:
One between  SDA and  VCC 
Another between SDL and VCC

In my case VCC = 3,3V