linux-sunxi/meta-sunxi #259: changing DEFAULTTUNE for sun50i

lorem-ipsum, 2019-06-25:
I've tried to set the `DEFAULTTUNE` for a NanoPi Neo 2 as described in the readme to `cortexa8hf-neon`. This however fails with the following error message:
> Error, the PACKAGE_ARCHS variable (all any noarch ${PACKAGE_EXTRA_ARCHS_tune-cortexa8hf-neon} nanopi_neo2) for DEFAULTTUNE (cortexa8hf-neon) does not contain TUNE_PKGARCH (${@bb.utils.contains('TUNE_FEATURES', 'aarch64', 'aarch64', '${ARMPKGARCH_tune-cortexa8hf-neon}' ,d)}).Toolchain tunings invalid:
> Tuning 'cortexa8hf-neon' has no defined features, and cannot be used.
This seems to be due to the fact that `conf/machine/include/sun50i.inc` does not include `
conf/machine/include/tune-cortexa8.inc`, but rather directly the machine features `conf/machine/include/arm/arch-armv8a.inc`

Replacing it with the `tune-cortexa8.inc` makes DEFAULTTUNE work, but changes the default tune features to `armv7athf-neon`. Looking at `beaglebone-yocto` in the poky repository they solve this by providing `DEFAULTTUNE ?= "cortexa8hf-neon"` before including the `tune-cortexa8.inc`. I've read through the discussion which lead to the comment in the README, but as even poky sets a default, maybe it is ok to change this and provide a sensible default (with ?=)?

I would prepare a PR if you're fine with this.

nandra, 2019-06-25:
Yes please create PR and we can discuss there. But using default version with ?= is preferred way of doing such things.

lorem-ipsum, 2019-06-25:
I've tried some things and also looked at other metas (mainly meta-raspberrypi). There seems to be no solution for thud as everyone does the same. My first suggestion fails, as it then tries to find the correct dtb as it is looking in `arm` instead of `arm64`. But for warrior there is a tune-cortexa53. I will post a PR to change to this once the build has passed.

lorem-ipsum, 2019-06-25:
I've posted this PR: https://github.com/linux-sunxi/meta-sunxi/pull/261 . The issue here is mostly me misunderstanding arm labeling. The previous sets the DEFAULTTUNE to aarch64 which is as good as it gets for 64 bit ARM before the warrior release. Warrior has the more specific tunes which I posted in the PR.

closed 2019-06-25 by lorem-ipsum