linux-sunxi/meta-sunxi #260: thud branch does not work with poky thud

lorem-ipsum, 2019-06-25:
poky has moved on to warrior as new release. However commit efe4acdeffb85e9d666579c7a4b2a6446d0d55b5 got into the thud branch of this meta which is incompatible with thud of poky. The change corrected by this commit has only been introduced in warrior, but not in thud. This breaks current builds.

It is probably best to create a new warrior branch and revert the change in the thud branch.

nandra, 2019-06-25:
OK I'll revert and and set master as warrior, can you pls then report back if build works with warrior? Thanks.

lorem-ipsum, 2019-06-25:
Currently I am running a build against warrior. This error is not present in the combination of meta-sunxi:master and poky/oe:warrior. Still waiting for the build to finish to see if anything else pops up.

nandra, 2019-06-25:
OK bumped thud with revert and setup warrior branch same as current master. Thanks for report :+1:

lorem-ipsum, 2019-06-25:
found now issues with the test builds. Thanks for the really fast fix.

closed 2019-06-25 by lorem-ipsum

nandra, 2019-06-25:
My pleasure :+1: thanks for testing. Can you pls share which board you did test? Thanks.

lorem-ipsum, 2019-06-25:
So far I only did a build test as I am out of hardware today and it only affects build flags and restores old state. But will test on a NanoPi Neo 2 in the next days to make sure that this doesn't break anything.