linux-sunxi/meta-sunxi #263: cannot patch linux-mainline on warrior

bernardoaraujor, 2019-08-02:
```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "orange-pi-zero"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7.1"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:01b8a8b54bc569e5ef3f5e6fc6abcee365ab25d9"
meta-oe              = "warrior:8d5dcd6522e9d15e68637b6d7dda0401f9bb91d0"
meta-sunxi           = "warrior:7c6cefcfe655b16285436d5d44560aedff8eb5e0"

Initialising tasks: 100% |#####################################| Time: 0:00:00
Sstate summary: Wanted 323 Found 0 Missed 323 Current 78 (0% match, 19% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-mainline-4.19.46-r0 do_patch: Could not apply patches for orange-pi-zero.
ERROR: linux-mainline-4.19.46-r0 do_patch: Patch failures can be resolved in the linux source directory /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/linux-4.19.46)
ERROR: linux-mainline-4.19.46-r0 do_patch: Function failed: do_patch (log file is located at /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/temp/log.do_patch.14283)
ERROR: Logfile of failure stored in: /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/temp/log.do_patch.14283
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/bernardo/yocto/poky/meta/recipes-extended/xz/xz_5.2.4.bb:do_populate_sysroot', '/home/bernardo/yocto/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', '/home/bernardo/yocto/poky/meta/recipes-devtools/quilt/quilt-native_0.65.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['xz-native', 'kern-tools-native', 'quilt-native', 'automake-native', 'gnu-config-native', 'libtool-native', 'gettext-minimal-native', 'autoconf-native', 'texinfo-dummy-native', 'm4-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_patch
| (1/2) 0003-ARM-dts-nanopi-neo-air-Add-WiFi-eMMC.patch
| [INFO]: check of .kernel-meta//patches//./0003-ARM-dts-nanopi-neo-air-Add-WiFi-eMMC.patch with "git am" did not pass, trying reduced context.
| (2/2) 0001-add-wifi-support.patch
| [INFO]: check of .kernel-meta//patches//./0001-add-wifi-support.patch with "git am" did not pass, trying reduced context.
| [INFO]: Context reduced git-am of .kernel-meta//patches//./0001-add-wifi-support.patch with "git am" did not work, trying "apply".
| error: patch failed: arch/arm/boot/dts/sun8i-h2-plus-orangepi-zero.dts:80
| error: arch/arm/boot/dts/sun8i-h2-plus-orangepi-zero.dts: patch does not apply
| [ERROR]: Application of .kernel-meta//patches//./0001-add-wifi-support.patch failed.
|          Patch needs to be refreshed. Sample resolution script:
|              .git/rebase-apply/resolve_rejects
| ERROR: Could not apply patches for orange-pi-zero.
| ERROR: Patch failures can be resolved in the linux source directory /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/linux-4.19.46)
| WARNING: /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/temp/run.do_patch.14283:1 exit 1 from 'exit 1'
| ERROR: Function failed: do_patch (log file is located at /home/bernardo/yocto/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/linux-mainline/4.19.46-r0/temp/log.do_patch.14283)
ERROR: Task (/home/bernardo/yocto/meta-sunxi/recipes-kernel/linux/linux-mainline_4.19.46.bb:do_patch) failed with exit code '1'
```

nandra, 2019-08-02:
Thanks for report. Pls give me some time to look on issue (probably patch must be only regenerated on  top of 4.19 kernel). marek

bernardoaraujor, 2019-08-02:
@nandra that sounds like a good guess.

If the patches were generated from a different kerbely version, the commit hashes will clash.

nandra, 2019-08-02:
#246 should revolve it (didn't test on board just compile tested)

nandra, 2019-08-07:
Fix merged. Closing.

closed 2019-08-07 by nandra