linux-sunxi/meta-sunxi #267: where do we keep firmware files for touchscreens?

slapin, 2019-09-10:
There are machine-specific touchscreen firmware files for popular Silead touchscreens. They are extracted from vendor device drivers. Where these are generally stored so recipes could pick them up? I made touch on my tablet work (generic nameless a13-mid), it would be not nice if I can't share.

nandra, 2019-09-10:
We have firmware for wifi in repo: https://github.com/linux-sunxi/meta-sunxi/tree/master/recipes-kernel/xradio-firmware so maybe we should move it to sunxi-firmware or so directory and put everything in.

slapin, 2019-09-10:
there is gsl-firmware repo in github, but I do not know how well it is
maintained, will try to contact there, but will initially add to
linux-sunxi repo.

nandra, 2019-11-07:
@slapin any update on this?

slapin, 2019-11-07:
Not yet as I got very well into actual touch working as I was not able to
fix offset problem.

nandra, 2020-05-26:
@slapin ping? Can we close this or?

slapin, 2021-03-31:
Hi,
Sorry, life took its toll. Feel free to decide on this for now, but I guess
some final decision should be commented on here
(I still have the stuff and it works more or less fine, would be nice to
finally put it where it belongs).

Thanks!

nandra, 2021-03-31:
@slapin maybe try to push your changes and we can discuss in pull request ;). Thanks.

closed 2021-12-15 by nandra