linux-sunxi/meta-sunxi #269: support mali r9p0

smzahraee, 2019-09-18:
is it possible to support newest mali blob with [mripard repository]( https://github.com/mripard/sunxi-mali) repository

nandra, 2019-09-18:
Hi @smzahraee . I have experimental branch with this located here: https://github.com/nandra/meta-sunxi/commits/mbe-gles-mali It's not yes pushed to meta-sunxi as I didn't verify if it works with opengl. Help welcomed ;)

smzahraee, 2019-09-18:
Hi thanks 
do you test lima on mainline kernel? 
is it comparable with mali blobs?

nandra, 2019-09-18:
Hi not yet but I have plan to do it. If I have something I'll share it here. Thanks.

nandra, 2019-10-05:
@smzahraee did you try my branch? Does it work for your needs? Thanks.

nandra, 2019-10-05:
I'm in process now to take this recipe ```https://layers.openembedded.org/layerindex/recipe/97617/``` and try to poke with configuration and play with latest kernel to check if acceleration works .

smzahraee, 2019-10-05:
I will test it in following week and inform you, thank you so much

nandra, 2019-10-16:
@smzahraee ping, any update on this? Thanks.

nandra, 2020-05-26:
Due inactivity, closing.

closed 2020-05-26 by nandra