linux-sunxi/meta-sunxi #275: USB not working on Bananapi m1 plus (power problem)

tanya1997, 2019-11-30:
Hello.
This is my bblayers.conf:
```
BBLAYERS ?= " \
  /media/empyrean/L/yocto/sources/poky/meta \
  /media/empyrean/L/yocto/sources/poky/meta-poky \
  /media/empyrean/L/yocto/sources/poky/meta-yocto-bsp \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-oe \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-multimedia \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-python \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-networking \
  /media/empyrean/L/yocto/sources/meta-sunxi \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-perl \
  /media/empyrean/L/yocto/sources/meta-openembedded/meta-gnome \
  "
```
and local.conf changes:
```
MACHINE="bananapi"

SSTATE_MIRRORS = "\
file://.* http://sstate.yoctoproject.org/dev/PATH;downloadfilename=PATH \n \
file://.* http://sstate.yoctoproject.org/2.6/PATH;downloadfilename=PATH \n \
file://.* http://sstate.yoctoproject.org/2.7/PATH;downloadfilename=PATH \n \
"
```
The branch version is warrior (I tried to use thud and zeus too) and kernel version is 4.19 (I tried to use 5.0.19 and other).  I tried to build: core-image-base, core-image-full-cmdline and core-image-sato. And I always got the same result.
Here is a part of dmesg output:
```
[    1.881235] vcc3v0: disabling
[    1.889935] vcc5v0: disabling
[    1.898413] usb0-vbus: disabling
[    1.907036] usb1-vbus: disabling
[    1.915557] usb2-vbus: disabling
```
This is the full dmesg output: [dmesg_bananapi_m1_plus_usb_problem.txt](https://github.com/linux-sunxi/meta-sunxi/files/3907237/dmesg_bananapi_m1_plus_usb_problem.txt)

lsusb return nothing. 
And I don't see any information about usb, when I run lsmod:
```
lsmod
Module                  Size  Used by
8021q                  28672  -2
garp                   16384  -2
stp                    16384  -2
mrp                    20480  -2
llc                    16384  -2
bnep                   20480  -2
bluetooth             311296  -2
ecdh_generic           28672  -2
```
This is not hardware problem, because usb ports work fine, when I use other images. Ethernet and HDMI work fine. I didn't have any warning message, while I was building image.

I tried to use sunxi version , but I got this error:
```
| make: Leaving directory '/media/empyrean/L/yocto/build/tmp/work/bananapi-poky-linux-gnueabi/u-boot-sunxi/1_v2014.04+gitAUTOINC+ea1ac32bf7-r0/git'
| The base file '.config' does not exist.  Exit.

```
This is the full log: [err_bananapi_sunxi.txt](https://github.com/linux-sunxi/meta-sunxi/files/3907354/err_bananapi_sunxi.txt)

nandra, 2019-11-30:
I don't have bananapi unfortunately so cannot really help. I would just check if usb is enabled in kernel config (check menuconfig and maybe compare with other working image zcat /proc/config.gz). If you please find something pls share it here with community. Thanks a lot.

slapin, 2019-12-02:
I'd check PMIC settings. My wild guess is in your dt nothing claims power sources you need so power gets disabled. Please check device dt regarding PMIC and USB.

tanya1997, 2020-01-29:
I found, that usb ports work fine on yocto Jethro (2.0) and early versions. Usb have not working since Krogoth (2.1)

This is the Jethro dmesg output: [dmesg_jethro.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130057/dmesg_jethro.txt)
This is the Jethro dts file: [sun7i-a20-bananapi-m1-plus.dts.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130087/sun7i-a20-bananapi-m1-plus.dts.txt). I created it from dtb file  (sun7i-a20-bananapi.dtb). I used this command for it (`dtc -b 0 -O dts -I dtb -o sun7i-a20-bananapi-m1-plus.dts '/media/empyrean/bananapi/sun7i-a20-bananapi.dtb`'). Github doesn't accept a dts file (I changed it to *.dts.txt). 
This is the Jethro  /proc/config.gz file:  [zcat_configgz_jethro.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130609/zcat_configgz_jethro.txt)

This is the Krogoth dmesg output: [dmesg_krogoth.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130359/dmesg_krogoth.txt)
This is the Krogoth dts file: [sun7i-a20-bananapi-m1-plus.dts.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130378/sun7i-a20-bananapi-m1-plus.dts.txt)
This is the Krogoth  /proc/config.gz file: [zcat_configgz_krogoth.txt](https://github.com/linux-sunxi/meta-sunxi/files/4130535/zcat_configgz_krogoth.txt)

I didn't find any interesting things inside menuconfig, that may turn off usb (maybe I'm wrong). But, there are a lot of changes inside dts files.

I used the 5V 2A adapter for both tests. This is not hardware problem. 
The usb ports turned off after 200ms -1s after power on. (I used flash-usb with led for check it) 
And turned on during kernel load. (Only Jethro)

nandra, 2020-01-30:
Hmm hard to say but I see in krogoth log that regulators for usb are disabled while in jethro are not:
```
[    1.875064] usb0-vbus: disabling
[    1.883866] usb1-vbus: disabling
[    1.892521] usb2-vbus: disabling
```

Maybe you can enable some more debugging info for usb in kernel and try to check. Also good option would be to test actual master (we should have pretty recent kernel) and report or ask to linux mailing list to get help. Thanks.

tanya1997, 2020-02-03:
I found a solution. I added AXP20x power supply driver to the mainline kernel (/meta-sunxi/recipes-kernel/linux/linux-mainline/arm/defconfig). 
I added these strings:
```
CONFIG_MFD_AXP20X_I2C=y
CONFIG_MFD_AXP20X=y
CONFIG_IIO=y
CONFIG_AXP20X_POWER=y
``` 
Both usb ports work fine now (except micro-usb - usb0-vbus). I tested it for 
Warrior (core-image-sato: [zcat_configgz_warrior.txt](https://github.com/linux-sunxi/meta-sunxi/files/4150023/zcat_configgz_warrior.txt), [dmesg_warrior.txt](https://github.com/linux-sunxi/meta-sunxi/files/4150024/dmesg_warrior.txt)), 
master (core-image-minimal: [zcat_configgz_master.txt](https://github.com/linux-sunxi/meta-sunxi/files/4149741/zcat_configgz_master.txt), [dmesg_master.txt](https://github.com/linux-sunxi/meta-sunxi/files/4149750/dmesg_master.txt)) 
and 
Krogoth (core-image-minimal: [zcat_configgz_krogoth.txt](https://github.com/linux-sunxi/meta-sunxi/files/4150106/zcat_configgz_krogoth.txt), [dmesg_krogoth.txt](https://github.com/linux-sunxi/meta-sunxi/files/4150108/dmesg_krogoth.txt))
branches.

Thank you!

nandra, 2020-02-03:
@tanya1997 thanks for reporting back. I'll create branch with this fix on master. Can you please re-test after change? Thanks.

tanya1997, 2020-02-04:
ok. 
Please write here after creating the branch.

nandra, 2020-02-05:
@tanya1997 can you pls share with me which machine do you build? Thanks.

tanya1997, 2020-02-05:
MACHINE="bananapi". (local.conf)
The board is Banana Pi BPI-M1+.

nandra, 2020-05-26:
@tanya1997 sorry for very late reply. Can you pls checkout this branch: ```bananapi-fix``` and try to build it. Thanks.

tanya1997, 2020-05-30:
I have tested this branch and it works fine. (My OS is Ubuntu 16.04)
I checked both build: core-image-minimal ([zcat_configgz_minimal](https://github.com/linux-sunxi/meta-sunxi/files/4705670/zcat_configgz_banana_minimal.txt),  [dmesg_minimal](https://github.com/linux-sunxi/meta-sunxi/files/4705671/dmesg_banana_minimal.txt))
and
core-image-sato ([zcat_configgz_sato](https://github.com/linux-sunxi/meta-sunxi/files/4705666/zcat_configgz_banana_sato.txt), [dmesg_sato](https://github.com/linux-sunxi/meta-sunxi/files/4705669/dmesg_banana_sato.txt)). 
I used USB-flash and USB keyboard/mouse for tests.

nandra, 2020-06-01:
ok thanks for help. I'll merge it then.

nandra, 2020-06-01:
Merged to master. Thanks.

closed 2020-06-01 by nandra