linux-sunxi/meta-sunxi #281: No audio devices on nanopi-neo-plus2

renshorus, 2020-02-04:
Hello,

I'm building an image for the nanopi neo plus 2 on `warrior` but when the nanopi is booted alsa shows `No soundcards found.` I do not see other errors during boot time and all seems to working fine. I cant seem to figure out whats going wrong. Could someone perhaps point me in the right direction? I'm using the following `local.conf` (I left out non-essential installs and configs):

```
MACHINE = "nanopi-neo-plus2"

EXTRA_IMAGE_FEATURES_append = " debug-tweaks "

USER_CLASSES = "buildstats image-mklibs"

PATCHRESOLVE = "noop"

BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

IMAGE_INSTALL_append = " kernel-modules "
... other image installs ...
IMAGE_INSTALL_append = " alsa-tools " 
IMAGE_INSTALL_append = " alsa-utils " 
IMAGE_INSTALL_append = " linux-firmware "
IMAGE_INSTALL_append = " alsa-lib "

NETWORK_MANAGER = "networkmanager"

LICENSE_FLAGS_WHITELIST += " gstreamer1.0-plugins-ugly-meta "

DISTRO_FEATURES_append = " systemd "
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED_append = " sysvinit "
VIRTUAL-RUNTIME_initscripts = ""

LICENSE_FLAGS_WHITELIST = "commercial"
LICENSE_FLAGS_WHITELIST += " gstreamer1.0-plugins-ugly-meta "

PACKAGECONFIG_append_pn-gstreamer1.0-plugins-bad = " webrtc webrtcdsp "
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-base = " opus "
PACKAGECONFIG_append_pn-libnice = " gupnp "
 
PREFERRED_VERSION_linux-mainline = "5.0%"

KERNEL_IMAGETYPES = "Image Image.gz"

HORUS_SID_MMAP_SIZE = "1024"
HORUS_SID_MMAP_ADDR = "0x01c14000"
HORUS_SID_MMAP_OFFS = "0x200"

IMAGE_FSTYPES = "ext4"

CONF_VERSION = "1"
```

nandra, 2020-02-05:
@renshorus are you sure you've added audio device to kernel config? Usually this is cause for such errors. Actual kernel config is default one and doesn't have device specific extensions. Thanks.

renshorus, 2020-02-06:
That was it, I already added the CODEC config entries but I also had to add `CONFIG_DMA_SUN6I=y`  Took me a while to figure out. I now have a `fragment.cfg` containing: 
```
CONFIG_SND_SUN4I_CODEC=y
CONFIG_SND_SUN8I_CODEC=y
CONFIG_SND_SUN8I_CODEC_ANALOG=y
CONFIG_SND_SUN50I_CODEC_ANALOG=y
CONFIG_SND_SUN4I_I2S=y
CONFIG_SND_SUN4I_SPDIF=y
CONFIG_SND_SUN8I_ADDA_PR_REGMAP=y
CONFIG_DMA_SUN6I=y
```

nandra, 2020-02-06:
Thanks a lot for reporting back. To have maybe other user not hit same issue can you pls create PR with those chnages for your device? So we can have them merged and working for everyone ;). Thanks.

Momomentum, 2020-05-21:
Hi I have a similar issue as described. I'm fairly new to Yocto, so please bear with me. I get the same alsa error message saying that it cannot find any audio devices. Do I always need to modify the kernel config, if yes could you give me a hint how to do that?

What I have so far:
```
# meta-custom/recipes-custom/images/custom-image.bb
require recipes-core/images/core-image-minimal.bb

IMAGE_INSTALL += "linux-firmware kernel-modules alsa-utils alsa-lib"
```
local.conf is untouched except for `MACHINE ??= orange-pi-pc`. 
I am using the dunfell branch.

nandra, 2020-05-21:
You need to create linux-mainline.bbappend file which enable above configurations. Please look how to create kernel configuration fragment [here](https://www.yoctoproject.org/docs/1.6.1/dev-manual/dev-manual.html#creating-config-fragments)

Momomentum, 2020-05-23:
I am sorry I got back to you so late. Thank you for the pointers. It seems like my kernel fragment does not get included in the resulting kernel configuration. At least the settings do not appear in  /proc/config.gz. 
Here is what I did so far:
1. Create a bbappend-file
```
# meta-custom/recipes-kernel/linux/linux-mainline_%.bbappend
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
SRC_URI += "file://custom.cfg"
```
2. Create a config fragment with the suggested options
```
# meta-custom/recipes-kernel/linux/linux-mainline/custom.cfg
CONFIG_SND_SUN4I_CODEC=y
CONFIG_SND_SUN8I_CODEC=y
CONFIG_SND_SUN8I_CODEC_ANALOG=y
CONFIG_SND_SUN50I_CODEC_ANALOG=y
CONFIG_SND_SUN4I_I2S=y
CONFIG_SND_SUN4I_SPDIF=y
CONFIG_SND_SUN8I_ADDA_PR_REGMAP=y
CONFIG_DMA_SUN6I=y
```
3. Clean sstates (virtual/config, linux-mainline, custom-image)
4. Rebuild

What am I missing here?

nandra, 2020-05-26:
First try use:
```
bitbake-layers show-appends linux-mainline
```
if your bbappend is found.

Momomentum, 2020-05-27:
Thanks for getting back to me. The bbappend file is found
Here the output:
```
=== Matched appended recipes ===
linux-mainline_4.19.103.bb:
  /Yocto/poky/build/../../meta-custom/recipes-kernel/linux/linux-mainline_%.bbappend
linux-mainline_5.4.18.bb:
  /Yocto/poky/build/../../meta-custom/recipes-kernel/linux/linux-mainline_%.bbappend
linux-mainline_5.5.3.bb:
  /Yocto/poky/build/../../meta-custom/recipes-kernel/linux/linux-mainline_%.bbappend
```

nandra, 2020-05-27:
ok fine do you have in directory where bbappend file is directory named:
```
linux-mainline
```
which contains ```custom.cfg``` file? Thanks.

Momomentum, 2020-05-27:
Yes I have that folder.
BTW I use the generated sdimage file in (`tmp/deploy/images/orange-pi-pc/custom-image-orange-pi-pc.sunxi-sdimg`). Do I somehow need to need to copy the contents of `modules-orange-pi-pc.tgz` onto the sdcard as well?

nandra, 2020-05-27:
OK fine. Do you have kernel-modules package installed? You can check in build directory /tmp/work/../linux-mainline/build/.config if it contains your options enabled.

Momomentum, 2020-05-27:
So the linux-mainline/build/.config file does not contain the options. There is a file called custom.cfg in `linux-mainline/build`-folder.

nandra, 2020-05-27:
Please double check if you have following structure:
```
.
├── linux-mainline
│   └── axp20x.cfg
└── linux-mainline_%.bbappend

```

and content:
```
cat linux-mainline_%.bbappend 
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI_append = " file://axp20x.cfg \
"
```

But if file is there this seems to be OK. You would need to inspect logs why it happened and options are not taken into consideration. There is task which combine defconfig with fragments. Which branch do you use? I'm using dunfell and with above works perfectly fine.

Momomentum, 2020-05-27:
Does it make a difference if I use `SRC_URI += file://custom.cfg` compared to `SRC_URI_append = " file://custom.cfg "`? That seems to be the only difference.

nandra, 2020-05-27:
This is only other definition. I posted what yesterday confirmed and works on dunfell branch. Pls try to look in linux-mainline logs (search for cfg file name), maybe you spot something ;)

Momomentum, 2020-05-27:
I just tried out `SRC_URI_append` - it makes a difference. The custom.cfg options still are not included in `.config`-file. Also when trying booting now I get a kernel-exception. I'm not sure if this is progress or unrelated.

Momomentum, 2020-05-27:
Anyway thanks for all your help - I really appreciate it. I will go look through the logs.

nandra, 2020-05-27:
OK np. Please get back with findings so maybe we can extend (create) wiki how to do such tasks properly. Thanks.

Momomentum, 2020-05-28:
Alright I finally got it working. Unfortunately I am not a 100% certain, what exactly fixed it. Basically I started from scratch and then did the following:
1. Generate a .config file by running `bitbake linux-mainline -c menuconfig`. In the menu I searched for all the configuration options that renshorus suggested and set them.
2. Generate a fragment.cfg by running `bitbake linux-mainline -c diffconfig`.
3. Copy resulting fragment.cfg to meta-custom/recipes-kernel/linux/linux-mainline/fragment.cfg
4. Add 
```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"
SRC_URI += "file://fragement.cfg"
```
to meta-custom/recipes-kernel/linux/linux-mainline_%.bbappend
5. Rebuild

What I noticed was that a lot more config entries were added to the fragment.cfg file by running the diffconfig command. So I guess some of those options are tied together and one option needs to have the other option set in order to be applied?

nandra, 2020-05-28:
Yes that could be caused that you enable some options but there was missing dependencies to kernel won't take it. can we then close it? Thanks.

Momomentum, 2020-05-28:
Yes sure. And thanks again!

closed 2020-05-28 by nandra