linux-sunxi/meta-sunxi #282: Support for Banana-Pi M4 ?

hsnhrn, 2020-02-05:
Can this layer be also used for M4? if not how could we use it for m4?

nandra, 2020-02-05:
@hsnhrn absolutely. We need to create new machine config and we currently have pretty recent kernel version so I suppose it should work. Pull requests are welcomed ;). If you need hand just say. :+1:

hsnhrn, 2020-02-05:
Thanks for a quick reply. let me try :D

hsnhrn, 2020-02-05:
MACHINE ??= "raspberrypi"
what should i put here?

nandra, 2020-02-05:
Well first you need to create machine configuration (look at [https://github.com/linux-sunxi/meta-sunxi/tree/master/conf/machine](machines) for inspiration)

hsnhrn, 2020-02-05:
require conf/machine/include/sun7i.inc

KERNEL_DEVICETREE = "sun7i-a20-bananapi.dtb"
UBOOT_MACHINE = "Bananapi_config"
SUNXI_FEX_FILE = "sys_config/a20/Bananapi.fex"

Thats what there for bananapi a20.. I dont have much knowledge about low level kernel development. :(

hsnhrn, 2020-02-05:
Is it possible for you to help?

nandra, 2020-02-05:
Sure but seems like this board doesn't have allwinner CPU (I overlooked it). So it doesn't fit unfortunately this meta-sunxi which have support for allwinner CPU arch. Thanks.

closed 2020-02-05 by nandra