linux-sunxi/meta-sunxi #286: Can't connect to wifi using wpa_supplicant

HerrMuellerluedenscheid, 2020-04-27:
Dear community,

I can't figure out how to connect the orange-pi-zero wifi to a local network using `wpa_supplicant`. I have used `wpa_supplicant` quite often, thus I'm aware how to basically set it up. I have the feeling that something is odd with the encryption during the authentication but I'm not sure. This is my wpa configuration:

```
ctrl_interface=/var/run/wpa_supplicant
ctrl_interface_group=0
update_config=1

network={
        ssid="abcdefg"
        #psk="11223344"
        psk=b757aa74ba63f1d91ea8350c5f2281d66cbe240d5d46cd0703b88e84e8ebf598
}
```

This is the output after firing up `wpa_supplicant`

```
root@orange-pi-zero:~# wpa_supplicant -c/etc/wpa_supplicant.conf -iwlan0 
Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan0: SME: Trying to authenticate with 6e:c7:ec:c2:ef:f7 (SSID=[ 2973.659645] wlan0: authenticate with 6e:c7:ec:c2:ef:f7
'abcdefg' freq=2437 MHz)
[ 2973.670529] wlan0: send auth to 6e:c7:ec:c2:ef:f7 (try 1/3)
[ 2973.694283] wlan0: authenticated
wlan0: Trying to associate with 6e:c7:ec:c2:ef:f7 (SSID='abcdefg[ 2973.698611] wlan0: associate with 6e:c7:ec:c2:ef:f7 (try 1/3)
' freq=2437 MHz)
[ 2973.712901] wlan0: RX AssocResp from 6e:c7:ec:c2:ef:f7 (capab=0x431 status=0 aid=1)
[ 2973.724210] wlan0: associated
wlan0: Associated with 6e:c7:ec:c2:ef:f7
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: WPA: Failed to set PTK to the driver (alg=3 keylen=16 bss[ 2973.805777] wlan0: deauthenticating from 6e:c7:ec:c2:ef:f7 by local choice (Reason: 1=UNSPECIFIED)
id=6e:c7:ec:c2:ef:f7)
[ 2973.820497] ieee80211 phy0: no station to remove
wlan0: CTRL-EVENT-DISCONNECTED bssid=6e:c7:ec:c2:ef:f7 reason=1 locally_generated=1
wlan0: WPA: 4-Way Handshake failed - pre-shared key may be incorrect
wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="abcdefg" auth_failures=1 duration=10 reason=WRONG_KEY
wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
wlan0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
```
It clearly says `wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="abcdefg" auth_failures=1 duration=10 reason=WRONG_KEY` but I am certain that the credentials are correct. I had different ones before using non-alphanumeric symbols. I changed it to rule out any encoding issues. I'm running out of ideas what else to try. Do you have an idea what is going on?

I'm using warrior branch, orange-pi-zero, xradio driver, on mainline kernel at version 5.2. I also tested on mainline at version 4.19.88 with the same result.
Any hint is highly appreciated!

Regards and thanks in advance

HerrMuellerluedenscheid, 2020-04-27:
I had to enable the following kernel modules:
```
CONFIG_LIB80211_CRYPT_CCMP=y
CONFIG_CRYPTO_CCM=y
CONFIG_CRYPTO_GCM=y
CONFIG_CRYPTO_ARC4=y
```

Now it works fine!

closed 2020-04-27 by HerrMuellerluedenscheid

nandra, 2020-04-28:
@HerrMuellerluedenscheid thanks for info ;).