linux-sunxi/meta-sunxi #288: uart1 & uart2 not working on Nano-Pi-neo

PJfixer, 2020-05-02:
Hi all, 

For a project i build a "core-image-full-cmdline" images on Nano-Pi-neo board.
i add a custom program (basic hello world) to test uart but i got no output on uart1(ttyS1) & uart2(ttyS2) but it work on uart0(ttyS0) so i think the program is ok.

for the wiring i am referring to this  : http://wiki.friendlyarm.com/wiki/images/c/c4/NEO_pinout-02.jpg

i was thinking , that maybe its related to the device tree ? any advice ? 

regards
Pierre.

PJfixer, 2020-05-02:
[image]

i'm not very experienced in BSP , but after display the device-tree 
 i see that there is no "phandle" address specified for the uart1 & uart2 so maybe this why it doesn't work ?

i imagine "phandle" address is a kind of a address that point to the uart0 peripheral base in the H3 processor ? i'm right ?

regards,
Pierre

PJfixer, 2020-05-02:
[image]

Using the datasheet, i find the base address of the uarts, and i search them in the device tree:
[image]

uart1 & uart2 are configured as "disabled"  so i think my assumption were correct ?
So if i'am right, where in the meta-sunxi are the file i need to edit so that the device tree will be build with the three uart enabled ?

nandra, 2020-05-02:
You need to change state from "disabled" to "okay" and create patch with this change and create bbappend recipe with that patch for your specific mate-layer.

PJfixer, 2020-05-03:
Thanks for your answer @nandra ,

I search how to create a patch in yocto environment

I think i need the sun8i-h3-nanopi-neo dts file ? in order to make a copy edit it and create the patch with diff right ?

But currently i don't know where find the sun8i-h3-nanopi-neo.dts file ? and i don't uderstand how the dtb file is generated during the build of the image ?

Could you please explain this ? :) 

Regards,

Pierre.

nandra, 2020-05-04:
Usually you can use in yocto [devtool](https://wiki.yoctoproject.org/wiki/TipsAndTricks/Patching_the_source_for_a_recipe)

PJfixer, 2020-05-05:
Hi all,  

I finally patch the device by creating a .patch in my meta : https://github.com/PJfixer/meta-pjdrones/blob/master/recipes-kernel/linux/patchs/0001-uart1-uart2-dts-patch.patch

Just for information if some beginners like me read this post :

Its not enough to just change uart status to "okay" you have to take care that there is "pinctrl-0" parameter assigned with a phandler.

The phandler point at a pin configuration node in the device tree which specify the TX & RX pins.

In my case the pin configurations for uart1 & uart2 didn't have a phandler so i define them, and took care that the values were not already in use.

After rebuild the images both uart are finally working fine.

thanks @nandra for the tips.

regards,

Pierre.

closed 2020-05-05 by PJfixer

nandra, 2020-05-06:
@PJfixer thanks for sharing.