linux-sunxi/meta-sunxi #293: do we have a fex file for orange pi zero board?

grugrugru1234, 2020-09-25:
Hi guys,
I was building a custom image and I was trying to read ADC value from another ICs using the SPI1 bus on the orange-pi-zero board and I couldn't get it to work (message error). So I looked it up and the documentation says that the board needs a fex file. Then I found out that the orange-pi-zero board doesn't have any fex file listed to build in the orange-pi-zero.conf configuration file. Plus the recipe also clearly says that it's only compatible with the sun4i/sun5i/sun7i devices whereas the orange-pi-zero is sun8i.  
However, I found a fex file here: 
https://github.com/linux-sunxi/sunxi-boards/blob/master/sys_config/h3/xunlong_orange_pi_pc.fex
which was for the pc version but still it confuses me. Does this mean that:
- The recipe and the configuration file need update and you still need a fex file for the orange-pi-zero board
or 
- The fex file is optional for the sun8i devices and you have to write your own recipe to include the fex file?
And I'm using linux 5.4.18 if that's relevant.
I'm new with the board and sunxi devices so I don't know what's the right thing to do.

nandra, 2020-10-14:
Hi, IIRC fex files are supported only with sunxi old kernel. We use mainline now you can use overlay (or simple patch dts for your needs). Thanks

nandra, 2021-03-31:
@grugrugru1234 ping? Did you resolve it? Can you please share? Thanks.

grugrugru1234, 2021-03-31:
I did solve it. How can I share it with you?

nandra, 2021-03-31:
Please write it as a text and I'll add it to wiki so then if somebody will search can get inspiration ;). Thanks.

closed 2021-04-17 by nandra