linux-sunxi/meta-sunxi #307: Forfun Q88DB Boots (mainline u-boot & kernel) but missing drivers?

bvarner, 2021-01-08:
Recently I got hold of a [Forfun Q88DB](https://linux-sunxi.org/Forfun_Q88DB) and decided to try out building a project around it.

Using a machine with the following config, I can get the device to boot `core-image-minimal` and `core-image-sato` with working framebuffer and UART.

```
#@TYPE: Machine
#@NAME: Q8 A13 MID
#@DESCRIPTION: Machine configuration for A starndard Q8 style tablet device based on allwinner a13 CPU
#https://linux-sunxi.org/Q8

require conf/machine/include/sun5i.inc

KERNEL_DEVICETREE = "sun5i-a13-q8-tablet.dtb"
UBOOT_MACHINE = "q8_a13_tablet_defconfig"
SUNXI_FEX_FILE = "sys_config/a13/a13_mid.fex"
```

However, literally nothing else works. No `/dev/input/` nodes. dmesg reports rtc and two of the regulators in error states, and there's no wifi, bluetooth, or sysfs nodes for cpufreq, etc.

I'm not sure if I'm running into a DTS issue or a kernel defconfig issue. 

I figure it's one of the two either the device tree isn't right, or the kernel modules necessary for the devices in the tree aren't getting configured in mainline since I didn't add a machine override for the kernel config.

If anyone can give me some pointers I'll be happy to address the issues in my fork and PR the results.

bvarner, 2021-01-09:
It may be a mix of issues, but so far it's clearly an issue of kernel drivers / modules not loading.

nandra, 2021-01-09:
Kernel defconfig inthis repo is generic one so it could happen that not all your device/drivers are loaded. Also there could be some parts missing in device-tree. I would focus on checking and adding it step by step what is necessary for you project. Please check if there are some clones of kernel for this device you can get some inspiration ;)

bvarner, 2021-01-11:
Thanks, @nandra. It looks like it's almost all drivers just not included.

Is the best practice to make a new layer based on this layer for my project / board, instead of trying to make a machine config and PR that back up to this repo?

Do the machine configs in this repo have the same situation for other boards?

bvarner, 2021-01-11:
Closing issue, as I've decided to do just that -- leverage this layer, but build my own layer up to handle kernel / additional config appends. Thanks you all for maintaining this layer! It's helpful!

closed 2021-01-11 by bvarner

nandra, 2021-01-11:
@bvarner thanks for sharing. If you have done modifications etc. you can post pull request and we can merge you machine changed to this layer. 

For kernel defconfig -> well we have setup just minimum stuff but we already have some overlays for various boards so maybe that would be an option also for your machine.

Thanks.