linux-sunxi/meta-sunxi #310: orange pi zero 2 - when will it be supported?

linuxgit88, 2021-03-16:
orange pi zero 2 - when will it be supported?

nandra, 2021-03-16:
Hard to say. Usually board owners post patch which add support for specific board and can the test. Thanks.

nandra, 2021-04-17:
Based on this wiki: https://linux-sunxi.org/Xunlong_Orange_Pi_Zero2 seems some parts for u-boot was merged to u-boot v2021.04 + some linux support is also added.

linuxgit88, 2021-04-22:
@nandra Cool, thanks for informing everyone!

nobodyguy, 2021-09-17:
Have anyone been succesfull with yocto build with the latest uboot + patched kernel (apritzel/h616-v9)?

nandra, 2021-09-17:
@nobodyguy do you experience some issues using this layer with support for opizero 2? Thanks.

nobodyguy, 2021-09-17:
@nandra I haven't tried it yet, will do it during this weekend and let you know the result.

nandra, 2021-09-17:
OK thanks. Do you have board to try? If need help pls write here I'll be able to help.

nobodyguy, 2021-09-17:
@nandra Thank you very much, yes I have a opizero2 board here.

nandra, 2021-09-17:
@nobodyguy ok pla form meta-sunxi and try to add support then we can review and merge back to meta-sunxi ;)

nandra, 2021-09-20:
@nobodyguy any update ;)? Thanks.

pblxptr, 2022-07-27:
Did anyone manage to succeed with Yocto with this board?I see that @nobodyguy was supposed to give it a try but I see no further comments and @nandra has closed the issue.

nandra, 2022-07-28:
As mentioned above u-boot changes should be mainlined, for kernel I found this fork: https://github.com/apritzel/linux/commits/h616-v5. I can prepare this kernel version to meta-sunxi and enable it for this device. But I have no way to test it. @pblxptr do you have board to try pls? If yes then we can collaborate and make it support. Thx.

nobodyguy, 2022-07-28:
Sorry, I had to postpone it because of a high work load and I also saw that none of the Armbian images for OPI Zero 2 actually booted.
We can give it a shot now.

nandra, 2022-07-28:
OK let me try to add support for it and if it builds then you can try. Thx.

jskarzynski, 2022-07-28:
I have an intention to prepare support for this board. I managed to boot it with kernel that @nandra mentioned.
I will also try to bring up UI graphics and video decoders. But it will be somewhere in the future, as I'm not focused on this board for now.

nobodyguy, 2022-07-28:
@jskarzynski That's great news that it boots.
@nandra Thank you, I'll prepare my meta layer meanwhile.

nandra, 2022-07-28:
@jskarzynski do you have some fork of meta-sunxi where can can collaborate? Thanks.

pblxptr, 2022-07-28:
> As mentioned above u-boot changes should be mainlined, for kernel I found this fork: https://github.com/apritzel/linux/commits/h616-v5. I can prepare this kernel version to meta-sunxi and enable it [...]

Unfortunately I don't have this board yet, so sadly I'm not able to help.

jskarzynski, 2022-07-28:
> @jskarzynski do you have some fork of meta-sunxi where can can collaborate? Thanks.

Some time ago I've created locally a new meta-layer for Zero 2 only, because I planned to use some weird drivers. But now I can make a fork of meta-sunxi and migrate changes important for us for this topic.

Also now I see on Orange PI website that there is some new software support. We should check this out.

jskarzynski, 2022-08-04:
Ok, I ported my changes to latest kirkstone of meta-sunxi. It's booting, but that's all. There is no Ethernet, WiFi, BT, HDMI etc.
@nobodyguy You can test it from here:
https://github.com/jskarzynski/meta-sunxi/tree/orange-pi-zero-2-poc

Also I found some side-topics that should be addressed:
- support for OPI Zero 2, which is present in https://github.com/apritzel/linux was merged recently to the main kernel's repository! It will be present in next kernel release. I can create recipe for master branch of mainline kernel in meta-sunxi and remove linux-h616 recipe.
- github.com/apritzel/arm-trusted-firmware is no longer maintained and there is no support for H616, we should migrate to mainline repository of ARM (github.com/ARM-software/arm-trusted-firmware)
- There are recipes in meta-sunxi that are not migrated to kirkstone's syntax, we should fix that. Also license of Linux kernel is deprecated.

nobodyguy, 2022-08-05:
@jskarzynski Great, thank you.
I'll test it during the weekend and let you know.

pblxptr, 2022-08-10:
@jskarzynski @nobodyguy 

My board will arrive today so I can also have a look at this.

nandra, 2022-08-15:
Any update on that? Thanks.

pblxptr, 2022-08-17:
Hello, nothing new from me. I was able to boot it and get into kernel but that's all (also managed to to this using Kernel master branch). Just for the purpose of tests I used armbian to verify if everything is working properly on this board and it seems that Ethernet is working but there were some problems with wifi drivers. So far I'm busy with something at my work but probably I should get back to this in let's say two weeks and try to at least enable Ethernet.

MartinLoeper, 2022-08-25:
Just want to let you know guys: We at Nesto will look into this too in a few weeks too since we plan to replace some of our hardware which is currently hard to source with the orangepi zero 2.

pblxptr, 2022-08-29:
> Any update on that? Thanks.

Just slight update from me, it seems that I've managed to run Ethernet on Linux Master.

`Configuring network interfaces... [   11.048530] dwmac-sun8i 5020000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.148929] dwmac-sun8i 5020000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
[   11.159003] dwmac-sun8i 5020000.ethernet eth0: No Safety Features support found
[   11.166332] dwmac-sun8i 5020000.ethernet eth0: No MAC Management Counters available
[   11.174002] dwmac-sun8i 5020000.ethernet eth0: PTP not supported by HW
[   11.180992] dwmac-sun8i 5020000.ethernet eth0: configuring for phy/rgmii link mode
udhcpc: started, v1.35.0
udhcpc: broadcasting discover
[   13.246195] dwmac-sun8i 5020000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
udhcpc: broadcasting discover
udhcpc: broadcasting select for 192.168.1.100, server 192.168.1.1
udhcpc: lease of 192.168.1.100 obtained from 192.168.1.1, lease time 7200
/etc/udhcpc.d/50default: Adding DNS 192.168.1.1
done.
Starting syslogd/klogd: done

Poky (Yocto Project Reference Distro) 4.0.2 orange-pi-zero2 /dev/ttyS0

orange-pi-zero2 login: root
root@orange-pi-zero2:~# lsblk
-sh: lsblk: not found
root@orange-pi-zero2:~# cat interface
cat: can't open 'interface': No such file or directory
root@orange-pi-zero2:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 02:00:7A:0B:7B:4F  
          inet addr:192.168.1.100  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1240 (1.2 KiB)  TX bytes:684 (684.0 B)
          Interrupt:31 `

nandra, 2022-09-09:
> @nandra
> 
> Just slight update from me, it seems that I've managed to run Ethernet on master branch of Linux
> 
> `Configuring network interfaces... [ 11.048530] dwmac-sun8i 5020000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0 [ 11.148929] dwmac-sun8i 5020000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211F Gigabit Ethernet] (irq=POLL) [ 11.159003] dwmac-sun8i 5020000.ethernet eth0: No Safety Features support found [ 11.166332] dwmac-sun8i 5020000.ethernet eth0: No MAC Management Counters available [ 11.174002] dwmac-sun8i 5020000.ethernet eth0: PTP not supported by HW [ 11.180992] dwmac-sun8i 5020000.ethernet eth0: configuring for phy/rgmii link mode udhcpc: started, v1.35.0 udhcpc: broadcasting discover [ 13.246195] dwmac-sun8i 5020000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx udhcpc: broadcasting discover udhcpc: broadcasting select for 192.168.1.100, server 192.168.1.1 udhcpc: lease of 192.168.1.100 obtained from 192.168.1.1, lease time 7200 /etc/udhcpc.d/50default: Adding DNS 192.168.1.1 done. Starting syslogd/klogd: done
> 
> Poky (Yocto Project Reference Distro) 4.0.2 orange-pi-zero2 /dev/ttyS0
> 
> orange-pi-zero2 login: root root@orange-pi-zero2:~# ifconfig eth0 Link encap:Ethernet HWaddr 02:00:7A:0B:7B:4F inet addr:192.168.1.100 Bcast:192.168.1.255 Mask:255.255.255.0 UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1 RX packets:3 errors:0 dropped:0 overruns:0 frame:0 TX packets:2 errors:0 dropped:0 overruns:0 carrier:0 collisions:0 txqueuelen:1000 RX bytes:1240 (1.2 KiB) TX bytes:684 (684.0 B) Interrupt:31 `

Perfect!

pblxptr, 2022-11-25:
> 

Slight update from me, I got back to work on this (I have very little free time, so I'm working on this only from time to time..) but I actually managed to run wifi chip on this board, and build it in Yocto but I used linux-orangepi kernel with a few modifications since the mainline lacks (even the newest one) almost everything so patches are needed, which wasn't necessary while using linux-orangepi kernel. Also, as far as I know, the UWE5622 wifi driver (which is not available in the mainline kernel) is compatible with 5.15 kernel. So the next step I'm going to take is to try to use the mainline 5.15 kernel with appropriate patches.

nandra, 2022-11-28:
@pblxptr thanks for update!

jskarzynski, 2022-12-01:
I think we should prepare some checklist with things to bring-up.
And provide comments how precisely we were able to enable specific feature (kernel source and version, etc.).
If you have some work done, it would be great to share our forks.

pblxptr, 2022-12-02:
I agree, I will clean up a bit of what I've already done and share my fork. Also, I agree as for the list it will be easier to see what is to be done.

ein-shved, 2022-12-10:
@jskarzynski  @pblxptr Hi! Seems that we are working on the same topic. I made the [wiki page](https://nixos.wiki/wiki/NixOS_on_ARM/Orange_Pi_Zero2_H626) and now trying to make usb work. I seems succeeded with correct dts overlay for usb and started with corect kernel configuration.

nandra, 2023-01-16:
@pblxptr any update on this pls?We can create wiki subpage with instructions how to build etc. Thanks.

pblxptr, 2023-01-17:
Hello, not really, I've been busy at work, I had a hard crunching time before the new year so I had no capacity to work on this, however I got back to this last weekend, prepared a setup again and I am going to do some cleanup. Then I will push my changes.

pblxptr, 2023-01-27:
Hi, 
@nandra here is my fork https://github.com/pblxptr/meta-sunxi/tree/orange-pi-zero-2-poc of @jskarzynski fork.
I've added some patches that enable Ethernet and Wifi, some of them I've taken from other projects, and some of them I had to modify. I used the device tree shipped with Linux 6.1.2 with some modifications.
It works with mainline Linux 5.15. 

//local.conf

`MACHINE ?= "orange-pi-zero2"
CORE_IMAGE_EXTRA_INSTALL:append = " ethtool networkmanager-nmcli kernel-modules linux-firmware uwe5622-firmware wpa-supplicant"
KERNEL_MODULE_AUTOLOAD += "sprdwl_ng"

DISTRO_FEATURES:append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"`

```
root@orange-pi-zero2:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 02:00:7A:0B:7B:4F  
          inet addr:192.168.0.242  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::b65f:31cc:c879:b81e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:82 errors:0 dropped:2 overruns:0 frame:0
          TX packets:46 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:26798 (26.1 KiB)  TX bytes:4476 (4.3 KiB)
          Interrupt:31 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:160 errors:0 dropped:0 overruns:0 frame:0
          TX packets:160 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11040 (10.7 KiB)  TX bytes:11040 (10.7 KiB)

wlan0     Link encap:Ethernet  HWaddr 68:B8:35:88:05:58  
          inet addr:192.168.0.116  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::38a3:3896:cf1c:2c7b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1591 (1.5 KiB)  TX bytes:1894 (1.8 KiB)
```
May I ask somebody to test it?

In the meantime I'm wondering about further steps, how should we actually approach this?

On one hand, Orange Pi Zero 2 is officially supported by mainline Kernel 6x but the support lacks some features e.g. wifi/bt support (for some people I guess the most important ones). I haven't found any wifi driver that would work with the newest kernel.

On the other hand, we have Kernel 5.15 (patched) that supports I guess most of the features but if we want to support them we have to patch the kernel as I did. 

What are your opinions about it?
I've already tried to compile the wifi driver in Kernel 6x, (just out of curiosity, because I knew it would highly likely fail) it failed but I haven't taken a detailed look at what went wrong because I was focused on enabling it in Kernel 5x. I will have a look at this again but this time I will try to find out if it's just a matter of Kernel's API changes which can be - let's hope - painfully adopted in wifi driver or if it's something more complicated.

nandra, 2023-01-27:
@pblxptr thanks for an update. From my point of view I would stick for kirkstone with 5.15 and in master add 6.0 kernel with this device even some parts (like wifi) are not fully working. We can make a note and in meantime work on it. Open to other suggestions.

pblxptr, 2023-01-30:
> @pblxptr thanks for an update. From my point of view I would stick for kirkstone with 5.15 and in master add 6.0 kernel with this device even some parts (like wifi) are not fully working. We can make a note and in meantime work on it. Open to other suggestions.

Sounds good to me.

nandra, 2023-02-01:
@pblxptr ok I'll took that fork and merge to this repo. Would you help me with testing please then. I'll ping you when have branches ready. Thanks.

pblxptr, 2023-02-01:
@nandra sure, just let me know :)

nandra, 2023-02-02:
@pblxptr please try opi2-zero branch in this repo. For me it builds fine. Thanks.

pblxptr, 2023-02-03:
> @pblxptr please try opi2-zero branch in this repo. For me it builds fine. Thanks.

I will check it probably on Sunday evening

nandra, 2023-02-06:
@pblxptr any update from you pls? Thanks.

pblxptr, 2023-02-06:
@nandra  it builds but stucks booting at uboot, will be investigating it.

Berveglieri, 2023-02-06:
Hello everyone, I just got a board opi-zero-2 and can help in the tests, I really would like to have yocto being fully supported for this board.
I can even test with TPM, I have a TPM module from infineon model slb9670.
I am not a C developers but I can help with any code that is python or Go.

Thank you

pblxptr, 2023-02-06:
@nandra I did just a quick look and see two things that might have caused boot problems:
- in my machine config there was MACHINEOVERRODES with values equal to sun50i and sun50i-h616, these SOC families are used in uboot and atf recipes 
- you haven't picked commits that @jskarzynski made, there was a change in atf recipe

Will have a look later

pblxptr, 2023-02-07:
@nandra 
I managed to boot it. I see a few issues:
- as above, in my fork, there was an MACHINEOVERRIDES used in the machine file. Without it uboot is not properly configured, as well as atf. We can e.g. create a new include file like sun50i-h616 or something.
- you've removed following lines from the machine file:

UBOOT_BINARY = "u-boot-nodtb.bin"
SPL_BINARY = "u-boot-sunxi-with-spl.bin"

They are needed

- you haven't committed a kernel config so wifi and ethernet drivers are not being compiled and included

We have to fix it if we want to merge it. Maybe it would be better if I created a pull request directly from my fork, then we would have a clear view of what we want to add, what we want to remote, and so on.

nandra, 2023-02-07:
@pblxptr ok I think pull request on top of kirkstone would be the best. Then I can review and we can discuss things. Sorry for double work. Thanks.

pblxptr, 2023-02-07:
@nandra I think that it will be just easier to keep the conversation, and comments  'closer' to the implementation in a pull request :) I've created it, you can have a look at this.

nandra, 2023-02-22:
pull request #366 merged. So we have Opi zero2. Thanks everybody involved. @pblxptr if you'll have time pls make other PR for master (even some peripherals won't work). Thanks.

closed 2023-02-22 by nandra