linux-sunxi/meta-sunxi #313: poky (dunfell) core-image-base doesn't boot

ipiazza, 2021-03-28:
Hello Team,
I am trying to create a custom image with Yocto and I am facing some issues at boot. The image baking ended successfully, though I am unable to let the image boot.
Down below the local.conf and bblayers.conf files followed by the minicom output of the boot up sequence.

### local.conf ###
`grep -v ^$ local.conf |grep -v ^\#`
 MACHINE ??= "orange-pi-zero"
DEFAULTTUNE = "cortexa7hf-neon-vfpv4"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_ipk"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
CONF_VERSION = "1"
BB_NUMBER_THREADS = "32"
PARALLEL_MAKE = "-j 32"
CORE_IMAGE_EXTRA_INSTALL += "dropbear htop opencv python3-opencv python3 python3-tensorflow-lite"
### bblayers.conf ###
`grep -v ^$ bblayers.conf |grep -v ^\#`
POKY_BBLAYERS_CONF_VERSION = "2"
BBPATH = "${TOPDIR}"
BBFILES ?= ""
BBLAYERS ?= " \
  /yocto-opizero/poky/meta \
  /yocto-opizero/poky/meta-poky \
  /yocto-opizero/poky/meta-yocto-bsp \
  /yocto-opizero/meta-openembedded/meta-oe \
  /yocto-opizero/meta-openembedded/meta-python \
  /yocto-opizero/meta-openembedded/meta-networking \
  /yocto-opizero/meta-openembedded/meta-multimedia \
  /yocto-opizero/meta-sunxi \
  /yocto-opizero/meta-tensorflow-lite \
  "
### minicom /dev/ttyUSB0 ###
U-Boot SPL 2020.01-g303f8fe (Jan 06 2020 - 20:56:31 +0000)                                       
DRAM: 512 MiB                                                                                    
Trying to boot from MMC1                                               
                                                                       
                                                                       
U-Boot 2020.01-g303f8fe (Jan 06 2020 - 20:56:31 +0000) Allwinner Technology
                                                                       
CPU:   Allwinner H3 (SUN8I 1680)                                       
Model: Xunlong Orange Pi Zero                                          
DRAM:  512 MiB                                                         
MMC:   mmc@1c0f000: 0, mmc@1c10000: 1                                  
Loading Environment from FAT... *** Warning - bad CRC, using default environment
                                                                       
In:    serial                                                          
Out:   serial
Err:   serial
Net:   phy interface0
eth0: ethernet@1c30000
starting USB...
Bus usb@1c1a000: USB EHCI 1.00
Bus usb@1c1a400: USB OHCI 1.0
Bus usb@1c1b000: USB EHCI 1.00
Bus usb@1c1b400: USB OHCI 1.0
scanning bus usb@1c1a000 for devices... 1 USB Device(s) found
scanning bus usb@1c1a400 for devices... 1 USB Device(s) found
scanning bus usb@1c1b000 for devices... 1 USB Device(s) found
scanning bus usb@1c1b400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
714 bytes read in 2 ms (348.6 KiB/s)
\#\# Executing script at 43100000
20893 bytes read in 3 ms (6.6 MiB/s)
4413896 bytes read in 200 ms (21 MiB/s)
\#\# Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-5.4.69
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4413832 Bytes = 4.2 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
\#\# Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
EHCI failed to shut down host controller.
   Loading Kernel Image
   Loading Device Tree to 49ff7000, end 49fff19c ... OK

Starting kernel ...
***
Then it never moves forward..
Can you please help me out understanding what I am missing here?
If you need anything from the tmp/deploy directory I am happy to provide it.

Thanks,
RJ

nandra, 2021-03-29:
Did you made some other changes? Maybe change SERIAL_CONSOLE or so which print kernel messages to other console? I'm using orangepi-zero dunfell build and it boots without issues. Can you try to run ```bitbake -e core-image-base | grep 'SERIAL_'```. Thanks.

ipiazza, 2021-03-29:
Actually the only changes I made are the ones in *local.conf* and *bblayers.conf*.
Down below the command you asked.
`bitbake -e core-image-base | grep 'SERIAL_'`
`# $SERIAL_CONSOLE [3 operations]`
`SERIAL_CONSOLE="115200 ttyS0"`
`# $SERIAL_CONSOLES [2 operations]`
`#     [_defaultval] "${@d.getVar('SERIAL_CONSOLE').replace(' ', ';')}"`
`#   "${@d.getVar('SERIAL_CONSOLE').replace(' ', ';')}"`
`SERIAL_CONSOLES="115200;ttyS0"`
`# $SERIAL_CONSOLES_CHECK`
`#     [doc] "Similar to SERIAL_CONSOLES except the device is checked for existence before attempting to enable it. Supported only by SysVinit."`
`# $SYSLINUX_SERIAL_TTY`

Could it be anything with the kernel itself? I didn't do anything to compile it.. Maybe some tweaking is needed?
Thanks!

nandra, 2021-03-30:
@ipiazza I found fix and it's pushed to dunfell release branch. Related PR to master is : #314

ipiazza, 2021-03-31:
Thanks @nandra! I am going to pull it, recompile and flash the sdcard with the new image.
I'll provide an update and eventually close the issue.

nandra, 2021-03-31:
@ipiazza just out of curiosity -> I saw your trying to compile also meta-tensorflow-lite. Will you try to use google coral with opi-zero? Thanks. :+1:

ipiazza, 2021-03-31:
I'am trying to see how much I can get in performance by doing object detection from rtsp video streams with a tf quantized model. If you are interested I can tell you more.
Now back to the issue. The uboot one is now sorted but but I'm getting kernel panic cause the root partition doesn't get recognized as a valid filesystem and thus it keeps rebooting.
The whole boot sequence and sdcard layout can be found here: [https://pastebin.com/p56nvDit](url)

Should I close this issue and file a new one or it is fine to keep working with this one? Kindly let me know.
 
Thanks a lot!

nandra, 2021-03-31:
@ipiazza working fine here:
```
[    1.229166] sun8i-h3-pinctrl 1c20800.pinctrl: 1c20800.pinctrl supply vcc-pg not found, using dummy regulator
[    1.236426] mmc0: new SDHC card at address 0001
[    1.247103] ehci-platform 1c1b000.usb: EHCI Host Controller
[    1.251317] mmcblk0: mmc0:0001 SD16G 14.5 GiB 
[    1.256259] ehci-platform 1c1b000.usb: new USB bus registered, assigned bus number 3
[    1.268603] ehci-platform 1c1b000.usb: irq 29, io mem 0x01c1b000
[    1.270660]  mmcblk0: p1 p2
[    1.298718] ehci-platform 1c1b000.usb: USB 2.0 started, EHCI 1.00
[    1.305522] hub 3-0:1.0: USB hub found
[    1.309362] hub 3-0:1.0: 1 port detected
[    1.314068] ohci-platform 1c1b400.usb: Generic Platform OHCI controller
[    1.320741] ohci-platform 1c1b400.usb: new USB bus registered, assigned bus number 4
[    1.328588] ohci-platform 1c1b400.usb: irq 30, io mem 0x01c1b400
[    1.403380] hub 4-0:1.0: USB hub found
[    1.407186] hub 4-0:1.0: 1 port detected
[    1.412791] sunxi-mmc 1c10000.mmc: allocated mmc-pwrseq
[    1.591287] sunxi-mmc 1c10000.mmc: initialized, max. request size: 16384 KB
[    1.598373] sun6i-rtc 1f00000.rtc: setting system clock to 2018-03-09T12:40:05 UTC (1520599205)
[    1.616762] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
[    1.624984] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    1.633000] devtmpfs: mounted
[    1.637333] Freeing unused kernel memory: 1024K
[    1.647608] mmc1: new high speed SDIO card at address 0001
[    1.668903] Run /sbin/init as init process
[    2.054029] random: fast init done
[    2.086295] udevd[117]: starting version 3.2.9
[    2.146394] random: udevd: uninitialized urandom read (16 bytes read)
[    2.155498] random: udevd: uninitialized urandom read (16 bytes read)
[    2.162062] random: udevd: uninitialized urandom read (16 bytes read)
[    2.204930] udevd[119]: starting eudev-3.2.9
[    2.817995] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    2.823470] ext4 filesystem being remounted at / supports timestamps until 2038 (0x7fffffff)
hwclock: settimeofday: Invalid argument
[    3.333886] urandom_read: 2 callbacks suppressed
[    3.333895] random: dd: uninitialized urandom read (512 bytes read)
ALSA: Restoring mixer settings...
INIT: Entering runlevel: 5
Configuring network interfaces... /usr/sbin/alsactl: load_state:1735: No soundcards found...
[    3.487480] dwmac-sun8i 1c30000.ethernet eth0: PHY [0.1:01] driver [Generic PHY]
[    3.489654] random: udevd: uninitialized urandom read (16 bytes read)
```

ipiazza, 2021-03-31:
thanks for the fast response. I dd'ed the rootfs.ext4 file to the mmcblk0p2 and it worked.
I am going to close the issue.
Thanks so much for the support, very much appreciated!

closed 2021-03-31 by ipiazza