linux-sunxi/meta-sunxi #318: Make layer ready for hardknott

TomHanlein, 2021-04-22:
Hi!

Yocto 3.3 hardknott was released a couple of days ago. Please be so kind and upgrade the meta-sunxi layer to match the new release.

At the moment I get the error:
```
ERROR: Layer meta-sunxi is not compatible with the core layer which only supports these series: hardknott (layer is compatible with dunfell zeus gatesgarth)
```

Thanks!
Tom

nandra, 2021-04-22:
Hi @TomHanlein . I've tried to build hardknott 2 days ago but I hit an issue like:

```
WARNING: core-image-base-1.0-r0 do_rootfs: libglib-2.0-0.postinst returned 126, marking as unpacked only, configuration required on target.
WARNING: core-image-base-1.0-r0 do_rootfs: eudev-hwdb.postinst returned 126, marking as unpacked only, configuration required on target.
ERROR: core-image-base-1.0-r0 do_rootfs: Postinstall scriptlets of ['libglib-2', 'eudev-hwdb'] have failed. If the intention is to defer them to first boot,
then please place them into pkg_postinst_ontarget_${PN} ().
Deferring to first boot via 'exit 1' is no longer supported.
Details of the failure are in /home/builder/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_rootfs.
ERROR: Logfile of failure stored in: /home/builder/build/tmp/work/orange_pi_zero-poky-linux-gnueabi/core-image-base/1.0-r0/temp/log.do_rootfs.8491
ERROR: Task (/home/builder/poky/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'
```

So I need to get in touch with poky maintainers to check what problem can be. Keep result posted here.

nandra, 2021-04-30:
OK despite of fact build issue is not caused by our layer I've added hardknott compatible + create hardknott branch.

closed 2021-04-30 by nandra